betaflight/betaflight #10151: Is it possible to ignore a string between limiting characters on the MSP connection

Thorone24, 2020-08-29:
I have problems connecting to my flight controller with the Bluetooth srial module RN4871. The module spits out an String on sucessful connection with my smartphon that contains the word "PARAM . This marke the FC change into DFU mode and de initialisation attempt od the speedy bee app fails.
As far as i understand it, the RN4871 is not able to suppress theese status strings gut one can change pre and post delimiters to them, so that the firmware recieving them can ignore them. Is there already a function to make betaflight ignore something on the MSP interface or is it possible to add such something like this?
The complete string, that is send by the RN4871 is "(CONNECT,1,5626E904A579)(CONN_PARAM,0006,0000,07D0)(STREAM_OPEN)(CONN_PARAM,0027,0000,07D0)(CONN_PARAM,0018,0000,0200)" while '(' and ')' are exemplary delimiters

Thorone24, 2020-08-30:
Just another thought: why does the processor go to the DFU Mode  with an simple "R" in the first place? This seems very unsave to me. Shouldnt the be a more complex command?

mikeller, 2020-08-31:
@Thorone24:
You should be able to masquerade the connection string as an MSP error message by prefixing it with `$M!<length of connection string in bytes - 2>`.
Re the `R` to trigger a reboot, this is legacy, probably from back in the days of MultiWii. There is an actual MSP command to trigger a reboot as well now, with parameters for booting into different modes. The fact that the `R` was never reported as causing problems most likely means that it is safe for normal applications.

Thorone24, 2020-08-31:
@mikeller Hi, do you habe an example for this? I tried several combinations but always ended up rebooting or having the interface hung up... Unfotunately the RN4871 has a maximum of 4 chars for e fornt and back delimiters to the status strings. So theres only a few possibilities. Furthermore the status strings contain variing numbers of bytes. 

I nothing else works, this would be an easy to add function but i think it will not be used widely aside my application

mikeller, 2020-09-01:
@Thorone24: Not sure what more example you need than `$M!<length of connection string in bytes - 2>`? The length is just one character containing the 8 bit value for the length, so this should fit into 4 bytes. But if the length of this in-band message is variable then I do not see a good way to make this work - my recommendation is to get yourself a bluetooth adapter that is fully transparent and does not send any in-band messages.

Thorone24, 2020-09-02:
This masking does not work for me (probably I'm just too stupid) but if its only for fixed bytelengths its not suited anyways.

Maybe I#m overseeing something, but adding

                case '[':
                    mspPort->c_state = MSP_STRINGIGNORE;
                    break;

 to line 125 in mspSerialProcessReceivedData() and 

		case MSP_STRINGIGNORE:      // Ignore all chars until ] appears
			if(c==']')
				mspPort->c_state = MSP_IDLE;
			break;

to line 131 in the same function together with defining MSP_STRINGIGNORE will already lead to a behavior of the MSP ignoring all bytes between $[ and ]

I understand there not much motivation to change this... so maybe i try to modify the Firmware of the RN4871 . Unfortunately there not much alternatives to this chip as it weights only 0.3g and is only 8x10mm in size including the antenna.

mikeller, 2020-10-04:
@Thorone24:
- MSP does not define any fixed message delimiters like `[` or `]` - it uses a message length field to detect the end of a message;
- even if it did include delimiters, there is no guarantee whatsoever that a `]` seen in the stream would be the end of a message - since all bytes inside a message can have any value, they can be `]`.

All in all it looks like the hardware you are trying to use is not usable with MSP, since MSP requires a transparent serial connection that is free of in-band control information, and the hardware does not meet these requirements.

closed 2020-12-25 by stale[bot]