betaflight/betaflight #10152: Defaults nosave giving an error

JustAskDave, 2020-08-31:
**Describe the bug**
When trying to reload a backup from the same firmware version, the defaults nosave option comes back with a red error - 
###ERROR: set: INVALID NAME###.

Then at the very end of the CLI page it show in red:
###ERROR: save: ERRORS WERE DETECTED - PLEASE REVIEW BEFORE CONTINUING###
###ERROR: save: PLEASE FIX ERRORS THEN 'SAVE'###

**To Reproduce**
Every time I try reloading the programming, or, just manually typing defaults nosave gives the same error.

**Expected behavior**
I would expect no error when sending defaults nosave!

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

``` # 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.2.2 Aug 16 2020 / 01:47:01 (e833ac612) MSP API: 1.43
# config: manufacturer_id: TTRH, board_name: TRANSTECF411, version: c03dceea, date: 2019-11-28T09:29:00Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name TRANSTECF411
manufacturer_id TTRH
mcu_id 006700513437510832363736
signature 

# resources
resource MOTOR 1 B10
resource MOTOR 2 B09
resource MOTOR 3 B00
resource MOTOR 4 B01

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 3 1700 2100 0 0
aux 1 1 0 925 1600 0 0
aux 2 2 0 1625 2100 0 0
aux 3 13 1 1300 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 500

# master
set gyro_lowpass2_hz = 238
set dyn_lpf_gyro_min_hz = 190
set dyn_lpf_gyro_max_hz = 475
set acc_calibration = -5,60,407,1
set serialrx_provider = IBUS
set motor_pwm_protocol = DSHOT600
set align_board_roll = 180
set align_board_yaw = -90
set bat_capacity = 800
set current_meter = VIRTUAL
set vbat_scale = 109
set ibata_scale = 130
set ibata_offset = 220
set ibatv_scale = 130
set ibatv_offset = 220
set pid_process_denom = 1
set osd_vbat_pos = 2465
set osd_flymode_pos = 2100
set osd_vtx_channel_pos = 2442
set osd_ah_pos = 2126
set osd_current_pos = 2455
set osd_mah_drawn_pos = 2487
set osd_battery_usage_pos = 2472
set osd_rate_profile_name_pos = 2081
set osd_pid_profile_name_pos = 2113
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 1
set vtx_freq = 5843
set gyro_1_sensor_align = DEFAULT

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 67
set dyn_lpf_dterm_max_hz = 162
set dterm_lowpass2_hz = 143
set level_limit = 15

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 67
set dyn_lpf_dterm_max_hz = 162
set dterm_lowpass2_hz = 143
set level_limit = 35

profile 2

# profile 2
set dyn_lpf_dterm_min_hz = 67
set dyn_lpf_dterm_max_hz = 162
set dterm_lowpass2_hz = 143

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 30
set pitch_rc_rate = 30
set yaw_rc_rate = 30
set roll_expo = 75
set pitch_expo = 75
set yaw_expo = 75
set throttle_limit_type = SCALE
set throttle_limit_percent = 50

rateprofile 1

# rateprofile 1
set roll_rc_rate = 50
set pitch_rc_rate = 50
set yaw_rc_rate = 50
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set throttle_limit_type = SCALE
set throttle_limit_percent = 75

rateprofile 2

# rateprofile 2
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 2

# save configuration
save
# ```

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

``` # resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: ADC_BATT
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: BEEPER
B07: FREE
B08: FREE
B09: MOTOR 2
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: INVERTER 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 1
TIM3:
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM4:
    CH4: MOTOR 2
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; TRANSTECF411 - purchased direct from Transtec
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable]; Flysky FS-A8S receiver, VTX provided by Transtec - don't know what it is
 - how are the different components wired up.-  Everything is simply plugged together, these were designed as a plug and play kit for school kids to build

**Additional context**
Add any other context about the problem here.
If I simply type save and press enter all seems to work OK, it just wont auto save when loading the backup.

mikeller, 2020-08-31:
@JustAskDave: This sounds like you are trying to restore a `diff` that was created by a different version of Betaflight to what you are running now - this is not supported and will lead to the errors that you describe.
When updating Betaflight versions, start with a fresh configuration, and apply all of the changes you want, and use the tuning guide for the version that you are using to tune.

JustAskDave, 2020-08-31:
I was wondering that too, but I get the same fault from a fresh install of 4.2.2 using Configurator 10.7.0, no changes made, just type in defaults nosave and the error shows up:

```
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# defaults nosave

# resetting to defaults
###ERROR: set: INVALID NAME###
```

haslinghuis, 2020-08-31:
Please flash again with `Full chip erase` enabled.

mikeller, 2020-08-31:
@JustAskDave: Now I get it - this is caused by a bug in the Unified Target configuration for your board, which tries to set a parameter that no longer exists: betaflight/unified-targets#341

closed 2020-09-01 by mikeller