betaflight/betaflight #10154: Dark Mode Issue on a couple of selectors in BF 10.8.0 Configurator

TimCiarlette, 2020-08-31:
Dark mode not functioning correctly in the BF 10.8.0 Configurator. This is working fine in the BF 10.7 Configurator.
The Board Selector on the Firmware Flasher Page and the Blackbox Mode Selector on the Blackbox Tab.

See pics below:
[image]
[image]

Asizon, 2020-08-31:
Good find @TimCiarlette ! This is related to the new Select2 implementation, lets fix it

TimCiarlette, 2020-08-31:
First, I want to thank all of the people that work on the Betaflight software team. Its awesome, I love it.
Any questions, I'm there. Thanks.