betaflight/betaflight #10155: [MAVLink] Receive Bytes over UART

nilay994, 2020-09-01:
Related to issues #6579 and #8867 

**Bug**
Not able to receive any serial UART bytes in the callback of MAVLink port (UART1). Tried checking the `TX` pin of the port, it is indeed giving out MAVLink telemetry. Also, was able to parse the bytes that were sent out by using MAVROS on a PC. Connected flight computer with a USB-FTDI cable (3.3V), applied standard serial port settings - 115200, no flow or parity, 1stop bit.

**Steps to reproduce the behavior:**
1. [Here](https://github.com/nilay994/betaflight/commit/033b896db3fa34e1f1bfa7ed1df2f4ee9776f16a) is a commit that enables receiving mavlink bytes on a UART port (rebased to a week old master).
2. Turned on MAVLink telemetry and serial rx on configurator.
3. To see if MAVLink packets can be received on the `RX` pin, tried to do a loopback test (`TX = RX`). 

**Expected behavior**
Tried to blink one of the onboard LEDs when few bytes are received (was the only way I thought I could debug). However, no toggles on the LED GPIO are observed - so I infer that the UART port does not receive any bytes.

**Flight controller configuration**
```
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.3.0 Aug  7 2020 / 18:56:46 (norevision) MSP API: 1.44

# start the command batch
batch start

board_name OMNIBUSF4SD

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# serial
serial 5 576 115200 57600 115200 115200

# master
set acc_calibration = -644,-319,-228,1
set serialrx_provider = SPEK2048

profile 0

rateprofile 3

# end the command batch
batch end
```

**Resources**:
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A15: OSD_CS
B00: FREE
B01: FREE
B03: BARO_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D02: FREE

Currently active Timers:Resources
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - OMNIBUSF4 (SD) V3 AIO No other connections made to the board.

**Additional context**
I'm not used to Betaflight firmware development yet. Feel like I am missing something simple in either the `Configurator settings` or somewhere in the `mavlink.c` file. Read somewhere that `UART6` of `OMINBUS F4 (SD)` has a hardware inversion on `RX` so avoided using it.

mikeller, 2020-09-01:
@nilay994: Looks all good at a glance, this _should_ work.

nilay994, 2020-09-01:
> @nilay994: Looks all good at a glance, this _should_ work.

Shall recheck, any suggestions on a better way to debug than checking bytes using LED toggles? Is it possible to have console based debug printfs over VCP?

etracer65, 2020-09-01:
@nilay994 See: https://github.com/betaflight/betaflight/pull/8905

mikeller, 2020-09-01:
@nilay994: The simplest way to log in Betaflight is to assign numerical debug output to the `debug[]` array and then watch the live output in the Sensors tab / log it into a blackbox log / watch it in OSD.

nilay994, 2020-09-01:
How about `serialrx_provider = SPEK2048`?  Should it instead have been something like sbus? I couldn't find a default UART protocol option.

haslinghuis, 2020-09-01:
> ```
> # serial
> serial 5 576 115200 57600 115200 115200
> ```

Did you try a non-shared port?
Why not use MSP for input with something like https://github.com/christianrauch/ros-multiwii

Also:
Raspberry with Ez-Wifibroadcast image uses SUMD as Control Link.
Ardupilot supports bidirectional Mavlink with MAVROS.

nilay994, 2020-09-02:
@haslinghuis , will try a non-shared port.

Is reading from `MAVLINK_COMM_0` correct in that case?
https://github.com/nilay994/betaflight/blob/033b896db3fa34e1f1bfa7ed1df2f4ee9776f16a/src/main/telemetry/mavlink.c#L114

@mikeller 
are these port settings okay for enabling receiving on the same port these is telemetry?
https://github.com/nilay994/betaflight/blob/033b896db3fa34e1f1bfa7ed1df2f4ee9776f16a/src/main/telemetry/mavlink.c#L219
```
    mavlinkPort = openSerialPort(portConfig->identifier, FUNCTION_TELEMETRY_MAVLINK, mavlinkReceive, NULL, baudRates[baudRateIndex], MODE_RXTX, SERIAL_STOPBITS_1);
```

haslinghuis, 2020-09-02:
`MAVLINK_COMM_0` is a channel not a port 😕 

enum | mavlink_channel_t {  MAVLINK_COMM_0, MAVLINK_COMM_1, MAVLINK_COMM_2, MAVLINK_COMM_3,  MAVLINK_COMM_0, MAVLINK_COMM_1, MAVLINK_COMM_2, MAVLINK_COMM_3,  MAVLINK_COMM_0, MAVLINK_COMM_1, MAVLINK_COMM_2, MAVLINK_COMM_3}
-- | --

🧰 There is also Mavproxy, mavlink-router and MAVSDK 😛

nilay994, 2020-09-08:
Shall close the issue for now since I'm inactive. MAVLink RX is not working for me yet. Will re-open once I've tried your suggestions! Thanks a lot for the replies :)

closed 2020-09-08 by nilay994