betaflight/betaflight #10156: RevoltOSD

Rainwise69, 2020-09-01:
Hello, I have a problem when i install betaflight (i tried some versions 4.1 legacy,4.2) in my RevoltOSD. The Crossfire doesn't work, i config the ports top serial and choose the right options like i usually do in other builds but it doesn't work. When i reflash to flightone it works with no problems, is there something else that i need to do or is this a known issue? Thanks

mikeller, 2020-09-01:
@Rainwise69: Yes, this is a known issue - Revolt hardware uses UARTs in weird and non-standard ways, and the solder bridges on the boards that can be used to configure inversion and other UART parameters are not documented, making UART use in Betaflight a hit and miss on these boards.
If you want boards that work well with Betaflight then go for manufacturers that provide full specifications of their hardware to Betaflight, like Airbot or Matek.

Rainwise69, 2020-09-01:
Well, I have boards that work well with betaflight, that's not the point. If it doesn't work why is there a target? Is there anything that i can do?

mikeller, 2020-09-01:
@Rainwise69: Ask FlightOne to provide schematics for their boards to Betaflight.

Rainwise69, 2020-09-02:
So Betaflight makes a target that doesn't work? What for? And based on what schematics then if they don't have it?

etracer65, 2020-09-02:
> So Betaflight makes a target that doesn't work? What for? And based on what schematics then if they don't have it?

Betaflight didn't "make" the target. People from FlightOne submitted the target. So if you're concerned it doesn't work you should probably contact them.

And I doubt there's anything actually wrong with the target. Rather you're trying to follow the FlightOne wiring diagram for Crossfire which is not appropriate for Betaflight. First you need to avoid UART1 as it has some ridiculous jumpering logic. Second the RX/TX from the Crossfire receiver need to go to the **same** UART's TX/RX unlike the silly FlightOne method of splitting them across different UARTs (some of their diagrams even say to not connect one of the lines). So maybe use UART6 and connect the receiver RX to the FC TX and the reverse. The point is you'll never get it to work following the FlightOne wiring diagram.

Without some kind of schematic from FlightOne detailing the implementation there's not much additional help we can provide. It's up to the manufacturer to provide documentation on how to properly connect and use their board.

Rainwise69, 2020-09-02:
Didn't knew they were the ones submitting the target, but there's an answer I can work with, gonna try to change to Uart 6 then, thanks!

Rainwise69, 2020-09-03:
I placed it on Uart 4 and it works fine, thanks. Now it doens't arm lol the flags are only MSP wich should go away when I dc and power cycle. I tried to disable DJI MSP on uart 3 also and nothing. When I toggle the warning at the motor tab it arms :/ and if I dc after that it arms, but if I power cycle the problem comes again. Any Ideia?

etracer65, 2020-09-03:
https://www.youtube.com/playlist?list=PLwoDb7WF6c8n8SCsG7mtIgQUgsuWxaiiO

closed 2020-09-03 by etracer65