betaflight/betaflight #10157: AK8963 magnetometer is not recognized

krygacz, 2020-09-01:
**Describe the bug**
I have a GY-91 sensor board connected over I2C, which contains an MPU9250 (MPU6500 gyro + AK8963 mag) and a BMP280. The BMP is working fine (so the i2c connection between fc and the sensor board is ok), however I can't get the AK8963 mag to show up. MPU is in bypass mode, so the mag is directly visible on i2c. I've tested the sensor board on an Arduino with identical wiring and it detects the mag at the correct address, 0x0C. I've set the correct mag bus type, bus device and address but Betaflight does not see it and shows 6 i2c errors.

**Flight controller configuration**
```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.2 Aug 16 2020 / 01:48:06 (e833ac612) MSP API: 1.43
# config: manufacturer_id: TMTR, board_name: TMOTORF7, version: d230cc5f, date: 2019-10-19T23:29:18Z

# start the command batch
batch start

board_name TMOTORF7
manufacturer_id TMTR

# name: KRYGACZ

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# map
map TAER1234

# serial
serial 1 2 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# led
led 0 6,7::CW:11
led 1 6,8::CI:12
led 2 9,8::CI:12
led 3 9,7::CW:11

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 2 1400 1600 0 0
aux 2 13 1 1800 2100 0 0
aux 3 26 4 1400 1600 0 0
aux 4 31 4 1800 2100 0 0
aux 5 35 2 1800 2100 0 0
aux 6 36 1 1800 2100 0 0

# adjrange
adjrange 0 0 3 900 2100 12 3 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set acc_calibration = 53,0,-46,1
set mag_i2c_address = 12
set mag_hardware = AK8963
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 2
set rssi_channel = 16
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set vbat_warning_cell_voltage = 360
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set osd_vbat_pos = 2433
set osd_rssi_pos = 2104
set osd_flymode_pos = 2456
set osd_vtx_channel_pos = 2082
set osd_mah_drawn_pos = 2423
set osd_craft_name_pos = 2443
set osd_warnings_pos = 14697
set osd_avg_cell_voltage_pos = 2401
set osd_disarmed_pos = 2186
set vtx_power = 1
set vtx_freq = 5865
set vcd_video_system = NTSC
set dashboard_i2c_bus = 0
set camera_control_key_delay = 125
set camera_control_internal_resistance = 290
set camera_control_button_resistance = 460,280,160,77,9
set pinio_box = 39,0,0,0
set gyro_2_bustype = NONE
set name = KRYGACZ

profile 2

rateprofile 0

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: LED 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: CAMERA_CONTROL
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: FLASH_CS
C09: FREE
C10: SERIAL_TX 3
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: PINIO 1
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE

# 
```

**Setup / Versions**
 - Flight controller: T-Motor F7, Betaflight 4.2.2

haslinghuis, 2020-09-02:
Please try `set mag_i2c_address = 0 to use `default address of device`.
Maybe related to: https://github.com/betaflight/betaflight/issues/4531 (although other MAG).
On a long shot perhaps need `#define USE_I2C_PULLUP` in target.h?

krygacz, 2020-09-02:
Already tried the default address and played with i2c pullup and overclock settings, still nothing :c

etracer65, 2020-09-02:
I could be wrong about this since I didn't spend a lot of time looking at it, but I think Betaflight only supports the magnetometer in a MPU9250 in I2C slave mode, but when the MPU9250 is connected via SPI.

This would be configured with `set mag_bustype = SLAVE`. The embedded AK8963 is treated differently than a standalone device (which can be I2C or SPI). So you'd have to have the MPU9250 connected to the SPI bus. I don't see any logic to connect via I2C to the MPU9250 and then use slave I2C to the mag.

krygacz, 2020-09-02:
Thanks for the reply, to my understanding that's what the passthrough mode of the MPU9250 is for, it bridges slave i2c bus with the main one. Once this is enabled, i2c scanner on Arduino detects the magnetometer standalone at 0x0C, so I don't see any reason why Betaflight wouldn't

etracer65, 2020-09-02:
There is no logic in Betaflight to enable the serial interface bypass multiplexer when connected to the MPU9250 via I2C and this setting is disabled by default (and after reset). Enabling the bypass multiplexer only happens when using SPI and `set mag_bustype = SLAVE` as I mentioned above. In short it looks like there was no intention of using the AK8963 inside the MPU9250 the way you want and instead it was intended to be used **_along with_** the MPU9250. And since the MPU9250 is a supported gyro this would make sense that a board with this device would want to use it that way. If somebody wanted to just use a AK8963 it would make far more sense to use the standalone variant of that part.

It probably wouldn't be that hard to modify the initialization code to enable the bypass multiplexer if you want, but that would be yet another variant of the interface for this mag (spi, i2c, slave, +???). And frankly this is an extremely unusual application and the part is quite old so I don't see how this would make sense for inclusion into Betaflight.

krygacz, 2020-09-02:
Thanks, it turns out someone already coded the i2c slave driver on Inav: 
https://github.com/0zoon0/inav/commit/e079fab3098087323cf0a6dae88f8990f10f423e
https://github.com/iNavFlight/inav/issues/1759
I'll try to adapt this to betaflight and make a pull request if everything works

PR #10159 (2020-09-20): Support MPU925x internal magnetometer over I2C
Enable support of the MPU9250/9255's internal magnetometer AK8963 on I2C bus. Inspired by: https://github.com/0zoon0/inav/commit/e079fab3098087323cf0a6dae88f8990f10f423e 😃 

Fixes #10157
closed 2020-09-20 by mikeller