betaflight/betaflight #10158: BF4.3 - Tramp pitmode not working via CMS

JyeSmith, 2020-09-02:
In 4.2.2 turning pitmode on and off works (e.g. decrease and restore power level).  FYI Pitmode is not remembered by the Tramp after power cycling, so I actually dont see the point.

In 4.3 turning pitmode on and off via the CMS does nothing.  Tramps power on start is not altered and never enters pitmode.

Entering Tramp pitmode via its button on power up works.

**To Reproduce**
Flash a clean version of 4.2.2 or 4.3 with minimal setup for the vtx and rx.

Tagging @pgreenland since the change is most likely from this PR https://github.com/betaflight/betaflight/pull/9638. No idea what I was on when I previously tested it.

pgreenland, 2020-09-06:
Hey Jye.

I use pit mode to keep my vtx cool while not flying. It's tied to my arming system, it leaves pit mode on arming and re-enters on disarm.

I've just checked out and built the head of the master branch (1c3346bae14f0de9e7d37f566c96ed3eb8e1d226).

Toggling pit mode works fine for me from the radio via my configured mode, along with the VTX pit mode toggle in the OSD.

I'm not super familiar with betaflight and google / the wiki hasn't helped. CMS is the old name for the OSD by the looks of the source code. Is that what you meant?

I initially didn't think it was working from betaflight configurator, however it seems you have to hit the save button before the MSP command is sent to the drone, after which the status of the toggle control thingy is respected.

Not sure how useful that toggle switch is myself. I suppose it's useful for testing, but looking at the MSP processing code in betaflight, it's value is passed to the normal vtxCommonSetPitMode function to toggle pit mode, therefore it's not saved.

Could perhaps do with a note in configurator saying that it's for testing only and wont be persisted in the config.

Hope that helps,

Phil

JyeSmith, 2020-09-06:
@pgreenland Are you using an IMRC Tramp?

pgreenland, 2020-09-06:
@JyeSmith 

Afraid not, I'm using a mateksys vtx-mini: http://www.mateksys.com/?portfolio=vtx-mini

Sounds like you may have stumbled on a compatibility issue.

Do you get any status info reported via the OSD or betaflight configurator?

Without wanting to muddy the waters by linking to a pull request, I added some additional debug in https://github.com/betaflight/betaflight/pull/10166 and have attached a batch of unified build outputs. Could you configure your FC's debug to log the tramp parameters and capture say 10 seconds of tethered logging while toggling pit mode via an rx channel. Would be interesting to see how far the module gets.

I've not got my drone to hand atm otherwise I would point you at the CLI voodoo to select the VTX debug variables. Let me know if you cant find it.

Phil

JyeSmith, 2020-09-06:
I think firstly we need to make it work with the IMRC Tramp, then other vendors can correct their implementation.  Otherwise we are going to end up with another SA/AKK saga.

From what Ive seen on the logic analyser, to set pitmode ON you send a '0', while OFF is '1'.  But the return is ON '1', and OFF '0'.  4.2 handles this inversion at the point of writing to the Tramp.  I need to have a closer look at 4.3 to see how its done.

@jflyper @raphaelcoeffic since the OG implementation have you ever receive a protocol from IMRC?  I would tag @tonycake but hes not active here.

pgreenland, 2020-09-06:
Agreed.

Yeah that got me to start with, the inversion is currently performed in vtxTrampSetPitMode. I await the results of your debugging. Let me know if you need anything.

raphaelcoeffic, 2020-09-07:
@JyeSmith nope, never received anything.

closed 2020-11-05 by stale[bot]