betaflight/betaflight #10160: Negative Camera uptilt compensation

wsnzone, 2020-09-03:
Hi!
I always fly with 25-30 deg camera uptilt. I don't need any compensations, I do it automaticly by my habbits.

But once, i need to do slower cinematic flight with 0 camera tilt and my muscle memory just completely get wrong. 
I want to set uptilt compensation with -30 deg value to have a feel like my normal flying.
This also will be convinient for winter whoop sessions.

etracer65, 2020-09-04:
This would be very difficult to implement vs. the value gained and how many people would actually use this. The fpv camera angle is internally stored as an unsigned 8-bit integer so this would have to be changed to a signed variable and all the logic that uses it adjusted. This includes the MSP interface and the Configurator (which has to remain compatible with previous versions that only use unsigned values). So the point is it would be a very large amount of work and it really doesn't seem worth it.

RipperDrone, 2020-09-05:
Just a thought - could the unsigned variable carry values 0-360, thereby including negative uptilt?

RipperDrone, 2020-09-05:
Not sure how the RC axes compensations would act on values >90, though. Maybe there are discontinuities in the math for values greater than 90deg...

etracer65, 2020-09-05:
> Just a thought - could the unsigned variable carry values 0-360, thereby including negative uptilt?

No. 8 bit integer.

phobos-, 2020-09-06:
I actually looked into this problem last winter and came to the same conclusion as @etracer65 . Too much effort. Just do the mixing in your transmitter config.

wsnzone, 2020-09-07:
> > Just a thought - could the unsigned variable carry values 0-360, thereby including negative uptilt?
> 
> No. 8 bit integer.

it's possible to store trom -63 to 63 angle value

etracer65, 2020-09-08:
> it's possible to store from -63 to 63 angle value

Not in an 8 bit unsigned integer it's not.

wsnzone, 2020-09-08:
> > it's possible to store from -63 to 63 angle value
> 
> In an 8 bit unsigned integer it's not.

It's yes! If we talk about "store". we can interpritate this 8 bits as we want. We can think about 0 as about -63, 1 as -62, ets

etracer65, 2020-09-08:
> It's yes! If we talk about "store". we can interpritate this 8 bits as we want. We can think about 0 as about -63, 1 as -62, ets

Maybe you should read the earlier comments about how all the logic would have to change to understand the use of a signed value. The issue is not really about whether the signed value could be stored. But all the ramifications of doing that.

wsnzone, 2020-09-08:
> Maybe you should read the earlier comments about how all the logic would have to change to understand the use of a signed value. The issue is not really about whether the signed value could be [...]

I'ver read all the comments. And I can said what 8 bit integer can store angle from -63 to 63 value. I can agree what this issue maybe hard to implement. That where are a lot of work, etc. But it's possible!

etracer65, 2020-09-08:
> I'ver read all the comments. And I can said what 8 bit integer can store angle from -63 to 63 value. I can agree what this issue maybe hard to implement. That where are a lot of work, etc. But it's possible!

Great. Glad we wasted time with something we already knew.

closed 2020-11-05 by stale[bot]