betaflight/betaflight #10161: JHEF7UDAL target voltage sensor problem

geniack, 2020-09-03:
**Describe the bug**
I flashed the latest betaflight to a F7-XSD board (target: JHEF7UDAL, ). Everything works except the voltage sensor. I have a 4s battery connected and the configurator shows a 8s/9s battery with something around 35 V. I tried to use the calibration tool and set the scaling factor from 110 to about 50. This seem to  have corrected the voltage but then it doesn't matter if I connect an empty or a full battery, it always shows the same voltage. I tried going backwards to betaflight 4.1.x but the problem remains with every firmware.

**To Reproduce**
Flash betaflight to this flight controller, check battery voltage.

**Expected behavior**
Showing the correct battery voltage.

**Flight controller configuration**
It's just the default configuration of any betaflight going back to 4.1.x

**Setup / Versions**
F7-XSD Flight controller
link to hardware: https://tinyurl.com/yx9o5eq2
vbat is connected to + of lipo, gnd to gnd of lipo

**Additional context**

etracer65, 2020-09-03:
**Important: This is not the right place to report faulty / broken hardware. Your issue will be closed without further comment.**

Contact the manufacturer or supplier of your hardware, or check RCGroups https://rcgroups.com/forums/showthread.php?t=2464844 or our Facebook group https://www.facebook.com/groups/betaflight to see if others with the same problem have found a solution.

The fact that you have the same problem regardless of the firmware version would indicate that this is not a Betaflight problem and likely either failed or incorrectly installed/configured hardware.

closed 2020-09-03 by etracer65

majoe17, 2020-11-08:
Hello, i have the same problem. Have you send a ticket to jhemcu?

geniack, 2020-11-08:
> Hello, i have the same problem. Have you send a ticket to jhemcu?

Hello, 

Nope. I contacted banggood, they sent me a 2nd one. 2nd one Was also faulty. So they refunded me and i have gone with another flight controller. I gave These 2 faulty fc's to a friend of Mine and He managed to swap 2 components on it making it work Finally. Hope this helps.

bingbingbongus, 2021-01-07:
>  I gave These 2 faulty fc's to a friend of Mine and He managed to swap 2 components on it making it work Finally. Hope this helps.

Do you have any recollection of what components he might have switched? I've been fighting this identical issue for days :/

ppoinha, 2021-01-12:
> Hello,
>
> Nope. I contacted banggood, they sent me a 2nd one. 2nd one Was also faulty. So they refunded me and i have gone with another flight controller. I gave These 2 faulty fc's to a friend of [...]

I also have same voltage issue, any updates how to fix this for the f7 XSD ?

isoos73, 2021-03-29:
> Hello,
>
> Nope. I contacted banggood, they sent me a 2nd one. 2nd one Was also faulty. So they refunded me and i have gone with another flight controller. I gave These 2 faulty fc's to a friend of [...]

Hi! 

Could you please write to me what and where were these two components in the board to change? I think these were the resistors of voltage divider for the voltage adc. 

Thx!

master0id, 2021-10-07:
> > I gave These 2 faulty fc's to a friend of Mine and He managed to swap 2 components on it making it work Finally. Hope this helps.
> 
> Do you have any recollection of what components he might have switched? I've been fighting this identical issue for days :/

can you share with us - what exact components were changed and with which ones?

isoos73, 2021-10-07:
> Got same problem here, what's the solution finally?

Hi!

No solution. I got refound from banggood.

master0id ***@***.***> ezt írta (időpont: 2021. okt. 7., Csü
18:43):

master0id, 2021-11-20:
found a solution. You need to replace the resistor component. This one on the photo. 
It is incorrect from factory - it is 10kOm, and you can put there 35kOm one. Then in betaflight battery settings set values as on the second screen. And everything will work smooth )

[image]

BF settings: 
[image]