betaflight/betaflight #10165: When can we unified H743 target?

DusKing1, 2020-09-06:
**Is your feature request related to a problem? Please describe.**
I'm making some H743 boards but due to board size matter and MCU package matter, the schematics are not all the same. So I might need to add 4 different targets but generally speaking, they are mostly the same, this would cause heavier load to the build system.

**Describe the solution you'd like**
If it is possible I wish the H743 target can be unified so that I can use one firmware for 4 (or even more) different FCs

**Additional context**
If devs need more samples to test out DM me in slack.

melonmarlon, 2020-09-07:
This would help me, as I'm developing flightcontrollers with the STM32H743 too.
+1 From me

mikeller, 2020-10-12:
@DusKing1:

Yes, there definitely is interest in creating a Unified Target for H743 if there is interest in releasing more boards with it.

haslinghuis, 2022-05-08:
Fixed in https://github.com/betaflight/betaflight/pull/10791

closed 2022-05-08 by haslinghuis