betaflight/betaflight #10167: New fly mode: maintain pitch during yaw/roll changes

Quitesimpletoo, 2020-09-07:
In ACRO, while turning using a roll and yaw input, the quad-copter will pitch down automatically (as the yaw rotation puts the nose 'down' into the turn). You leave the turn with pitched down more than you entered the turn. As a consequence one of the most counter intuitive things people have to overcome is to learn to pitch back while turning to avoid crashing into the concrete.  

Proposal is a flight mode that will always maintain pitch regardless of roll/yaw inputs => a PID-loop to counteract any pitch changes unless there is a pilot pitch stick input. So not a auto-leveling mode like ANGLE or HORIZON where pitch gets reduced without stick input but rather a pitch HOLD until changed by the pilot and counteracted by the PID loop when external changes or Y/R changes create an error vs pitch setpoint. 

Usefulness?   
- for cinematic users you could focus on an object in a turn and hover wide around it (orbit style) without correcting pitch (which would create wobbles and height shifts - you basically hand over those tiny corrections to a fast PID loop than have it done by the pilot) 
- no crashes by losing altitude in a turn by novice acro pilots. It gives new pilots a better view on a turn as it stabilizes one rotational axis for them.

CONS
- Experienced ACRO users can still do whatever they want but when pulling down the pitch stick after a turn will now result in pitching up during/after the turn, slowing down. 
- another source for noise?
- F4/F7 still able to digest yet another loop?

naming proposal: HOVER

haslinghuis, 2020-09-09:
Looks like a variant of #9318

Quitesimpletoo, 2020-09-09:
yes indeed - it is very similar but there's a major difference: in #9318 uses a pre-flight setpoint for camera angle. In this HOVER proposal in ACRO mode we would only maintain, hold the pitch once pitch stick input is zero/low. In essence this is much easier as we only have one single known error to correct. I agree with simonbethke in #9318 commented on Dec 30, 2019 'How can you ever know the offset between the viewing vector and the motion vector? You never know the motion vector at all...'  -- which makes #9318 a lot more complicated and probably is mixing up acro with self-leveling modes (and needs accelerometer) 

In this proposal however the new pitch setpoint is a known variable and produces a predictable errror due to yaw/roll (only using the gyro in ACRO or even without using a static calculation). We don't use the accelerometer or gyro to level out. The easiest way to understand is actual when airspeed is zero and you roll & yaw, there is always a known degree of pitch from that combined stick input (so no need to 'measure' pitch with sensors - you already know ); combine this error with a rate or strength and you're done. If of course the gyro could be used it would be more universal and tunable but more complicated (idk if the current throttle up pitch correction can also be omitted then). Again we don't need to correct against a deviation from a flight path (leveling out to horizon) - only correct against a pitch change during zero/low pitch input.

Also reading into #9318, I totally forgot about upside down flight, but guess gravity and thrust prevents real flat corners upside down anyway.

Quitesimpletoo, 2020-09-23:
Closing topic

closed 2020-09-23 by Quitesimpletoo