betaflight/betaflight #10168: Unified UBLOX gps configuration

DickyMcCockpants, 2020-09-07:
The UBLOX gps modules are having various performance issues in betaflight due to incomplete configuration of satellite channels, gps ephemeris, dynamics mode (air mode<1g, 2g, 4g) and so on. Is there a road-map for such implementation (incl. gui in the configurator) or will users be expected to disable auto-config and use the UBLOX u-center to configure?

DickyMcCockpants, 2020-09-07:
I almost forgot, with an update frequency of 4hz (250ms) the BN-220 can receive 3 networks simultaneously. (GPS, GLONASS, Galileo).

mikeller, 2020-09-07:
@DickyMcCockpants: Yes, improving GPS support is on the roadmap, under 'as soon as somebody steps up to put in the work'. Feel free to contribute.

haslinghuis, 2020-09-16:
🚧 @DickyMcCockpants Lets fix this?

kap-yra, 2020-10-12:
Есть проблема с отображением и калибровкой компаса Gps-модули UBLOX, есть какое то решение? полётный контроллер KAKUTE F7

Bwsbmw, 2020-10-30:
> 
> 
> @DickyMcCockpants: Yes, improving GPS support is on the roadmap, under 'as soon as somebody steps up to put in the work'. Feel free to contribute.

GPS has been working on every quad i have put it on. I only run a Beitian BN-180, and BN-220, baud 57600, Ublox, Auto Config, ground assist NONE. I live in the Northeast USA. I have not had a problem with any of the 6 quads i have with GPS. All acquire full GPS lock, and over 7+ satellites within 45-60seconds on a cold start. Never longer than 2min for a Lock (atleast 7) on a horrible day.. (dont turn the GoPro on till you have GPS lock) its the one thing that i have noticed that will affect acquisition time, but once locked the GoPro doesn't seam to have any effect.. Ill get up to 13-17 satellites by the end of the first battery depending on the conditions. I have had horrible trouble trying NMEA protocol with other GPS modules in the past. I have found the best is the BN-180, and 220 with 18-A BT on the Chip, it will work perfectly with the settings above. GPS modules with the SR03 on the chip don't even work in U-Center. They acquire maybe 2 or 3 then drop, and start over again, almost as if they aren't tuned. One has not a single tuning mark, and it wont even connect via u-center outside in a field on a perfect day. All my GPS issues have been rectified on every build by using the above settings. The best thing is to check your GPS prior to installing using a FTDI adapter, and U-Center if you can. This will really let you know how the GPS is functioning, and if it is working correctly. If you are using the BN220 with Flash memory, you can Enable Autonomous Now Offline, which will help with faster locking times. I hope this can help someone. Happy flying.

kap-yra, 2020-10-30:
Спасибо дорогой друг!

-- Юра

closed 2021-02-08 by github-actions[bot]