betaflight/betaflight #10169: @pam4fh just use Betaflight Configurator 10.7.0 and enable FrskyOSD in PORTS tab

DubLD, 2020-09-08:
Yeah my brother also has the MatekF722PX F.C. also. But we had his flashed before this whole thing started. We also had my exact same board that I had ordered after his, and as a consequence, arriving after his. But little did I know then that the second consequence was waiting for me that I had nothing to do with. 
But low, and behold! I flashed the exact same version from the exact same version configurator, I don't have that option either. There is no selecting the FrSkyOSD in the ports tab for me. But the exact same thing on my brother's, does have the option. So I believe you are only telling a half-truth.
And, also(To be Clear), not saying that you're wrong (because you were mostly right before current issue developing). 
Just wanted to make you aware of the stipulation that may fall into play.

@pam4fh just use Betaflight Configurator 10.7.0 and enable FrskyOSD in PORTS tab

_Originally posted by @Asizon in https://github.com/betaflight/betaflight/pull/9808#issuecomment-678117594_

Asizon, 2020-09-08:
This is not the correct place to seek support for these kinds of problems. This is the place to report bugs with the firmware. Please try an alternate source like:

Facebook Group: https://www.facebook.com/groups/betaflightgroup/
Betaflight Slack: http://slack.betaflight.com
RCGroups Thread: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread/

DubLD, 2020-09-08:
I wasn't seeking anything. It was informing. If it WAS anything at all. 

https://github.com/betaflight/betaflight/issues/10169#issuecomment-688694824

Asizon, 2020-09-08:
But it is clearly user configuration problem.

Asizon, 2020-09-08:
Just make a fresh firmware flash with full chip erase, and in configurator part, delete manually ProgramFiles Betaflight-Configurator folder and reinstall again

DubLD, 2020-09-08:
Don't know how the hell that could because I damn sure didn't figure anything fuking down wrong I mean I wasn't given the option so there wasn't the wrong choice to make. It was just a fucking story dude.

https://github.com/betaflight/betaflight/issues/10169#issuecomment-688750924

DubLD, 2020-09-08:
Yeah I'll get right on it.
Even though that's not what I'm doing with that flight controller. 

https://github.com/betaflight/betaflight/issues/10169#issuecomment-688752845

closed 2020-09-08 by etracer65