betaflight/betaflight #10179: DJI FPV OSD show 0

TenchiMasaki, 2020-09-13:
**Describe the bug**
I get 0V and 0A on OSD using DJI FPV using DJI Air Unit when upgraded from 4.1.2 to 4.2.x and 4.1.7 BETAFLIGHTF4 target (Also noticed with other targets as well)

**To Reproduce**
Steps to reproduce the behavior:
Upgrade BETAFLIGHTF4 target to 4.1.7 or 4.2.x, enable MSP on UART6, Serial RX on UART2, change battery Voltage Meter and Current Meter to Onboard ADC, add Battery Voltage and Battery current draw and view OSD in DJI FPV goggles.

**Expected behavior**
OSD elements should show correct values and not 0 in DJI FPV goggles.

**Flight controller configuration**
```
# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.2 Aug 16 2020 / 01:46:41 (e833ac612) MSP API: 1.43
# config: manufacturer_id: FPVM, board_name: BETAFLIGHTF4, version: f92ddc38, date: 2020-04-28T07:37:21Z

# start the command batch
batch start

board_name BETAFLIGHTF4
manufacturer_id FPVM

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# serial
serial 1 64 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

# master
set acc_calibration = -97,-61,-304,1
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set battery_meter = ADC
set osd_vbat_pos = 2059
set osd_current_pos = 2090
set osd_avg_cell_voltage_pos = 43

profile 0

rateprofile 0

# end the command batch
batch end
```

```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: PULLDOWN 1
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: MOTOR 4
C09: MOTOR 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: INVERTER 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FR
```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
- iFlight SucceX-E mini F4 bought from Rotorvillage
 - DJI Air Unit Firmware v01.00.06.00
 - DJI Air Unit Wired to UART6 and nR2

**Additional context**
N/A

mikeller, 2020-10-11:
Interesting - this works for me with Betaflight 4.2 and the DJI firmware v01.00.06.00. Do you get the correct voltage displayed when you connect the flight controller to Betaflight configurator with a battery connected (and PROPS REMOVED!)?

TenchiMasaki, 2020-10-11:
I guess it must have been a bad solder of UART connection, seems okay now.

closed 2021-02-08 by github-actions[bot]