betaflight/betaflight #10180: Change RC rate type in OSD to match rate type in use

ctzsnooze, 2020-09-14:
Hi

A person who used 'Actual Rates` instead of `Betaflight Rates` noticed that the labels for roll, for example, remained `RC R RATE`, `ROLL SUPER` and `RC R EXPO`.  And the rate values were 0.02 not 20, for example.

For actual rates, perhaps these should be `ROLL CENTER`, `ROLL MAX`, `ROLL EXPO`, etc.

To be 'perfect', the value for the rates (centre and max) would need to be multiplied by 1000 to match what the user sees in configurator when Actual Rates were selected.

One solution is to make Actual Rates the default in 4.3, and change all the scaling and labels to suit, effectively deprecating Betaflight Rates.  The simple new labels 

The alternative is to somehow (might be difficult) to re-code the OSD to be responsive to the rates type.

Actual Rates has more flexibility in creating rates curves (greater range of curve options), simpler numbers to set and remember, and is easier to adjust, especially via the OSD,  because each parameter affects only one single aspect of the rates outcome. 

My thinking is that if we changed to Actual Rates by default, in 4.3, all the other rates types will become 'history', for niche/historical users only.  

I'm putting this up for discussion since it involves quite a significant change.

dronotron, 2020-09-14:
Exactly. Since it's not extremely difficult to figure out they show as 1/1000 of the values in the configurator. It would be nice to have them the same.
I also agree that actual rates will be the future since the values make more sense (degrees per second) and of course, they're easy to remember

etracer65, 2020-09-14:
The choice of the default rates type should be driven by the user preferences which can be reported from the analytics gathered by the Configurator. I suspect that Betaflight rates are used by an overwhelming majority.

atomgomba, 2020-09-14:
I wonder how many users are there who never touch the rates just leave them on default. For these people it doesn't matter which one is the default rate type, but should changing them cross their minds then how to make changes to Actual Rates might be easier to understand.

wuzle, 2020-09-14:
ActualRates are infinitely easier to finetune to preferences. So personally I am all for the transition to ARs to become the default. It's also easier this way to rebuild any previously liked rate curve. Like when you want to transfer velocidrone rates (I think it's based on 3.1.x or something and still has setpoint weight and transition in there) to the actual flying quad, it's easier to model the wanted curve with ARs where you actually have control about the different zones of the curve.

Though I can see problems arising when it will just land as default in Betaflight and loads of uninformed people just hack in their usual numbers and start ruining their quads despite a warning text in the release notes. ^^

McGiverGim, 2020-09-15:
> The choice of the default rates type should be driven by the user preferences which can be reported from the analytics gathered by the Configurator. I suspect that Betaflight rates are used by an overwhelming majority.

This is not the only reason to use. Is a point, but not the only one. 

The major part of users have their rates from older versions, and use the same system. For them, is not a problem to choose Betaflight rates and configure them with the values from older versions.

I think we must decide which rates we want to promote, and Actual rates are far easier to understand for newbies pilots, and they don't seem to have any drawback against Betaflight rates. We have made an effort in the latest versions to make Betaflight more accessible and this is one thing that helps.

So I think we must decide if Actual is better than Betaflight rates and if this is yes, use them as default. Have we received feedback from users trying to use Actual and abandoning it because it was a bad rate system? Or only because they usually use Betaflight and don't want to change?

IllusionFpv, 2020-09-15:
This is a know issue discussed with @mikeller too, I've done some fix to use osd but the current labels have remained unchanged.
About deprecating other rate systems I'm not shure because there are other new systems that aren't for "niche/historical users only" so we need to adapt the code to support all (almost the new ones) systems.

ctzsnooze, 2020-09-18:
I wasn't suggesting that we remove or deprecate any of the other rates systems, we should continue to support them all.

mikeller, 2020-09-21:
Interesting data here (last 30 days):

[image]

These are only events where a user explicitly selected a rate in the UI and then saved the configuration, so it does not take into account cases where users restore rates from a backup in CLI So it does look like an overwhelming majority of users still prefer Betaflight rates. I wonder if this is due to the fact that the overwhelming majority of setup guides / videos use Betaflight rates?

atomgomba, 2020-09-21:
It'd be interesting to sample a couple of months to potentially see some trends because new rate systems are, well, new. If I understand correctly this table also doesn't show users who installed Betaflight and never have touched the rates. How does the selection of Betaflight rate system compare to the total number of installs?

mikeller, 2020-09-21:
Ratio over time seems to be fairly constant:

[image]

And in the last 30 days we had 696448 sessions, vs. 60706 sessions where the user changed the rates type.

IllusionFpv, 2020-09-21:
For me the best solution is to leave betaflight rates as default and add the code in OSD for supporting the new rate systems.

haslinghuis, 2020-09-22:
If `actual rates` is a better default and not changing behavior when untouched please consider making `actual rates` default because I think the majority of the userbase does not even consider changing the rates anyway.

mikeller, 2020-09-22:
@haslinghuis: I am torn on this. As the stats show, then user counts are:

does not change rates > uses betaflight rates > uses actual rates > uses other rates

While I totally agree that actual rates are a better way to adjust rates, and the way forward, changing what the default is will hardly change whatever system any given user uses to change their rates - this requires reading up on how the new chosen system is works and what action is required to get the desired effect. So all that changing the rates type default (alone) achieves is to force more users to take one extra step than how it is now.

If anything, this should be accompanied by documentation changes to explain how actual rates work and why they are better.

McGiverGim, 2020-09-22:
I don't think changing defaults is as problematic.
- The users that is not the first time that they play with rates, know what rates they want. So they can easily change to Betaflight rates and adjust the rates as they are used to. In a future maybe they move to Actual, but is not a problem to remain with Betaflight rates.
- The new users don't know nothing about rates, and if they read about Betaflight rates, they will not understand nothing. Few users understand the values that they change. They usually get a curve moving the values and seeing the curve that the Configurator draws until it gets a form that they think is what they want. For this users I think is better to show them Actual directly, at least they can enter the max degrees directly and I'm sure they can understand better the value introduced in the new system, so we are giving new users a more friendly system..

So I continue thinking we must show Actual as default, because we think is better and easier. For experimented users it will not be a problem to change to Betaflight rates in any way or recover them from a diff, so I don't see any drawback.

Another think is that if we think that Actual is not a better system, for X or Y reasons. If someone has some reasons then we don't must change until discussed.

etracer65, 2020-09-22:
If the majority of users don't change the rates and simply fly with the defaults, then how is it an advantage to switch the default to something that's supposedly "easier to understand" when they have no need to understand them? It is however a disadvantage in that all these users that don't understand or change their rates will experience an unexpected and unwelcome change in the flight feel when they next upgrade and are unaware that the rates (which they didn't change) are now defaulting to a different curve.

Also I feel that the "easier to understand" or "easier to use" arguments are flawed. Few people other than more advanced users or people constantly analyzing blackbox logs actually care that it's easier to set a specific "actual rate". A normal user isn't looking to set the rate to exactly a specific max rate. Beginners generally start with defaults and relatively increase or decrease them based on preference and improving skills. The actual "number" that the rates are is really irrelevant to most users. People don't tune their rates to a "number", they tune them to feel.

McGiverGim, 2020-09-22:
If a user let the default curve, without any modification, is because it is a newbie user. They will not get a different flight feel because we can give a new default one almost exactly to the first one. At least enough exactly for a newbie user. I don't think a lot of advanced users fly with defaults.

> Also I feel that the "easier to understand" or "easier to use" arguments are flawed

I don't know why. All of us have started with defaults, and changing them we all have reached to our "curve", I don't say Betaflight rates don't work, only that we can make easier to get to the desired curve.

An user maybe wants to:
- Make the quad faster/slower in flips and rolls.
- Make the quad more/less responsive in the center.
- Make the change between center and final stick more/less agressive.

Now the user must relationate what he wants with what value he must change:
- RC Rate, Rate and RC Expo.
- Center sensitivity, Max Rate and Expo.

What do you think will scare less the user and make him easier to get to what he wants? 

The Betaflight rates are usable because we have a curve in the Configurator that lets the user see the changes he is doing. And maybe the user don't know what is the max degrees he wants exactly when is newbie, but he will know, more or less, how fast is 360deg/sec or 1500deg/sec, but he will not know how fast is 0.6 or 1.5 Rate.

So is clear that Actual is easier and is better for newbie users. Another think is that this is not the only reason maybe to make them defaults.

@mikeller have give a good reason, the major part of users (that are not newbies) will want to continue with Betaflight rates to not play again with rates, and they will need to change one combo more.

If there is no more reasons that we can find, we can discuss if is better to make things easier for newbies or less clicks in the Configurator to old users. I prefer to make things easier, but is more a personal preference.

IllusionFpv, 2020-09-22:
@McGiverGim for example actual rates have less resolution than quick rates(there are more similar to default bf rates), and there are more similar to default bf rates, why not set quickrates by default?

McGiverGim, 2020-09-22:
I suppose you are talking about some part of the curve. We have the same number of input and output points with both systems. I don't notice the lost of resolution, so I prefer easy concept over resolution, but again it's only me.
To me quick rates is better than Betaflight rates because simplifies it, but actual simplifies it more. Only my preference.
If there is a real difference in resolution that people feel, is a reason to add to the list.

IllusionFpv, 2020-09-22:
@McGiverGim resolution of settings, for example rcrate cam be increased by 10 points

mikeller, 2020-09-22:
@IllusionFpv:

Looking at https://github.com/betaflight/betaflight/issues/10180#issuecomment-696084554, about 10 times more users seem to be wanting to switch from the current default of Betaflight rates to Actual rates than are wanting to switch to Quick rates - this suggests that Actual rates are a lot more popular with users than Quick rates.

dustinlacewell, 2024-05-31:
Wait, should this be closed? Seems things got derailed by a discussion over defaults. Can we please just at least support Actual Rates in the OSD, it's an issue that we have today that I bump into nearly every time I fly.

ctzsnooze, 2024-05-31:
Thanks @haslinghuis - we should make a PR for this, I completely forgot about it!

brassballl, 2026-08-18:
I just got into fpv, and all sources online suggest using actual rates, and pretty soon in the sim I developed a feel for what max, expo, and center change on the behavior. Recently in the field I tried to tune my rates a bit, and saw its betaflight type with names like rc and whatnot. So I went ahead, and converted my desired actual rates to betaflight rates, and filled them in the osd, needless to mention it flew like shit.
Would it be hard to query betaflight for rate type and display the labels accordingly? I am surprised to see now that its the actual values scaled down and mislabeled there, I wouldnt have expected that.