betaflight/betaflight #10182: STM32L4+

Hphipps, 2020-09-15:
Is it possible to compile Betaflight for a STM32l4+ microcontroller?
[https://www.st.com/content/st_com/en/products/microcontrollers-microprocessors/stm32-32-bit-arm-cortex-mcus/stm32-ultra-low-power-mcus/stm32l4-plus-series.html](url)

haslinghuis, 2020-09-15:
❓ Afaik no. STM32G4 is being added to the codebase right now. Maybe @jflyper has some insight?

haslinghuis, 2020-09-15:
❓ Not sure. STM32G4 is being added to the codebase right now. Maybe @jflyper has some insight? I have seen some reports about using Betaflight with STM32L476.

💡 STM32L4 is pin-to-pin compatible with the different STM32 series and opens the door to the full STM32 ecosystem.

mikeller, 2020-09-22:
@Hphipps: No, this is not possible out of the box, and will require some work to create at least linker scripts to work on these MCUs. Not sure what the advantage you see in using STM32L4+ MCUs - the MCU power consumption is normally insignificant in our application. And the lower clock rate compared to F4 will be a disadvantage.

closed 2020-12-25 by stale[bot]