betaflight/betaflight #10186: BF 4.2.2 yaw spin to the moon - with crash_recovery ON

emmetscopters, 2020-09-17:
I encountered a tiny problem with BF 4.2.2. today - yaw spin to the moon, when messing with crash_recovery settings.

Here is what happened:
1. I left yaw_spin settings on default
2. I turned crash recovery ON - left crash_limit_yaw on default - 200
3. I hit the gate at nearly full speed and got a massive yaw spin. Never  saw THAT MASSIVE yaw spin to the moon. I disarmed at ~40meters and fell to the ground.

A couple of months ago somebody said that yaw spin recovery has priority over crash_recovery settings. But it didn't look like that today. Never saw anything like this and  the way I see it is that the crash triggered crash recovery and since yaw recovery was on default 200 (limit of how much reverse crash recovery will apply to stop yaw spinning after crash) it was getting worse and worse since  crash recovery was all around roll and pitch too.
 
Any ideas what could go wrong ? Is it some kind of bug in BF 4.2.2 ? Yaw  spin recovery should always have a priority over crash recovery - its  dangerous the other way... Apparently yaw spin recovery didn't triger  and 200 on crash_limit_yaw wasn't enough to deal with this spin. Is it possible that new AUTO mode on yaw_spin_recovery doesn't have priority over crash recovery ?

The quad is trusted, well tuned and known setup, raced a lot previously. The problem never happened prior to 4.2.2 with crash_recovery ON.
FC - FURYF4OSD

etracer65, 2020-09-17:
Please provide the information requested when you opened the issue.

etracer65, 2020-09-17:
Firstly it's important to note that yaw spin recovery simply tries to maximize the PID controller's ability to address yaw by disabling roll/pitch control. There is still no guarantee that the spin can be recovered. For example if a prop was damaged or motor de-synced then no matter what the PID controller does it may not be able to recover from the physical problem.

Secondly setting `crash_time = 150` is likely a bad idea. Only allowing 150ms for recovery before the feature deactivates is not good.

Lastly I wouldn't recommend using crash recovery without lots of testing and experimentation. It's always been an experimental feature that's suffered from insufficient testing little to no development after its initial implementation. It's my understanding it was originally developed and tested in the context of micro quads flying indoors bouncing off walls.

emmetscopters, 2020-09-18:
The main thing I wanted to be sure of is If we got 100% certainty that yaw_spin recovery got priority over crash recovery now. If not - that explains this incident since 200 crash_yaw wasnt enough to do anything.

All in all If you do not rcommend to use it I might resign from using it. But I got very good results with crash recovery so that would be pitty.

Thanks for Your work and responses.

etracer65, 2020-09-18:
I don't really understand your question. Yaw spin recovery and crash recovery are two separate and unrelated things. Yaw spin recovery will disable the roll/pitch PIDs when triggered so I guess you can say it "takes priority". But yaw spin recovery will only be active when the yaw rate is above the trigger threshold which looks like about 715 deg/sec in your case. Below that it does nothing. Additionally if it is triggered, it will disable once the yaw rate is 100 deg/sec below the trigger.

The way crash recovery works leads to a strong possibility of a resonant type of runaway in cases where the dominant aspect is a yaw spin, but with a substantial "wobble" in the axis of rotation. It's not optimized to handle this and in theory could lead to a self-sustaining aspect. There's no way to know if that's what happened in your case without a blackbox log.

Crash recovery needs a significant rewrite to be more effective but it's not the kind of thing people want to work on since it involves lots of crash testing and nobody wants to do that.

butterSnake, 2020-09-19:
If you cut your throttle as soon as you enter the yaw spin it will recover nearly instantaneously. If you hold the throttle open it makes it a lot harder

emmetscopters, 2020-09-19:
OK. Thanks for making it clear. I understand the problem now.

closed 2020-12-25 by stale[bot]