betaflight/betaflight #10189: Motors gradually increase to full throttle with QX7

Ashtrans, 2020-09-17:
Hello Guys,

I upgraded to BetaFlight 4.2.2 (did not use previous versions) and after configuration, everything seems fins. When motors are tested by moving the Slider, all seems perfect (not taking into account the pid).

However when the drone is connected directly to qx7 and when throttle is used, the motors go crazy from 1050 to 2000 from just 2 notch throttle!!

What may be the cause of this and how to fix?! TY

etracer65, 2020-09-17:
It's normal, read this: https://drones.stackexchange.com/questions/127/why-do-the-motors-on-my-drone-keep-spinning-up-and-max-out-when-i-run-up-the-thr

**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

closed 2020-09-17 by etracer65

Ashtrans, 2020-09-17:
Hello Bruce,

Thank you for the email.

I forgot to mention that this occurs with props on! If I dont use the kill switch, it will keep going higher. Two notches on the throttle raise the motors to 1800!!

I already viewed all the video on YouTube along with Joshua's video and cannot figure it out...i also uploaded the tuning from the manufacturer, HGLRC....

Please assist if possible
Best

Closed #10189<https://github.com/betaflight/betaflight/issues/10189>.