betaflight/betaflight #10194: No, dont exist. I think its actually not possible with BF

4noxx, 2020-09-19:
No, dont exist. I think its actually not possible with BF

https://youtu.be/Vd6EBr4QYa8
https://www.youtube.com/watch?v=fmOxB7QSLUk

_Originally posted by @4noxx in https://github.com/betaflight/betaflight/issues/10193#issuecomment-695228468_

sorry, the other post was closed too quickly

i think its not possible with BF to change VTX Power in flight. i read the BF CLI COMMANDs, but found nothing. things like "set vtx_low_power_disarm" are not
that what i mean.

here are a list of all my VTX command in my settings....

SET_VTX_POWER_LEVEL is only i inav command. i didnt found a command like this for betaflight. or i look wrong..... then sorry.

#master
set osd_vtx_channel_pos = 2511
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 1
set vtx_low_power_disarm = OFF
set vtx_freq = 5740
set vtx_pit_mode_freq = 0
set vtx_halfduplex = ON

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 400
vtxtable powerlabels 25 100 200 400

# vtx
vtx 0 0 0 0 0 900 900
vtx 1 0 0 0 0 900 900
vtx 2 0 0 0 0 900 900
vtx 3 0 0 0 0 900 900
vtx 4 0 0 0 0 900 900
vtx 5 0 0 0 0 900 900
vtx 6 0 0 0 0 900 900
vtx 7 0 0 0 0 900 900
vtx 8 0 0 0 0 900 900
vtx 9 0 0 0 0 900 900

dkustec, 2020-09-19:
This 
```
vtx 0 2 0 0 1 900 1200
vtx 1 2 0 0 2 1300 1700
vtx 2 2 0 0 3 1800 2100
```
will enable a 3-position switch on Aux3 to switch power levels 1/2/3.

etracer65, 2020-09-19:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

closed 2020-09-19 by etracer65