betaflight/betaflight #10197: Add option to display the OSD time & date on arming

molor, 2020-09-20:
**Is your feature request related to a problem? Please describe.**
No, it's just a idea for BF improvement. I use DVR all the time, and sometimes when I watch old recordings of my flights, I get nostalgic and want to know exactly when it was. As far as I understand, there is no such possibility now :c

**Describe the solution you'd like**
Display current time & date received from GPS, if possible, below "Armed" message.
If the time is unknown then display "Armed" message only.

**Additional context**
The ability to customize the date & time display format will be an additional advantage.

etracer65, 2020-09-20:
The RTC can already be displayed on the post-flight OSD stats screen.

molor, 2020-09-23:
> The RTC can already be displayed on the post-flight OSD stats screen.

Sounds like a alternative. But I think that my idea is more universal & convenient..

etracer65, 2020-09-23:
> > The RTC can already be displayed on the post-flight OSD stats screen.
> 
> Sounds like a alternative. But I think that my idea is more universal & convenient..

You missed my point. The RTC is **already** displayed on the post-flight OSD stats. No changes are needed.

[image]

closed 2020-09-23 by molor

jonathanoeijoeng, 2023-12-20:
How we can set date and time in betaflight so rtc will show correct date and time?
Btw, I did check rtc but still not showing date time in google. I have connected to remote as well.

> > > The RTC can already be displayed on the post-flight OSD stats screen.
> > 
> > 
> > Sounds like a alternative. But I think that my idea is more universal & convenient..
> 
> You missed my point. The RTC is **already** displayed on the post-flight OSD stats. No changes are needed.
> 
> [image]

SteveCEvans, 2023-12-20:
To ensure your RTC is set correctly (which also ensures your blackbox logs are timestamped correctly) see https://github.com/betaflight/betaflight-tx-lua-scripts#background-script

jonathanoeijoeng, 2023-12-20:
Thanks so much for your quick response. I have downloaded the script and it worked. Date and time are showed in the OSD but my vtx table suddenly changed? it should be only downloading and not changing vtx on start up, right?