betaflight/betaflight #10203: Add universal support for OGN tracker

Quitesimpletoo, 2020-09-22:
Future of UAV will have some sort of tracking need anyway (either voluntary or by law); I recently was only aware of paragliders in the area when I touched down a xxx km/h drone after a max speed test... So that's were I come from writing this feature request.

- TBS is going for FLARM messaging, which is not open source, DJI has yet another solution (I guess). Usually all nicely protected technologies and quiet expensive too...

- Betaflight could 'easily' implement an OGN (open glider network) tracker. OGN has an open architecture (that also includes FLARM and others) - see http://wiki.glidernet.org/dev

 * STM32F7xx (probably)
 * a GPS is connected to the FC and has a 3D fix
 * FC has SPI interface and to this SPI interface is a connection to a sx1276 (board) RF transceiver
So for any long range drone/fixed wing the only additional hardware is probably this few grams transceiver 

The FC compiles a text message including ID, GPS location, speed etc. based on the OGN message structure and broadcasts this by sending this text message every 5 seconds to sx1276 transceiver over SPI, every 1 second if stick input ROC > treshold value%.

OGN already has a github page with all its software available to create the message, including some DIY trackers
A first implementation can be restricted to sending messages only (to become visible to whatever traffic is already in the air) 
--> these messages can readily be verified through the existing tracking website - https://live.glidernet.org/
>> collision warnings can be implemented later (OSD warning; add to fail save options to cut power at < 5 meter clearance?), but this is much more MCU intensive.

**Describe alternatives you've considered**
- TBS crossfire with FLARM license

mikeller, 2020-09-22:
@Quitesimpletoo: 

> The FC compiles a text message including ID, GPS location, speed etc. based on the OGN message structure and broadcasts this by sending this text message every 5 seconds to sx1276 transceiver over SPI, every 1 second if stick input ROC > treshold value%.

Is there any reason that this should be tied into the flight controller? This could be done by a completely independent piece of hardware (aka a radio tracker).

Quitesimpletoo, 2020-09-22:
Adding a dedicated tracker adds a lot of weight to a drone. This solution only needs a lightweight sx1276 module (worst case with an UART/SPI bridge). Also this solution is incredibly cheap vs anything commercially available (a few $ vs a few hundreds) and still meets the goal of making sure you are visible to others. OGN already has a widespread use and the network is compatible with FLARM that TBS is using. We need something that is easy to implement by any drone builder that is so low threshold it actually makes less sense _not_ to use it. All the infrastructure (ground stations, servers etc) already exist so we just need to gather GPS data and send a message through SPI (might not have exposed solder points on an fc today) or I2C (usually has solder pads) or even UART (followed by SPI bridge) to an sx1276. Whatever serial protocol we choose, it is a very limited hardware scope to implement.

I can imagine that future FC builders might just want to incorporate a sx1276 on the SPI bus if this is available. Yes it's going to take time to process on the F7, but it's not a time-critical task (on a MCU scale) and it is only needed on 1-second basis worst case.

etracer65, 2020-09-22:
How does a dedicated tracker add "a lot of weight"? All the same components are needed with the possible exception of a small micro controller that will weight a fraction of a gram.

And without a dedicated all-in-one solution how would this work when only a small fraction of Betaflight users actually have GPS on their quads?

Quitesimpletoo, 2020-09-22:
A blue pill STM32 board, a separate GPS, BEC, sx1276 transmitter board and antenna will indeed enable you to build an OGN tracker (see their dev site). All these separate items take space (for some users the weight probably matters less than the space they occupy). Not counting even a box to put this in. (Still it's a lot lighter than ADS-B units)

An FC connected tracker will indeed only work/be useful for those people already adding a gps to their craft (drone, fixed wing, hexacopter...) for either speed measurement, fail save, long range retrieval when crashed or just because they can .

Having a GPS connected to your FC, now only adding the sx1276 module and a small antenna will still have your gps in your BF OSD, have your return to home location work while now just adding OGN tracking capability. 
You will also know where your drone is if the battery is still attached, even if your vtx failed (no osd!) and had you crash in galaxy far far away...
The FC's onboard barometer data can also be enclosed/used in the message. 
A separate unit has no access to OSD, barometer, stick input, accel/gyro data ... It would mean you add 2 gps modules (one in the tracker and one for BF fail saves/OSD).

>> it starts with a proposal today to send out a message where we are and where we are going to be; eventually you could also process other people's messages and warn for a future collision (and propose an exit path) - in that case FC can provide more data (on attitude).

etracer65, 2020-09-22:
I think you're misunderstanding our target user base. Only a small percentage of users have GPS units installed because Betaflight has limited support or utility for GPS. Secondly again only a relatively small fraction of flight controllers come with an onboard barometer. And again Betaflight has limited use for it and only uses it to display relative altitude so it's not a priority for manufacturers to put them on their boards.

Quitesimpletoo, 2020-09-22:
Hard to argue on target groups: only few people use a gps, only few people fly long range, only few people need angle mode (perhaps nobody) etc.
I agree there is other software that is more oriented towards gps flight planning and it makes less sense on using a gps on a race quad. Even if it is displayed on the first bf config screen...

But the gps use in itself is not the point: the idea is now we have the gps integration already and the data available, can we please send it to a transceiver in fixed format so we have the cheapest OGN tracker ever? So I don't accidentally fly into a silent paraglider in the Swiss Alps at 60 mph?

mikeller, 2020-09-22:
@Quitesimpletoo: Do you have any more links to the actual state of OGN, and what the coverage for it is at the moment - the URL you are listing has latest updates from more than 2 years ago, suggesting that this is pretty much dead.

I concur with @etracer65 that the group of potential users for this will be relatively small - users will either want to / need to do tracking / controlling that uses certified and legally compliant hardware, or they will not bother to do so. Adding code to Betaflight to do homegrown tracking will not cover the needs of either of these groups.

If you are worried about becoming liable for causing in-air accidents involving aircraft then I would recommend you stay within the airspace that is legally cleared for remote control aircraft operation, and will be free of other aircraft. If you are looking for somebody to help you facilitate activity that is outside of the law then you are wrong here.

Quitesimpletoo, 2020-09-23:
I do keep in my airspace. I live in Europe. EU law on drones is changing on 1st of Jan 2021 (implementation was delayed by covid19 for 6 months). Today in my country I cannot even fly above 10m. In the future this will be above 100m. So that is my genuine concern: with this added freedom comes a higher potential for air traffic conflicts - unintended ones. And usually vs light craft or para gliders - silent craft in general. This was also the basis for development of Flarm and OGN. And don't get me wrong: if needed I will invest in a dedicated tracker. However betaflight offers a great option to provide a very low cost solution. Even with a small use base today there is an opportunity to proactively work on this before authorities force some device to be used on us. OGN still works today btw - I can still see gliders and helicopters in my area. 

Of course I appreciate your time assessing this proposal. I think I made my point so if you guys don't feel it's worth the time to develop, no hard feelings. I just wanted to defend my point strongly as I feel we will be forced by authorities some day to have this.

Quitesimpletoo, 2020-09-23:
https://www.easa.europa.eu/sites/default/files/dfu/EASA_AERO_GA_Roadmap_EN.pdf
page 16

fb
https://www.facebook.com/OpenGliderNetwork

mikeller, 2020-09-23:
@Quitesimpletoo:

> I do keep in my airspace. I live in Europe. EU law on drones is changing on 1st of Jan 2021 (implementation was delayed by covid19 for 6 months). Today in my country I cannot even fly above 10m. In the future this will be above 100m. 

In most jurisdictions the minimum altitude for GA outside of the controlled airspace around an airport is 150m, so if you stay within the law you should be fairly safe from any collisions even without a tracker.

And I am not saying that this should absolutely not be implemented in Betaflight - there will be use cases where this makes sense, like if the pilot already has a GPS unit connected to Betaflight for other reasons and does not want to install a second one. But I think that these will be a small minority.

> I just wanted to defend my point strongly as I feel we will be forced by authorities some day to have this.

From seeing what is happening in jurisdictions where regulation for electronic limitation of remote control aircraft has been introduced, in practically all cases the requirements were for the limiter device to be _approved_ and _resistant to modification_, and I suspect that this will be the case for pretty much all other jurisdictions where this happens. This pretty much precludes the use of a community driven open source project like Betaflight for these purposes.

closed 2020-12-25 by stale[bot]