betaflight/betaflight #10204: Buzzer disable mode (switch)

Fabiandelete, 2020-09-22:
**What?**

The possibility to disable all buzzer 'alarms' with the flip of a switch.

**Why?**

In case you are at an event and don't want to bother everyone with a buzzer. Or to disable the buzzer when it's going off due to low batt and you cannot retrieve it directly (active race track as example). And many more possible use cases.

**How?**

Best would be a mode which is basically the exact opposite of the current beeper mode. Instead of beeping it silences the beeper.

**Additional's**

I think it is best to include all warnings in the silence mode. So also the rx lost.
In this case the user can decide for himself if the buzzer should always come back on when rx is lost by assigning values in the failsafe tab. But the user can also decide to keep the set value and have it be silenced.

Activating the buzzer manually (with the current buzzer mode) should override the buzzer silence mode.

The code is too complex for my skills so I hope someone who has the skills can implement this. If a feature like this is already out there please let me know how to find and use it.

Thanks for all the great work!

Fabian

dkustec, 2020-09-23:
@Fabiandelete 
Can you please test this firmware. There is a feature you want.
Note that board init sound (the moment when a batt is connected) can't be silenced using a box (with the flip of a switch).

[betaflight_4.3.0_STM32F7X2_norevision.zip](https://github.com/betaflight/betaflight/files/5270007/betaflight_4.3.0_STM32F7X2_norevision.zip)
[betaflight_4.3.0_STM32F405_norevision.zip](https://github.com/betaflight/betaflight/files/5270010/betaflight_4.3.0_STM32F405_norevision.zip)
[betaflight_4.3.0_STM32F411_norevision.zip](https://github.com/betaflight/betaflight/files/5270011/betaflight_4.3.0_STM32F411_norevision.zip)
[betaflight_4.3.0_STM32F745_norevision.zip](https://github.com/betaflight/betaflight/files/5270013/betaflight_4.3.0_STM32F745_norevision.zip)

Follow installation instructions from here: https://youtu.be/I1uN9CN30gw

Fabiandelete, 2020-09-23:
@dkustec **AWESOME!!!!**

I just flashed it and it seems to be working fine on the bench. Will field test it tomorrow if the weather permits flying. Thanks for the good work!

Are there other new features in this firmware to be aware of?

dkustec, 2020-09-23:
@Fabiandelete , Thank you for the feedback. However, I can not guarantee that you can fly safe with this build. Anyway, for an added piece of mind, I can build this feature on the 4.2.3 base for you.

Fabiandelete, 2020-09-23:
Okey, I understand. The function works on the bench. And I honestly have no idea why it wouldn’t work in the field.
If implementing it into the base version is relatively simple that would be awesome!

dkustec, 2020-09-23:
@Fabiandelete No probs at all. It's an easy job.
So here it is. Those are hexes based on the 4.2.3.
This build is exactly the same official 4.2.3 except introducing this new feature. 

Please test it thoroughly. Take in care both situations when the beeper is muted and when it's not.  Check for the warning beeps when it's not muted, etc; you will know. And write some note here if any undesired behavior. 

Fly safe!
Regards.

[betaflight_4.2.3_STM32F7X2_norevision.zip](https://github.com/betaflight/betaflight/files/5271911/betaflight_4.2.3_STM32F7X2_norevision.zip)
[betaflight_4.2.3_STM32F405_norevision.zip](https://github.com/betaflight/betaflight/files/5271913/betaflight_4.2.3_STM32F405_norevision.zip)
[betaflight_4.2.3_STM32F411_norevision.zip](https://github.com/betaflight/betaflight/files/5271914/betaflight_4.2.3_STM32F411_norevision.zip)
[betaflight_4.2.3_STM32F745_norevision.zip](https://github.com/betaflight/betaflight/files/5271915/betaflight_4.2.3_STM32F745_norevision.zip)

Fabiandelete, 2020-09-23:
Thanks again for all the good work!
I will revert back to the base+ tomorrow and will post updates here about experiences with the muting.

Fabiandelete, 2020-09-25:
I'm getting weird behavior on the rx lost alarm. For some reason it doesn't activate the beeper. Also happening when I reverted back to the original 4.2.3 btw.

Betaflight is registering rx lost correctly (checked when connected to usb and leds come back on when ledlow was activated).

dkustec, 2020-09-26:
@Fabiandelete , Thank you for your input.
In this (current) scenario, all beeps start after the board initialization are suppressed by the 'mute switch'. We can add here one setting in the CLI (OFF by default) which can preserve state of the beeper silence. eg. `suppress_alarm_warning = ON` or something, and this will disable all beeps. `suppress_alarm_warning = OFF` would be default and this will disable muting of alarms.
Did you find any undesirable flow of this firmware modification (about this new mode only)?

Fabiandelete, 2020-09-28:
I tested in the field last weekend. Everything works perfect. I have set up failsafe in a way that it disables muting when rx is lost. Setting up like this also works perfectly fine. No negative (side)effects were observed.

dkustec, 2020-09-28:
@Fabiandelete , Thank you for your clarification. I'll made an pull request for this feature.
Regards.

haslinghuis, 2020-09-30:
Does is also work with Dshot ESC beeper?
Could you describe how to setup? After PR is accepted we could put it in the wiki.

Fabiandelete, 2020-09-30:
> Does is also work with Dshot ESC beeper?
> Could you describe how to setup? After PR is accepted we could put it in the wiki.

I don’t know if it also applies to dshot beeper. I did not (consiously) test that part. But why would you want to have it in those cases? As far as I know dshot beeper only works if you manually activate it or on rx lost (in which case you have no rx link to mute or unmute it).

Setup is simple as can be. Flash the new beta firmware (follow the links) and setup the beeper mute mode.

haslinghuis, 2020-09-30:
> Warning: running motor beeper for too long could result in overheat, even burns out your motors in some cases. It’s best to beep the motors once or twice then stop.

Thanks. Most of the times you approximately know where you crashed, so after regaining control link you could unmute it 🙃

dkustec, 2020-10-01:
> Does is also work with Dshot ESC beeper?

No, this feature is not Dshot related.

> Could you describe how to setup? After PR is accepted we could put it in the wiki.

Thank you for supporting this. 
It's a pretty simple: the 'Beeper mute' mode (if enabled) will suppress any buzzer sound **after** the board finish its initialization. After the final beep, the board will stay in silence. 
Using a manual 'Beeper switch' will activate a buzzer and override the 'Beeper mute' mode only when is active. 
All warnings will stay in silence unless the FS value is correctly set. 

Eg., if we use an AUX5 for the 'Beeper mute' mode and set its active range >1500, the beeper will stay in silence. In the FS procedure, AUX5 should be 800-1400 for deactivating the 'Beeper mute' mode.

closed 2020-10-08 by mikeller

dkustec, 2020-10-08:
@haslinghuis , This feature request is accepted. If you want to document it, there is a small change:
- we removed condition for checking is beeper activated manually (via switch). A buzzer can't be activated using this method now. But it's possible to use one switch to activate beeper and deactivate beeper mute mode. We can do that in the "Modes" tab of the Configurator. 

Eg.:
AUX5 <1500 -> 'Beeper mute' is active, 'Beeper' is inactive.
AUX5 > 1500 -> 'Beepr mute' is inactive, 'Beeper' is active.

So, all of this can be done using only one switch.

haslinghuis, 2020-10-08:
Not very impressive but I added your notes to the wiki. Updated menu, Also sorted the menu.

PR #10225 (2020-10-08): New feature 'Beeper mute'
According to #10204 (Buzzer disable mode via switch), this PR will introduce the feature "Beeper mute".
A beeper will be disabled right after the board is initialized if it's activated via dedicated switch.
This setting is overridable by the FS via dedicated channel used for triggering this mode.

The feature is thoroughly tested.

Fixes: #10204