betaflight/betaflight #10207: DJI MSP over softserial flashes on and off on BF 4.1.2 and greater (worked fine before)

Hawkers, 2020-09-23:
**Describe the bug**
I'm running the BetaFPV AIO V3 board on the BETAFPVF411 release. I'm running my GPS on the hardware UART so that leaves me to run DJI OSD over softserial. The OSD is perfect on 4.1.1 but any version after that it glitches on and off. Sometimes mah drawn shows up scrunched on the left side of the display despite the fact that I don't have that element enabled. And sometimes the elements on the top of the screen move to the bottom but they're half cut off. I tested 4.1.2-4.1.7 as well as 4.2.3. Every version exhibits the glitching except 4.1.1

**To Reproduce**
Steps to reproduce the behavior: Flash any version newer than 4.1.1 and fly for about a minute. The OSD will begin to disappear and reappear.

**Expected behavior**
It should remain on 100% of the time.

**Flight controller configuration**
```
resource SERIAL_RX 11 B10

# feature
feature GPS

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 1 2 115200 57600 0 115200
serial 30 1 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 3 1700 2100 0 0
aux 2 46 1 1700 2100 0 0
aux 3 13 2 1700 2100 0 0
aux 4 35 1 1300 1700 0 0

# master
set acc_calibration = -141,26,-351,1
set rssi_channel = 12
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set beeper_dshot_beacon_tone = 2
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_initial_alt = 40
set gps_rescue_ascend_rate = 400
set gps_rescue_allow_arming_without_fix = ON
set deadband = 3
set yaw_deadband = 3
set osd_rssi_pos = 2100
set osd_mah_drawn_pos = 149
set osd_gps_speed_pos = 2089
set osd_gps_sats_pos = 2103
set osd_home_dir_pos = 2096
set osd_home_dist_pos = 2084
set vcd_video_system = NTSC
set name = Whiffles

profile 0

# profile 0
set p_pitch = 75
set d_pitch = 47
set p_roll = 77
set d_roll = 44
set d_min_roll = 30
set d_min_pitch = 32

rateprofile 0

# rateprofile 0
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

# end the command batch
batch end
```

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FLASH_CS
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: SERIAL_TX 11
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: SERIAL_TX 11
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; 

BetaFPV AIO V3 (BETAFPVF411) from AirbladeUAV

 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable]; 

Caddx Vista, TBS M8 GPS

 - how are the different components wired up.

The GPS is wired to the 2nd hardware UART and DJI MSP is wired to the softserial "UART6". This was a decision based on the critical nature of GPS when flying long range.

**Additional context**
Add any other context about the problem here.

Everything works great on 4.1.1, but I'd like to try the newer versions of Betaflight without this issue.

etracer65, 2020-09-23:
Running the DJI MSP connection over softserial is not recommended. Use a hardware UART. If there are no hardware UARTs available, then get a more capable flight controller.

durchblicker71, 2020-09-30:
I can confirm Hawkers report on different hardware.  
On my Flywoo Explorer LR with caddx Vista the OSD works flawless since I downgraded to BF 4.1.1.
The Explorer quad has the Vista connected via softserial, the FC is a Flywoo Goku F4. 
With BF4.2 the OSD glitches, moves and sometimes disappears. 
"Tasks" or "status" in CLI shows high cpu load at these times.

So this appears to be a more general problem, I do not want to believe the solution is to get a more capable FC... ?

glenb2, 2020-10-08:
I have a Flywoo Explorer LR as well. I understand the use of softserial is not optimal but the LR is hugely popular.

jseaber, 2020-10-14:
The soft serial MSP connection can freeze on the Explorer also, meaning the OSD values become static and do not change until the flight controller is rebooted.

KnzHz, 2020-10-26:
Are we to interpret tracer's comment as "We don't care if it broke, don't use it that way?"

etracer65, 2020-10-26:
> Are we to interpret etracer's comment as "We don't care if it broke, don't use it that way?"

@KnzHz I don't care how you choose to interpret my comment. But the reality is that it was never meant to work so it's unsurprising that it doesn't and therefore there is nothing to fix. However if you think it's an important thing for you then feel free to submit improvements to make high baud rate softserial reliable and robust.

durchblicker71, 2020-10-26:
First let me thank you for replying. But to be honest, all we  asked was to
take a look why the osd over softserial works in 4.1.1 flawless and we get
max overhead of data in 4.2.?
But anyway, we are just a minority of users under your wrath...

Thanks and best regards,

Christian

mikeller, 2020-10-26:
@durchblicker71:

> all we  asked was to take a look why the osd over softserial works in 4.1.1 flawless and we get
max overhead of data in 4.2.?

If you want an answer to this, then it is probably best to open a support case with DJI - we know that their hardware queries the version of Betaflight it is connected to, and tailors the stream of MSP request it sends to the Betaflight version - so potentially they are sending more requests, or use a different timing for Betaflight 4.2 vs. Betaflight 4.1.

etracer65, 2020-10-26:
> First let me thank you for replying. But to be honest, all we asked was to take a look why the osd over softserial works in 4.1.1 flawless and we get max overhead of data in 4.2.? But anyway, we are [...]

@durchblicker71 And my response was that running the DJI MSP connection over softserial is not recommended. And by not recommended I mean it's not expected to work well or at all. If an unsupported mechanism seemed to previously work that doesn't change the fact that it wasn't intended to work. And again, nobody is saying that it can't be made to work or improved. Feel free to do so and submit the code. Likewise feel free to investigate the code to look for unrelated improvements that might have affected this unsupported use of softserial. But expecting that we consider this a bug and therefore we're somehow "required" to fix it is missing the point.

KnzHz, 2020-10-26:
Nobody is requiring anybody to fix anything. Not replying arrogantly and dismissively would be nice though. Even replying what you're replying now would be better, as that at least explains why. I simply refuse to believe you'd speak to people in real life like you do on here. Anyway, thank you for not caring about how I interpret it, appreciate it. My comment at least got the truth out, that it's not going to be looked at and that you don't consider it a bug.

jflyper, 2020-10-26:
Please note that software serial running on the same timer does not support full-duplex communication well, especially the line is busy in both directions. This is because active transmission is clobbered by receiving bits.

If an application using this configuration works, it is by pure coincidence of reception not overlapping with transmission.

So, if someone will be looking at this issue in depth, I would recommend to check if
1. software serial under question is configured to run on the same timer, and if this is the case, if
2. transmission and reception are overlapping (using an LA or by checking `transmissionErrors` in `serial_softserial.c`.

durchblicker71, 2020-10-26:
I understand a little bit more now. At no time I wanted you to feel this is
a requirementl. We are just enthusiastic users (in my case since about 2.4)
and where more or less politely asking for help. My background is medicine
and UX-Management, I do not know how to code. So I fear there will be no
solution.

Best regards,

Christian

jseaber, 2020-10-26:
See also PR #10229, part of release 4.2.4. 

Soft serial performance did deteriorate, but it seems developers are more interested in resolving the limitations of current MSP usage. This issue may be moot if MSP utilization is optimized and DJI cooperates: 
https://github.com/betaflight/betaflight/pull/10229#issuecomment-703263006

mikeller, 2020-10-26:
@jseaber:

> Soft serial performance did deteriorate, 

What is the basis for you to make this statement?
There were no changes in the firmware that would suggest that this is the case.
Please read https://github.com/betaflight/betaflight/issues/10207#issuecomment-716673421 - the case for this to work in 4.1 but not in 4.2 is very likely on DJI's side, not Betaflight's.

@durchblicker71:

> We are just enthusiastic users (in my case since about 2.4) and where more or less politely asking for help.

The help was rendered in https://github.com/betaflight/betaflight/issues/10207#issuecomment-697961281:

> Running the DJI MSP connection over softserial is not recommended. Use a hardware UART. If there are no hardware UARTs available, then get a more capable flight controller.

I think the message that this is working the way it was intended to work, and that different hardware or code improvements are needed is pretty clear - even if this is not the answer you would have liked to hear, it is the answer to your request.

Another possible avenue to improve your situation would be to change GPS to softserial, and use the hardware UART for MSP - if the GPS is configured for NMEA (i.e. inwards data only) then this should work on a softserial, if required at a speed less than 115200.

jseaber, 2020-10-26:
@mikeller Thank you, I agree with the suggestions and development strategy. I did not intend to suggest that (a lack of) changes to Betaflight firmware are responsible for this non-ideal behavior, but rather that others should read further to appreciate the work being done.  DJI needs to step up their involvement.

KnzHz, 2020-10-26:
Mikeller: Can you elaborate on how DJIs implementation might be the cause when going from 4.1 to 4.2? I get the jist of the rest of your post but I just don't understand this part? I understand it might just have worked out of dumb luck up until now but I'm confused on how their firmware would differ in behavior between the two BF versions? Is there a change in MSP version numbering or something else? Thanks!

jseaber, 2020-10-26:
@KnzHz Agreed, we may be missing information shared elsewhere. Time allowing, I'll probe via logic analyzer and compare timing with different Betaflight versions.

mikeller, 2020-10-26:
@KnzHz: As said in https://github.com/betaflight/betaflight/issues/10207#issuecomment-716673421, the DJI hardware queries the version of Betaflight (or rather the MSP API version supported by it), and then tailors the MSP requests to the MSP version supported by Betaflight - it is possible that they use more / different MSP commands (or that the MSP responses have become longer), thus causing a higher data load on the serial connection.

Hawkers, 2020-10-26:
Would the DJI firmware make a distinction between such a minor version change in Betaflight? We're talking about 4.1.1 vs 4.1.2.

mikeller, 2020-10-26:
@Hawkers: Sorry, I misread, I thought the problems were showing up between 4.1 and 4.2. For between 4.1.1 and 4.1.2 I don't know if DJI makes a distinction, since their stuff is closed source. They _should_ not, as they should be using the MSP API version, which is the same for both. On the other hand, the changes in Betaflight between these versions are also minimal, and don't contain anything that stands out as a possible culprit: https://github.com/betaflight/betaflight/releases/tag/4.1.2.

Hawkers, 2020-10-26:
> @Hawkers: Sorry, I misread, I thought the problems were showing up between 4.1 and 4.2. For between 4.1.1 and 4.1.2 I don't know if DJI makes a distinction, since their stuff is closed source. They [...]

That's what makes this so puzzling. I looked over the changes as well and nothing stands out as substantial. Yet, the OSD definitely breaks on 4.1.2 and is fine on 4.1.1. I wonder if pruning out those changes from a later version can shed more light.

durchblicker71, 2020-10-27:
I found the osd over softserial works fine up to 4.1.1

KnzHz, 2020-10-27:
I'm guessing but as you have to enable OSD for DJI for DJI to read the placement of the OSD elements, and as quite a few DJI boards actually have a OSD chip, could this cause some timing issue:

https://github.com/betaflight/betaflight/pull/9198/files

It's the only OSD related change. I'm too stupid to figure out what it does though as I haven't looked at the rest of the code.

mikeller, 2020-10-27:
@KnzHz: The fix in #9198 is only applies to the situation when an on-board MAX7456 is used, so this should not make a difference to the data exchange on the MSP UART.

KnzHz, 2020-10-27:
I know, but DJI sues the OSD to map out where to place the OSD elements it gets from MSP commands, so it still uses the OSD functionality to some extend I would assume?

jseaber, 2020-10-29:
Here's raw data captured on the soft serial pins of a Flywoo F411 via Saleae Logic @ 50MHz (https://www.saleae.com/downloads/). Each capture was started just before plugging in a Lipo to the flight controller. 

[Soft-serial-data.zip](https://github.com/betaflight/betaflight/files/5461250/Soft-serial-data.zip)

Initial data on the TX line is the same for BF 4.1.1 and 4.2.0:
[image]

Note that I the software serial pins were misconfigured after loading BF 4.1.1, so OSD was not working and there is only data on the TX line. I can try setting up 4.1.1 if anyone would find it useful, but initial data shows nothing out of the ordinary.

mikeller, 2020-10-30:
@KnzHz:

> I know, but DJI uses the OSD to map out where to place the OSD elements it gets from MSP commands, so it still uses the OSD functionality to some extend I would assume?

Not sure how this is done, I would assume it uses the MSP command to read the OSD element configuration, and then extracts the settings for the elements it is emulating.

KnzHz, 2020-10-30:
@mikeller Yeah, it's hard to know exactly what they do, but my experience is that OSD needs to be enabled for screen location. So if anything changed in the timing on calculating that or the like, that might do something. I'm not saying it's completely logical, but with so few changes between those two builds, OSD stops working and one of them is a OSD thing, it might be it. I know how to build firmware from a branch, I have no idea how to cherry pick. I assume I would fork 4.1.1 and then try to take only the other changes since 4.1.2?

jflyper, 2020-10-30:
Do bisection.

git bisect reset
git bisect start
git checkout 4.1.1
Build and make sure it works
git bisect good
git checkout 4.1.2
Build and make sure it fails
git bisect bad

Then git will automatically bisects commits between the two versions, checks out a new bisecting commit.
You will build and test it, and tell git if the commit was good or bad.

mikeller, 2020-10-30:
@KnzHz:

> Yeah, it's hard to know exactly what they do, but my experience is that OSD needs to be enabled for screen location.

Yeah, I saw that too - my suspicion is that this is DJI's weird way to allow the user to 'enable / disable the DJI OSD through Betaflight' - I can not imagine that they would do something as loopy as grabbing OSD displayport frames and using them to somehow detect element positions. But you never know. :wink:

KnzHz, 2020-10-30:
I have a one of these in the mail as we speak, once it arrives I will try to confirm the error and then do a build without the OSD commit, if nothing else it will take that theory out of the equation. Btw, could it be something as "simple" as CPU load?

@mikeller: They use MSP commands for all data, not displayport AFAIK, that's why they don't have Warnings :P 

I actually have no idea how they get the element locations. You have to enable Custom OSD in the goggles, so I don't think they use the OSD toggle in BF to enable/disable, it must be something like they grab something that OSD needs. There's DJI boards without OSD chip, I think but I'm pretty sure the feature still needs to be enabled, also, you'd need to, to pick elements. 

And thank you for your help, I appreciate that this is a mouthful for most of you as my feeling is that none in the team use DJI and have the system readily at hand.

jseaber, 2020-10-30:
@KnzHz 

> Btw, could it be something as "simple" as CPU load?

Could be. It's possible more data framing errors occur on the software serial RX line under higher usage. If this hypothesis is correct, the solution still boils down to relying on hardware UARTs to reduce errors, or begging DJI to improve their side. Have you tried reducing the number of OSD elements on 4.2 to see if there is a correlation to OSD reliability?

The Vista TX outputs data every ~10ms, with 18 transmits of 6 bytes of data before the pattern repeats (related to number of OSD elements?). The first 3 bytes are consistent [0x24, 0x4D, 0x3C] and the final three bytes vary. 

[image]

The number of bytes of output from the FC varies, with some rather long data chunks. There are occasional framing errors on this line, although I did not witness a collapse of OSD elements during the 1 minute of data collection (of course, disarmed).

[image]

tobstare, 2020-11-11:
I could deal with it. But I need someone with the error described, who could test the firmware on his Flywoo Explorer and give me appropriate feedback as to whether the error still exists until we get the error with bisect.

enzofpv, 2020-11-26:
ciao ho acquistato F4 2-4S AIO Brushless Flight Controller 20A v3, ma non riesco a far funzionare correttamente l'osd, praticamente il simbolo della direzione di casa non viene fuori e il punto di partenza non funziona Ho provato a cambiare il carattere, ma la casa simbolo di direzione se vede un trattino come il meno - nopn la freccia ... Ho inserito anche la batteria ma non cambia comunque il carattere. Ho aggiornato il firmware all'ultima versione, ma continua a non funzionare. Ho installato il gps su softserial, compaiono solo i satelliti di altitudine ma non la freccia di direzione home e i metri per i punti home. Uso il sistema hd caddx

carlcrott, 2020-12-27:
> solution

> Could be. It's possible more data framing errors occur on the software serial RX line under higher usage. If this hypothesis is correct, the solution still boils down to relying on hardware UARTs to reduce errors, or begging DJI to improve their side.

I'm seeing a videos where its claimed that Caddix Vista was working over  soft serials (https://www.youtube.com/watch?v=5PaUaXQbnDE&t=50s).  But maybe *just* the OSD dropping out is unrelated?  Not sure...

Also others videos (https://www.youtube.com/watch?v=xLo_bHVqhZ0) where the GPS is switched to the soft serial, and they claim that fixes the misbehaving OSD. 

Yet another video that I seem to have lost claims that theres a whole other fix not related to shielding, swapping UARTS, or flashing betaflight.

closed 2021-02-16 by github-actions[bot]