betaflight/betaflight #10208: VTX power adjustment plus level shown in OSD

flyandride, 2020-09-24:
I would like to see VTX power adjustment added to adjustments in Betaflight and the level shown as 1-4 in the OSD as an option. 

iNav has this feature and I use it in flight often.  Power adjustment can be accomplished with a 3 way switch, one way increases, the other way decreases.  I use the throttle trim switch for this.  If I am interfering with another pilot's video at a meet I can quickly reduce power or increase as my craft gets farther away.

etracer65, 2020-09-24:
VTX power level can already be changed on a switch - see the CLI `vtx` command. Also the VTX band/channel/power is already available as an OSD element.

flyandride, 2020-09-24:
> VTX power level can already be changed on a switch - see the CLI `vtx` command. Also the VTX band/channel/power is already available as an OSD element.

Nowhere have I been able to find any documentation on how to do that.

etracer65, 2020-09-25:
Well, start with `help vtx` in the CLI.
```
vtx - vtx channels on switch
	<index> <aux_channel> <vtx_band> <vtx_channel> <vtx_power> <start_range> <end_range>
```

For example the following will configure a 3-position switch on Aux3 to switch power levels 1/2/3.
```
vtx 0 2 0 0 1 900 1200
vtx 1 2 0 0 2 1300 1700
vtx 2 2 0 0 3 1800 2100
```

This can also be used to switch to specific band/channels based on a switch but since the band/channel in the example above are 0 it won't change channels - only power. See the help for the vtx command.

And for the OSD element:

[image]

haslinghuis, 2020-09-28:
Added to the wiki: https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings#change-vtx-power-level-using-aux-channel

closed 2020-12-25 by stale[bot]