betaflight/betaflight #10210: Betaflight 4.2 OSD freezes, Motors stop and total loss of control

Silver-Power, 2020-09-26:
During a test flight i was executing mild manuevers with the intent to observe in my OSD (Debug Mode: Dmin) the D term Value variation during a propwash situation. I rolled left up to the set limit of 50 degrees in a coordinated turn and when completed the turn and return to level the OSD Froze the motors stop and i had no control of the drone which resulted in an unrecoverable crash. 
Please Review the attached DVR Footage.
[Matek_F722_SE_Failure2.zip](https://github.com/betaflight/betaflight/files/5287266/Matek_F722_SE_Failure2.zip)

**Flight controller configuration**
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.2 Aug 16 2020 / 01:48:06 (e833ac612) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722SE, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722SE
manufacturer_id MTKS
mcu_id 002700463338510339393832
signature 

# name: Silver Power

# feature
feature GPS
feature ESC_SENSOR

# map
map TAER1234

# serial
serial 0 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 1000 1300 0 0
aux 1 1 1 1350 1650 0 0
aux 2 46 1 1700 2000 0 0
aux 3 40 2 1700 2000 0 0
aux 4 47 1 1000 1300 0 0

# rxrange
rxrange 0 1153 1845
rxrange 1 1153 1845
rxrange 2 1153 1845
rxrange 3 1153 1845

# vtxtable
vtxtable bands 8
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LO_BAND1 U CUSTOM  5325 5348 5366 5384 5402 5420 5438 5456
vtxtable band 7 LO_BAND2 O CUSTOM  5474 5492 5510 5528 5546 5564 5582 5600
vtxtable band 8 HI_BAND  H CUSTOM  5653 5693 5733 5773 5813 5853 5893 5933
vtxtable powerlevels 5
vtxtable powervalues 0 1 2 3 4
vtxtable powerlabels 25 200 400 800 1W

# rxfail
rxfail 3 s 1300
rxfail 5 s 1850

# master
set acc_calibration = 88,-30,110,1
set align_mag = CW270FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 2700
set mag_hardware = AUTO
set mag_declination = 428
set mag_calibration = -62,2,-113
set min_check = 1020
set max_check = 1980
set serialrx_provider = SPEK1024
set dshot_bidir = ON
set failsafe_switch_mode = STAGE2
set failsafe_throttle_low_delay = 300
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 2400
set vbat_min_cell_voltage = 250
set vbat_warning_cell_voltage = 300
set ibata_scale = 101
set ibata_offset = -860
set small_angle = 180
set gps_provider = UBLOX
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 25
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1000
set gps_rescue_throttle_max = 1850
set gps_rescue_ascend_rate = 300
set gps_rescue_descend_rate = 100
set gps_rescue_throttle_hover = 1450
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_dth = 50
set gps_rescue_alt_mode = FIXED_ALT
set osd_rssi_alarm = 0
set osd_alt_alarm = 50
set osd_tim1 = 3843
set osd_tim2 = 5121
set osd_vbat_pos = 2423
set osd_rssi_dbm_pos = 87
set osd_tim_1_pos = 2327
set osd_flymode_pos = 2448
set osd_g_force_pos = 2241
set osd_throttle_pos = 2369
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2454
set osd_mah_drawn_pos = 2392
set osd_craft_name_pos = 427
set osd_gps_speed_pos = 2305
set osd_gps_lon_pos = 2433
set osd_gps_lat_pos = 2401
set osd_gps_sats_pos = 2137
set osd_home_dir_pos = 2128
set osd_home_dist_pos = 2131
set osd_altitude_pos = 2120
set osd_debug_pos = 2177
set osd_power_pos = 2316
set osd_pidrate_profile_pos = 44
set osd_warnings_pos = 385
set osd_pit_ang_pos = 2156
set osd_rol_ang_pos = 2262
set osd_battery_usage_pos = 385
set osd_disarmed_pos = 2379
set osd_nheading_pos = 2416
set osd_esc_tmp_pos = 2145
set osd_esc_rpm_pos = 161
set osd_core_temp_pos = 2113
set osd_rate_profile_name_pos = 2099
set osd_pid_profile_name_pos = 2093
set osd_profile_name_pos = 55
set osd_stat_max_dist = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_flight_dist = ON
set debug_mode = D_MIN
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 4
set vtx_freq = 5865
set vcd_video_system = NTSC
set displayport_max7456_wht = 0
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900
set name = Silver Power

profile 0

# profile 0
set vbat_pid_gain = ON
set acro_trainer_angle_limit = 50
set auto_profile_cell_count = 6

profile 1

# profile 1
set vbat_pid_gain = ON
set acro_trainer_angle_limit = 50
set auto_profile_cell_count = 5

profile 2

# profile 2
set vbat_pid_gain = ON
set acro_trainer_angle_limit = 50
set p_pitch = 60
set d_pitch = 49
set p_roll = 55
set d_roll = 46
set p_yaw = 59
set d_min_roll = 30
set d_min_pitch = 33

# restore original profile selection
profile 2

rateprofile 0

# rateprofile 0
set rateprofile_name = 6S
set roll_rc_rate = 60
set pitch_rc_rate = 60
set yaw_rc_rate = 70
set throttle_limit_type = SCALE
set throttle_limit_percent = 65

rateprofile 1

# rateprofile 1
set rateprofile_name = 5S
set roll_rc_rate = 60
set pitch_rc_rate = 60
set yaw_rc_rate = 70
set throttle_limit_type = SCALE
set throttle_limit_percent = 75

rateprofile 2

# rateprofile 2
set rateprofile_name = General
set roll_rc_rate = 60
set pitch_rc_rate = 60
set yaw_rc_rate = 70

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: SDCARD_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
```
⦁	6S Battery Li-ion 2600mAh (35A Max) Throttle Limit Scaled at 65%
⦁	Fight Mode:
		1) Acro Trainer with 50 Degrees Tilt Limit on Both Pitch & Roll Axis
		2) Angle with 55 Degrees Tilt Limit on Both Pitch & Roll Axis
⦁	Props: DalProp CYCLONE (Purple) T5046C Pro
⦁	Default PID Tuning Betaflight 4.2 (Configurator 10.7)
⦁	VTX Power Settings: 800mW
	VTX Antenna: Right Hand Circular Polarization 
⦁	GPS (Beitian BN-880Q GPS+GLONASS Dual GPS Antenna Module)
⦁	Motors: EMAX ECO Series 2207 1700KV
⦁	ESC: T-MOTOR 4-in-1 Velox 45A BL-Helli 32 with ESC & Current Sens Telemetry
⦁	Flight Controller: Matek System F722-SE F7 Dual Gryo
⦁	Frame: Geprc MARK4 225mm 5 Inch
⦁	Receiver: Upgraded R720X 2.4G 20CH DSM2 DSMX Compatible Micro Receiver with 9cm Monopole antenna inside 
        protective tube with Cap.
⦁	VTX Eachine Leaf TX806 72CH 1000mW
⦁	Camera Caddx Turbo EOS2 NTSC 16:9
⦁	VTX Antenns iflight RHCP Pagoda MMCX Connector
⦁	Power Connectors XT60 & LOW-ESR Capacitors 1x 470μF 35V Rubicon ZLH on Flight Controller & 1x 470μF 35V 
        Rubicon ZLH on ESC
```

etracer65, 2020-09-26:
This would generally be a power problem leading to the flight controller browning out (for low voltage) or crashing due to a voltage spike from the ESCs. Could also be a wiring problem causing an intermittent break or short. Or finally a failing flight controller.

You're just going to have to go through the quad and check everything and try swapping parts.

Silver-Power, 2020-09-27:
etracer65, Thank you for your Comment!
The Flight Controller in my built was powering the following Components:

- The RC Receiver Connected to 3.3V Pad with Measured Operating Current: 76mA

- The GPS Module Connected to 4.5V Pad with Measured Operating Current: 57mA

- The FPV Camera Connected to 5V Pad with Measured Operating Current: 112mA

The flight Controller is stated as capable to provide 2A Continuous @ 5V. 
Furthermore the 3.3V LDO is stated as capable to provide 200mA @ 3.3V

I did not measured the Operating Current draw of the flight controller board (@ the 5V Internal Bus) because i thought that the 2A continuous carrent rating was adequate.
Do you suggest that we should power every peripheral Component from a seperate UBEC just for safety?

etracer65, 2020-09-27:
This is not the place to get hardware support. Try other options like RCGroups:  https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

Slack: https://slack.betaflight.com

Facebook: https://www.facebook.com/groups/betaflightgroup

closed 2020-09-27 by etracer65