betaflight/betaflight #10213: Total Flights and Total Time Stats failing to update

Justicefpv, 2020-09-28:
**Describe the bug**

When enabling osd_stat_total_flights and osd_stat_total_time stats they are not updated after each flight and remain at 0

**To Reproduce**

Enable both Total flight and Total time on the Post Flight Statistics screen. Make a flight or 2 and check the statistics screen.

**Expected behavior**

Stats should update after each flight.

**Flight controller configuration**

Aikon F7 

**Additional context**

Tested on build #2288 (26-Sep-2020 01:38:12)

etracer65, 2020-09-28:
See https://github.com/betaflight/betaflight/pull/10011

Justicefpv, 2020-09-28:
stats_min_armed_time_s is set to -1 by default is that off i.e no recording? and anything over is the delay before recording begins.

closed 2020-10-06 by Justicefpv

dmak, 2024-06-08:
> stats_min_armed_time_s is set to -1 by default is that off i.e no recording? and anything over is the delay before recording begins.

Is `stats_min_armed_time_s` documented somewhere? And maybe Betaflight configurator could display a warning when "Fly time total" OSD is enabled, but `stats_min_armed_time_s == -1` (because otherwise it stays zero)?

[image]