betaflight/betaflight #10215: "Lipo counter" in OSD

MBoerlage, 2020-09-28:
Training aid? Bragging rights? As fanatic racers we always look at how many packs we flew on an event. Its our way of measuring how intense the training was, how many tries it took to get it right. Its also great to know what count you where on for "at-home DVR reviews". Especially during training events, funflys (not a formal event with a race director), counting packs while having fun is pretty hard. Reality is that half way the day we all loose count and then we just make up a number and round up - oooh, I said it... 

It would be awesome to have an automatic counter that helps me estimating how many packs I fly at an event. 

**What it would count: "lipo packs flown on a single quad"**

- One approach could be to count the number of times you armed with a fully charged pack: average cell voltage "max cell voltage... max cell voltage*90%", e.g. "4.35 to 4.00 volt".   
- It should not count arming right after turtle mode; its very unlikely that after a turtle mode the average cell voltage is higher than 4.20.
- The counter could be either continuously running and restart at "0" after "999", or manually resettable. We never fly more than 99 packs on a day.

**Where it would show:** 

- in OSD during flight - I would put "#n" where "n" is the lipo number in my screen
- in the final stats, is chosen so with selector switch in BF OSD configurator

**What this would unlock in user experience:** 

- data based bragging rights... important
- DVR review with #1 v.s. #20: investigate how you improved your line, or became more sloppy
- number of packs flewn with this quad before I crashed it to pieces and had to replace the flight controller

**Alternatives:** 

- you could use an FC that has an SD card reader and an SD with a lot of storage. The blackbox file number increases with every time you arm (it also counts recovery after turtle mode)
- before take off stand in front of your quad and count with your fingers

etracer65, 2020-09-28:
Use the already existing persistent stats. `set stats = ON`. Then the total number of flights can be displayed on the OSD post-flight stats screen. You can reset the counter anytime you want with `set osd_stat_total_flights = 0`.

MBoerlage, 2020-10-20:
I have updated the request after extensively testing the existing feature @etracer65 suggested, many thanks for that!

haslinghuis, 2020-10-20:
What is the value of this if you already have the stats in post flight? Adding a static number to OSD will only give one more distraction in flight imho.

MBoerlage, 2020-10-20:
Thank you for your question @haslinghuis. 

We often review DVR and also parts of DVR, in this way you do not have to search for the final stats screen. It would allow me to quickly compare pieces of DVR more easy. Another value is that I can watch someone else channel during raceday and see what pack he/she is on - we often have funflys where we compete to packs flown during that day. 

Fore sure there are elements that are of value to some users and a nuisance to others - that's why my request would be to be able to turn this off similar to other elements. This particular element is surely valuable for just a small group of fanatic racers.

haslinghuis, 2020-10-20:
Fair enough. Lets see if this gets a feature request label.

limonspb, 2020-10-20:
I actually like this feature request as I like to count my practice packs too :)

I'm down to try to implement it as well if it gets a feature request label. Or what's the process here, @haslinghuis ?

As a further improvement would be also good to have a "reset" from OSD menu if you want to reset the counter.

haslinghuis, 2020-10-20:
Before waisting time for me the feature request label is only a conformation if requests are accepted by the members.

mikeller, 2020-10-20:
I still don't see much value in it. But then, if all it does is show the current flight count from the stats in the OSD it won't cost us much flash space either.

And reset through CMS should be minimal effort as well.

PR #10300 (2020-10-25): Total flights stats for OSD (firmware part)
Adding total flights count from persistent stats to OSD.
Firmware part
Feature request: #10215 

Pull request for configurator part: https://github.com/betaflight/betaflight-configurator/pull/2254

Fixes #10215.
closed 2020-10-25 by mikeller