betaflight/betaflight #10219: Anti-gravity also applying for throttle drops

TonyAcroFPV, 2020-09-30:
Feature Request 

When the throttle is dropped it tends to move the nose around a bit, quite similar to how the nose behaved with throttle pumps without anti-gravity.

If anti-gravity were to apply to throttle drops, I think it could help with keeping nose stability.

I have noticed this instability with throttle drops on lots of quads regardless of tune or hardware and I think this could be a suitable fix.

RipperDrone, 2020-10-01:
Hmm, seems this has been discussed many times - though there is no viable solution for me either. On all builds with strong motors I'm getting this effect.

https://github.com/betaflight/betaflight/issues/6018

RipperDrone, 2020-10-01:
@ctzsnooze has there been any revision of AG as a remedy since, as mentioned in the old thread...?
[image]

Justicefpv, 2020-10-02:
https://github.com/betaflight/betaflight/pull/10163

closed 2020-10-04 by TonyAcroFPV