betaflight/betaflight #10220: Serial Port Issue with SERIAL_BIDIR/SERIAL_BIDIR_PP

tonycake, 2020-09-30:
The existing SRXL-2 driver, and the new GHST driver use bidirectional, half duplex serial communication on one pin of the selected UART, the Tx pin. 
When the Rx drives the FC there is no problem, the scope traces are clean, Rx drives Push-Pull, rise and fall times are correct with 400k+ serial comms. 

When the bus is turned around however, and the FC drives the Rx (Telemetry) the default mode for SERIAL_BIDIR, the FC uses Open Drain (OD) mode, and the weak pullups on the Rx are not enough to create a clean rising edge. It 'works', but is far from ideal, and wouldn't take much noise to start corrupting date. 

The natural fix would appear to be SERIAL_BIDIR_PP, but what this seems to do is hold the serial line high even when in Rx mode, thus suppressing the control link between the Rx and FC. In this mode, the telemetry data is clean however, and as it should be. 

To reproduce, just hook up a SRXL-2 receiver to a Tx pin, configure BF in SRXL-2 mode, and put an oscilloscope on the Tx pin. 

The way I would expect this to work is that in SERIAL_BIDIR_PP mode, the Tx pin would be in input mode while receiving, and would switch to PP output mode when sending telemetry. 
[image]

This issue has been noted on F4, and F7 processors ruling out a single-processor driver issue.

mikeller, 2020-10-01:
@tonycake: Do you have any proposals for how to address this?

tonycake, 2020-10-01:
Not yet... but I am happy to dive in there and fix it, or at least understand whether it does have a clean fix.

tonycake, 2020-10-01:
Bit more news on this... firstly, it seems that in order to run in BIDIR_PP mode, SERIAL_BIDIR | SERIAL_BIDIR_PP is required. This avoids the bus being held, suppressing incoming packets. 

However... When these options are requested, with a non-inverted serial stream (SRXL2, GHST), the serial line sits 1/2 way between 3.3V and Gnd. 
The reason for this appears to be this code (in this case from serial_uart_stm32f7xx.c: 

    if ((options & SERIAL_BIDIR) && txIO) {
        ioConfig_t ioCfg = IO_CONFIG(
            ((options & SERIAL_INVERTED) || (options & SERIAL_BIDIR_PP)) ? GPIO_MODE_AF_PP : GPIO_MODE_AF_OD,
            GPIO_SPEED_FREQ_HIGH,
            ((options & SERIAL_INVERTED) || (options & SERIAL_BIDIR_PP)) ? GPIO_PULLDOWN : GPIO_PULLUP
        );

What this does is request weak pulldowns if the serial line is inverted (correct) OR if in Bidirectional PushPull mode, which appears to be incorrect. 
Bidirectional PushPull mode, if non-inverted, with weak pullups to the (high) idle level, doesn't want to fight against weak pulldowns on the FC side. 

There is a serial option for 'NOPULL' SERIAL_BIDIR_NOPULL, but this only appears to be used in the softserial driver. 

Not sure what the correct fix is. One option would be to accompany the SERIAL_BIDIR_NOPULL option with a SERIAL_BIDIR_PULLUP, and SERIAL_BIDIR_PULLDOWN option, but one would think that just by virtue of specifying whether the port is inverted or not would allow this to be determined. 

Can anyone who has worked on this code chime in?

tonycake, 2020-10-02:
Two ways to fix this: 

Option 1) Remove pulldowns for Inverted OR BIDIR_PP. Shouldn't affect anything, but more risky than option 2)
    if ((options & SERIAL_BIDIR) && txIO) {
        ioConfig_t ioCfg = IO_CONFIG(
            ((options & SERIAL_INVERTED) || (options & SERIAL_BIDIR_PP)) ? GPIO_MODE_AF_PP : GPIO_MODE_AF_OD,
            GPIO_SPEED_FREQ_HIGH,
            ((options & SERIAL_INVERTED) || (options & SERIAL_BIDIR_PP)) ? 0 : GPIO_PULLUP
        );

Option 2) Just remove pulldowns for BIDIR_PP, which will leave them enabled for INVERTED. 
    if ((options & SERIAL_BIDIR) && txIO) {
        ioConfig_t ioCfg = IO_CONFIG(
            ((options & SERIAL_INVERTED) || (options & SERIAL_BIDIR_PP)) ? GPIO_MODE_AF_PP : GPIO_MODE_AF_OD,
            GPIO_SPEED_FREQ_HIGH,
            (options & SERIAL_INVERTED) ? GPIO_PULLDOWN : GPIO_PULLUP
        );

Personally I am hedging towards option 2. 

(Note that this affects F7, F3, G4, and H7 builds)

ledvinap, 2020-10-02:
IMO this is copy/paste/refactoring mistake. option 2) looks correct to me.

In general you want pullup/pulldown to IDLE state on serial input pin (input in bidir). Without it, noise can couple to unconnected input, causing interrupt storm and possibly severe consequences.

mikeller, 2020-10-04:
@tonycake, @ledvinap: In this case I am assuming that we should not be enabling push/pull when inverting the UART? Or else we will be left in the same situation with inverted and forcing push/pull AND a pulldown, with the pull down fighting the push/pull.

mikeller, 2020-11-08:
@tonycake: Do you want to open a pull request for this - going for option 2 seems to be reasonable if the above statements hold true.

tonycake, 2020-12-04:
Ok, time to fix this. After spending a few days debugging noise-related problems with certain FCs and receivers it turns out that this bug was the source of the problem. 

Basically, in a noisy system (most of them), any Rx protocol that uses bidirectional comms (SRXL-2, GHST, FPORT) can randomly drop packets with no warning. 

I will prepare a PR for it, with the one-line fix for each of the F7, F3, G4, and H7 builds.

closed 2021-02-16 by github-actions[bot]