betaflight/betaflight #10233: RC smoothing filter Input Cutoff Frequency calculated incorrectly when using Frsky 8+ channels

mrbig, 2020-10-02:
I'm using an Frksy XM+ receiver with betaflight 4.2.1, connected with s.bus.

When I use more than 8 RC channels, and have RC input smoothing turned on at the defaults I get this weird hook looking ripple on any of the channels that uses the smoothing:
[image]

The interval of these steps is ~18ms, equivalent to 2 s.bus frames.

If I set rc_interp to manual and set rc_smoothing_input_hz to 25hz, or below, then the input turns smooth:
[image]

**To Reproduce**
receiver: Frsky XM+
firmware: ACCESS, d16 LBT, RSSI16
more than 8 active channels
rc_interp = auto
everything on defaults for rc filter.

**Expected behavior**
It would be nice if the auto cutoff calculation could handle this case.

I'm not an expert on rc protocols, but as far as I know when using more than 8 channels, then the RC input is split into two frames when transmitted from the RX to the FC. Seems like to me that the calculation uses this higher frame rate as the update frequency. Whoever the real frequency is only half of the calculated one.

**Flight controller configuration**
Not much to see here. Everything on defaults for rc_ values.

Aux channels:
```
# aux
aux 0 0 0 1300 2100 0 0
aux 1 13 3 1550 2100 0 0
aux 2 28 1 1400 2100 0 0
aux 3 35 4 1700 2100 0 0
aux 4 47 1 900 1325 0 0

set rssi_channel = 16
```

**Setup / Versions**
 - MAMBAF722 (MBF7)
 - betaflight: 4.2.1 Aug  1 2020 / 01:41:07 (aa96576d7) MSP API: 1.43 (based on caa0d68)
 - frsky xm+ XM+LBT170313-RSSI16 firwmare
 - receiver connected over s.bus

** Workaround **
Set Input cutoff type to MANUAL, and set Input Cutoff Frequency to 25 or lower:

```
set rc_interp = MANUAL
set rc_smoothing_input_hz = 25
```

etracer65, 2020-10-02:
This is a well-known limitation of Frsky D8/D16 protocols and not anything we can fix. The receiver continues to send data frames to the flight controller at 9ms intervals even though the RC channels are only updated every other frame. This is because the over the air protocol from the transmitter to the receiver only has the bandwidth to send 8 channels so it alternates and sends 1-8 in one transmission and then 9-16 in the next.

Short answer: don't bind with more than 8 channels or set manual cutoffs of 25hz for RC smoothing.

mrbig, 2020-10-02:
I wonder is a way for the auto tuning to measure the update frequency correctly? Now the code treats every frame as equal. Could be changed to measure the interval between updates on a specific channel?

etracer65, 2020-10-02:
This problem is not something new and has been discussed at great lengths and there is no viable solution. Betaflight doesn't have enough information to tell whether the channels are updated every from or every other frame.

Channel values frequently don't change between frames - just think about the sticks being centered for example. So it's not possible to differentiate between a 8 channel value that simply hasn't actually changed vs. a 16 channel value that's only updated every other frame.

The receiver sends all 16 channels with every frame so we can't tell which partial frame and which half of the channels were updated from the transmitter.

mrbig, 2020-10-02:
Thank you very much, now I have the full picture.

I might have an idea: what is instead of trying see how often a channel has changed, we would be looking for a distinctive pattern that would give away the situation? And this would be the special "staircase like" signal that is changing with every second frame.

So the algorithm could roughly be something like this:
 - look at one or few of the special channels like throttle
 - count how many frames have passed between each change. (we could stop counting after 3, or a few more if want to be more generic)
 - if we see a change after exactly 2 frames, increment the `likely_a_halfing_protocol` counter.
 - if this counter reaches a threshold then turn on a flag for the halfing workaround.

I have to admit, this has some potential issues:
 - it's very specialized for a specific problem
 - it will have a delay until getting it right
 - could have false positives, but I don't think it would affect more users than the current situation does.
Pros:
 - userfriendly :)

If there is a better place to discuss, or threads I should read before coming up with stale ideas, I'd be glad to point them out. Also feel free to close this issue if you believe it is a duplicate/wontfix scenario.

etracer65, 2020-10-02:
Doing this kind of signal pattern analysis in a microcontroller firmware is extremely difficult. Additionally you are not considering the behavior of other radio brands that also use the SBUS protocol. Remember, we don't know the type of radio and only that it's talking to the flight controller using SBUS (but could be Frsky, Futaba, DJI, Crossfire, Jumper, and so on) so making some kind of assumption about the behavior of the signal without knowing it's a Frsky system would lead to problems.

As I said, this has been discussed a lot and you haven't brought up anything new. So far no reasonable solution has been proposed. But feel free to work on it if it's something you feel strongly about.

mrbig, 2020-10-03:
I've implemented a proof of concept solution to this in PR #10241 It might explains my idea better. We could discuss the topic there if you like. I understand that special kind of care should be taken with such kind of change, so I'm not pushing it, it's just a proposal.

mikeller, 2020-10-04:
@mrbig: A fix for this has been identified in OpenTX a long time ago (opentx/opentx#6037), and according to the maintainers of OpenTX this has been fixed by now.
If not please propose a fix to OpenTX to get this resolved - adding hacks to everything around OpenTX to work around their poor engineering is not the way to go.

closed 2020-10-04 by mikeller