betaflight/betaflight #10234: Slip indicator in OSD

rgfuller, 2020-10-02:
A slip indicator would be really nice for FPV because even with camera angle compensation it is difficult to estimate turn coordination. [https://en.wikipedia.org/wiki/Turn_and_slip_indicator#Turn_coordinator](url)

rjalexa, 2020-12-15:
I completely agree. Now flying FPV but coming from full scale saiplane piloting, the turn&slip was the most important instrument with the second being the variometer :)

AirCombat, 2025-04-15:
I know this is old... I feel there is less utility in a quad. I fly helicopters and we typically don't have slip indicators as the nature of the type of aircraft makes turning efficiently less critical to flight (there is no real impact on fuel use or speed). I feel the same is true for quads. I know BF is used on other types of aircraft but I feel this feature would not be utilised by the majority of users. Just my opinion.

The funny thing is most helis just use a short peice of wool on the end of a stick in front of the glass to measure coordination, something that would be quite entertaining to see in front of a camera on a quad

closed 2026-09-26 by blckmn