betaflight/betaflight #10248: SP Racing H7 Zero

Peppe911, 2020-10-05:
New board, tested with only battery voltage from 4in1 esc, motors wire and current sensor wire.
No short circuits on the board.

When i measure ADC pad 2, in idle condition and when disconnected from the wire, i read about 800mv; total idle current from battery is less than 130mA.
When i connect the current sensor wire from ESC this voltage drops to about 250mv;
if i measure directly from this pad, with current sensor connected and with various loads, this voltage don't change.
If i measure directly esc current output, without connecting the wire to the FC, i read 0mv in idle, 30mv at 1.36A (4 motors at 1100) and 70mv at 2.64A (4 motors at 1150).

So i think someting is broken in the FC or in this settings.

DUMP:

`
# dump

# version
# Betaflight / SPRACINGH7ZERO (SP7Z) 4.2.2 Sep 14 2020 / 12:12:18 (e833ac612a) MSP API: 1.43

# start the command batch
batch start

board_name SPRACINGH7ZERO
manufacturer_id 

# name: x220s

# resources
resource BEEPER 1 D07
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 C06
resource MOTOR 8 C07
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B15
resource PWM 1 B15
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 B14
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 D01
resource SERIAL_TX 5 B13
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 E01
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 B15
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 D09
resource SERIAL_RX 4 D00
resource SERIAL_RX 5 B05
resource SERIAL_RX 6 C07
[... 1063 lines cut ...]
set rateprofile_name = -
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end
`

etracer65, 2020-10-05:
**Important: This is not the right place to report faulty / broken hardware. Your issue will be closed without further comment.**

Contact the manufacturer or supplier of your hardware, or check RCGroups https://rcgroups.com/forums/showthread.php?t=2464844 to see if others with the same problem have found a solution.

closed 2020-10-05 by etracer65

Peppe911, 2020-10-05:
> **Important: This is not the right place to report faulty / broken hardware. Your issue will be closed without further comment.**
>
> Contact the manufacturer or supplier of your hardware, or check [...]

Im not sure about is broken the hardware or it's just a wrong setting/bug in firmware.
For this reason i open here an issue instead write directly to @hydra