betaflight/betaflight #10251: Arming the quad when it is not suppose too

Ric-Alves, 2020-10-06:
Hi, I was testing new setup on modes tab and testing each one. I tough if I don't turn on the "safety" switch on motor tab, the motors will not spin at all, even I had the props on (a big mistake I know).
When I was testing the arm switch and the pre-arm switch, the motors start to spin...
I notice if I only activate arm switch the arm mode goes red with the message "(disable)", as suppose.
But if I activate the Pre-arm switch and next the arm switch, like I would do to fly, the arm mode goes yellow and the motors start to spin!
I thought that with the usb cable connected, betaflight configurator connected, the motors would only spin with the "safety" switch activated, at motors tab.

Is this a normal behavior? It could be danger to someone that doesn't know this.

It seams to me a configurator issue, or it would be a Betaflight issue?

Link youtube https://youtu.be/ABIy5fBAxbQ
Betafligh configurator 10.7.0
Betaflight : 4.2.3

`# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.3 Sep 20 2020 / 20:42:51 (2696b7c88) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FURYF4OSD
manufacturer_id DIAT
mcu_id 003d002f374d500320313048
signature 

# name: Alves

# feature
feature -AIRMODE

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# aux
aux 0 0 3 900 1125 0 0
aux 1 13 1 1925 2050 1 0
aux 2 26 0 900 1125 0 0
aux 3 28 2 1375 1600 0 0
aux 4 31 3 1800 2100 1 0
aux 5 31 1 1275 1400 1 0
aux 6 35 2 950 1150 0 0
aux 7 36 1 1700 1825 0 0
aux 8 39 0 1850 2100 1 0
aux 9 39 1 1450 1550 1 0
aux 10 39 3 1800 2100 1 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 400
vtxtable powerlabels 25 100 200 400

# master
set gyro_lowpass2_hz = 375
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 360
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_hardware = NONE
set acc_calibration = -39,9,99,1
set min_check = 1010
set max_check = 1990
set rssi_channel = 16
set rssi_src_frame_errors = OFF
set rc_smoothing_auto_smoothness = 7
set blackbox_p_ratio = 64
set dshot_idle_value = 320
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set bat_capacity = 680
set vbat_warning_cell_voltage = 355
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 2
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_cap_alarm = 700
set osd_logo_on_arming = FIRST_ARMING
set osd_tim2 = 1281
set osd_vbat_pos = 6594
set osd_rssi_pos = 6201
set osd_tim_2_pos = 6614
set osd_throttle_pos = 6177
set osd_vtx_channel_pos = 6645
set osd_current_pos = 6625
set osd_mah_drawn_pos = 4554
set osd_craft_name_pos = 6209
set osd_battery_usage_pos = 2536
set osd_disarmed_pos = 6187
set osd_esc_rpm_pos = 4247
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bb_no = OFF
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 3
set vtx_channel = 3
set vtx_power = 1
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5665
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 700
set name = Alves

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dyn_lpf_dterm_curve_expo = 7
set dterm_lowpass2_hz = 225
set vbat_pid_gain = ON
set anti_gravity_gain = 4000
set iterm_relax_type = GYRO
set yaw_lowpass_hz = 100
set throttle_boost = 7
set throttle_boost_cutoff = 25
set p_pitch = 48
set i_pitch = 72
set d_pitch = 43
set f_pitch = 84
set p_roll = 44
set i_roll = 68
set d_roll = 40
set f_roll = 79
set p_yaw = 47
set i_yaw = 72
set f_yaw = 79
set d_min_roll = 26
set d_min_pitch = 29
set ff_spike_limit = 70
set ff_smooth_factor = 20
set idle_min_rpm = 20

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 71
set tpa_breakpoint = 1550

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# `

mikeller, 2020-10-06:
Transferring this to the firmware repository.

@Ric-Alves: This is very odd - looking at the video, this should not be possible. When trying to reproduce, did you use the same sequence of steps, including changing configurator tabs and plugging in the battery?

Did you change anything in your configuration between the being able to reproduce and not? OpenTX configuration / Betaflight configuration?

Ric-Alves, 2020-10-06:
I will try to describe as close as I did in the first time. Sorry if I give to you too much not necessary information.

Just switching to a new radio (X9 Lite S) with more switchs compare the previous one (X9 lite), flashed the xm+ from ACCESS to D16, bind it to the radio using an external power source.
Connect the drone, at 4.2.0, changed the motor idle and dyn_idle values, get the diff all and save in (attached file).
Update to 4.2.3, paste the diff all from 4.2.0 at CLI, checked the receiver tab to adjust the end points at the radio and re-arrange the modes, since I switched to a new radio with more buttons. 3 buttons/functions at the same aux channel, Pre-arm, buzzer and Black box erase.
When testing the pre-arm settings this issue happens, tried several times, and all times occurs as described, checked the motor tab and the switch was off, switched to modes tab and test again, issue still present. So I did the video to reproduce, the first time, didn't film the radio, so I needed to do the video again. The issue was always present.

Yesterday I tried to reproduce it without success. I tried to flash the 4.2.0, paste the diff all of 4.2.3, flash again the 4.2.3, past the diff all of 4.2.0, test again but the issue is not present.

I think I described every step I did.

In the next days, I will try again as last time, but past the diff all of 4.2.0 when the FC has 4.2.0.

Diff all 4.2.0
[BTFL_cli_Alves_20201004_155057.txt](https://github.com/betaflight/betaflight/files/5337022/BTFL_cli_Alves_20201004_155057.txt)

Ric-Alves, 2020-10-10:
Today I tried to replicate the issue as described. Not able to reproduce it BUT the configurator had a weird behavior.
Steps done:
- Flash 4.2.0
- Paste Diff from 4.2.0
- At modes tab the modes, when I tried to set the modes when the issue occurred, cursors didn't react to the radio aux channel changes, for example aux4 (arming aux channel), changed as suppose too but when I was defined the new values from pre-arm at aux2 the little yellow cursor below the slider bar doesn't move when I flip the switch. 
And some other aux channel had the same behavior. 
If I press "save" button, they start to react to the radio for some seconds. 
After that the weird behavior returns.

I didn't had time to do more testes.

Could this be related with the first issue? Would you like a video with the issue?

mikeller, 2020-10-11:
@Ric-Alves: Hmm, this now sounds much more like a configurator issue than firmware. Can you try completely uninstalling configurator, downloading configurator again, and re-installing it?

KAJE1977, 2020-10-13:
Maybe usb cord not fully connected or wire inside broken. So the usb not connected and battery is so it will arm. Sometimes if i use bad wite the configurator still shows open and not close. But is disconnected from configurator so safety not active. Could be a stretch but had wierd issue similar with bad connection.

Ric-Alves, 2020-10-17:
I wasn't able to reproduce any of the issues after uninstalling the configurator, delete the configurator folders from harddrive and re-installing...

Ric-Alves, 2020-11-05:
Like said before, I wasn't able to reproduce the issue. But I was helping a friend to start on this hobby and he experience the same issue. The drone will arm when it isn't suppose too.
Betaflight configurator: 10.7.0
Drone: Iflight Nazgul v2 6s, beta 4.2.0. fully stock

When this happens, what do you need to know? Or there is some log file that could help to figure out where is the problem?

McGiverGim, 2020-11-06:
I doubt we can have some log that helps for that.
I think we only have two options:
- Or some developer has an idea and can fix the issue without more info.
- Or you and your friend are able to reproduce it with a list of steps

The only thing that maybe can help if it shows something unusual is the first page of the Configurator after connecting, in the right there is a list of flags with reasons to not arm. When this happens check this list to see if there is something there.

mikeller, 2020-11-07:
Agreed with @McGiverGim - without a reproduction it will be hard to fix this.

closed 2021-02-08 by github-actions[bot]