betaflight/betaflight #10253: Add TX/RX swapping support on F7 hardware UART (SWAP bit in USARTx_CR2)

VladimirP1, 2020-10-06:
I was switching from SBUS to F.PORT on a GEPRC BNF quad when I realized that this board has RX4 routed to the reciever connector. While this works for SBUS, for F.PORT, half duplex UART can only be enabled on TX4, which is routed to a solder pad. To keep the wiring clean and the connector detachable, I wanted to use half-duplex UART on RX4.

Closely reading the reference manual for the F7, I figured out, that it is possible to swap TX/RX, not through AFIO, but in the UART peripheral itself. The only problem is that the datasheet does not clearly explain if it works in half-duplex+inverted mode.

I have written some proof-of-cocept code [here](https://github.com/VladimirP1/betaflight/commit/f9e57c2e7d0aa998610cae78b805a812fce5ba8f) and have tested it in flight, I was really able to enable inverted half-duplex uart on RX4, F.PORT works with telemetry. The code is nowhere near production ready although.

Also, this feature would be useful when using UARTs in normal mode (without halfduplex or inversion). I think everybody could remember oneself after soldering some hardware and assembling a quadcopter to realize that the TX/RX are swapped.

I would like to see support for TX/RX swapping in upstream, although I am not very sure if I am ready to improve the code and merge or it would be better for a developer with more expirience in Betaflight to implement it.

jflyper, 2020-10-08:
The RX/TX pin swapping comes in pretty handy, and has been considered some time ago, so we know it can be done. However, devs didn't reach an agreement on where to handle the feature; at hardware resource (pin assignment to MCU function) level, UART driver level or UART user (final feature) level.

Give us your thoughts.

VladimirP1, 2020-10-09:
To swap RX/TX, the SWAP bit has to be set and it seems to me that the best place to do that is at UART driver level.

The other question is about how to present the setting to the user: should there be a 'swap' option, or should there be possibility to simply reassign the pins using resource commands (this one looks harder to implement, because AFAIK there is no way to check if the user has assigned TX to RX).

In my code I have added one parameter (an integer, whose bits show wether to swap pins or not) because I needed a quick and dirty solution. I think it would be better to add a parameter for each UART just like for baudrate settings.

ps. It would be interesting to see the code for the PoC you are talking about.
pss. I have never worked with Betaflight code before and my experience with it is only ~10 hours.

ledvinap, 2020-10-09:
@VladimirP : It may be possible to use resource assignment. It will complicate the logic a bit, but should not be that bad.

uartPinConfigure can check first normally, if mapping fails, try again with rx/tx swapped and set invert flag in `uartdev` if successful.
Then check logic for using single pin for both RX and TX, it should reveal most cases that may need modification.

But I don't remember this code well, some edge cases may surface.

etracer65, 2020-10-10:
I don't really think it should be handled at the `resource` level by an implicit swapping of the pins. Since this swapping the UART pins won't work on F4 it creates a poor user experience where resourcing the pins to swap the RX/TX will work in some cases and not others. I'd rather see an explicit pin-swap feature at the UART level with the capability only exposed for MCUs that support it.

ledvinap, 2020-10-10:
I'm not sure here, both variants will work.
Possible pin assignment is already target/cpu specific and only checked by BF. Available pin table can be augmented by inversion notice.
Also with explicit inversion, listing resources will show unswapped pins, also leading to confusion.

etracer65, 2020-10-10:
Our typical user doesn't care about the MCU pin assignments. They care about whether the physical RX and TX pads on the flight controller are swapped. Soldered the rx/tx backwards for your GPS or Crossfire receiver? No problem, just enable a "swap" switch on the UART in the Configurator Ports tab.

mikeller, 2020-10-11:
To me there are two distinct ways to look at this:
1. if this is a feature that is intended for target maintainers to use when designing targets (i.e. having a board with a TX UART pin on a pad that is labelled 'RX' and vice versa) - not sure if there is any possible use case for this. In this case using `resource` for this would be the right approach;
2. if this is a feature that is intended for users (i.e. to get around faulty assembly or a destroyed pad) then having it as a config setting would be the right approach.

BUT at the moment there is no way to do per-UART configuration - things like speed, inversion, one-wire/two-wire are all configured per protocol and not per port. This is not a good approach and the whole UART configuration facility should be pivoted to allow per-UART configuration, and this should be added in as another per-UART option.

jflyper, 2020-11-11:
- "Swapped by design" case
When there is a design with left over pin capable of doing UART RX, then it would be nice to convert it to UART TX, so that it can be used to provide an extra half-duplex UART.
Board designers can take advantage of this feature and actively use the swap feature for more options in pin assignment.
For this case, the swapping should be handled at `resource` level, and board pads/THs will be labelled according to the design.
There is one very minor drawback in this approach; users who wishes to create a brand new config may fall into some confusion.

- `resource` or config
For replacement of broken pads/THs, the swapping may be configured somewhere other than `resource`. However, since pin swapping is not part of the application protocol, it should be transparent to `uartOpen`. If the swapping is not handled by resource, it should be handled somewhere between `resource` and `uartOpen`.

- `resource` and config?
Can we do both `resource` oriented and user config oriented at the same time?

mikeller, 2020-11-11:
@jflyper: We _could_ have both, but we don't need to to get this off the blocks. Right now, resource seems to be easy and achieveable (just figure out if a TX pin for a hardware UART is an RX pin on the MCU, and if so, configure swapping). Doing it as a configuration setting will be harder as the UART config is a mess at the moment - we are probably better off to wait until this has been reworked.

Also, thinking of the 'work around incorrect wiring or broken pad' use case, in pretty much every other case this requires the user to do this via resource reconfiguration as well, so I don't think this will be too far from what users are already used to doing.

hydra, 2021-02-17:
As a designer of lots of FCs I would love to see support for pinswap.

I agree with @jflyper that turning a spare RX pin into a bi-directional uart via the pin-swap is very useful indeed, especially when it comes to routing a PCB as often all the orientation of the CPU is dictated by available space around the edge of the board and by the location of the USB port and USB signals and sometimes you don't want to, or can't route traces from opposite sides of the board to where the connector/pad is.

The AF mapping of the MCU pins and clockmux and interrupt priority also plays a large part in designing pin layout schemes which influences availability and physical pin location of UARTS.

PCB routing is often also constrained by location of connectors, pads and other components on the opposite side of the board from the MCU, so having greater flexibility of PCB routing is most welcomed.   For smaller MCU packages it's not too much of an issue, but for larger packages (LQFP100 on a 20x20 mount) or for high-density designs (e.g. AIO FC+4in1 ESC) then every mm of PCB space is critical.

There's also another use-case that's not been considered and that is stacking of RX and/or VTX units using through-hole pins to corresponding pins on the FC.   There's no industry standard with regards to ordering of GND/5V/VIDEO/RX/TX signals.  So being able to support multiple VTXs or RXs using the same set of pins with PINSWAP is required.

I think the `resource` approach is all that's needed, and the resource allocation should fail and inform users for MCUs that don't support it.

Having a UI for resource allocation instead of the CLI approach would help.

hydra, 2022-02-18:
I did this in https://github.com/betaflight/betaflight/pull/10940

This PR can be closed.