betaflight/betaflight #10259: Follow-up #7722/#7754 (M4 in CLRacingF7 not working)

Christiaan1990, 2020-10-07:
**Describe the bug**
I have bought the FC "clracing f7 v2.1" and I am having an issue, that is already reported in the past: M4 is not spinning. It was reported in #7722 and it is (probably) already fixed in #7754. With an oscilloscope I also have checked if the problem is maybe somewhere else, but the signal is only flat for this M4. My firmware is updated to the newest software, and i didn't make any changes after that (except for calibration). Could you tell me what I do wrong? 

**To Reproduce**
Install Betaflight-Configurator 10.7.0, update the firmware of the FC to 4.2.3, connect the FC, and start the motors with the configurator. M4 does not spin.

**Expected behavior**
All 4 motors spin

**Flight controller configuration**
```
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.3 Sep 20 2020 / 20:44:12 (2696b7c88) MSP API: 1.43
# config: manufacturer_id: CLRA, board_name: CLRACINGF7, version: b69c3a8d, date: 2019-10-19T22:37:25Z

# start the command batch
batch start

board_name CLRACINGF7
manufacturer_id CLRA

# master
set gyro_to_use = FIRST

profile 0

rateprofile 0

# end the command batch
batch end

# 
```

``` 
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: PINIO 1
A15: OSD_CS
B00: LED 1
B01: FREE
B02: FREE
B03: CAMERA_CONTROL
B04: BEEPER
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 2
B08: MOTOR 3
B09: MOTOR 4
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: PREINIT 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: CAMERA_CONTROL
TIM3: FREE
TIM4:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

**Setup / Versions**
 - Flight controller [clracing f7 v2.1, banggood];
 - Other components [no other components];

**Additional context**
My assumption is that the fix from #7722/#7754 is already included in the betaflight configuration. Therefore, I assumed that no changes were required. 

It was also written in #7722 that _"The solution was to use the LED pad."_ However, I thought that this problem is already fixed automatically in the new firmware that I have installed. Therefore, I have opened this issue. If this LED-solution is the actual solution to my problem, would it be possible if you write the CLI commands I have to send and save?

etracer65, 2020-10-07:
Sounds like failed hardware. Replace the flight controller.

MBoerlage, 2020-10-20:
Is M4 selected on the "M4 - M6" selector pad on the flight controller? Otherwise contact CLRacing, I have the same FC and no issues because my FC is by default selected to M4.

deltawhiskey1147, 2020-11-13:
On that flight controller you have to bridge two pads on  the board to get motor 4 working

deltawhiskey1147, 2020-11-13:
[image]

Christiaan1990, 2020-11-15:
Thank you for the replies :) Ok then my hope is not lost yet. 

@deltawhiskey1147 :
My FC looks a little bit different then on the picture, but I guess I should do it like this right? 
[image]

deltawhiskey1147, 2020-11-15:
Bridge the M4 and the pad right next to it. Leave the LED pad alone

Christiaan1990, 2020-11-17:
Thanks it works! :)
I've connected M4 and M5 just as you said.
Therefore, I will close the issue

[image]

closed 2020-11-17 by Christiaan1990

deltawhiskey1147, 2020-11-17:
Happy to help. Email me if you ever need help. I would be glad to help.
Happy flying