betaflight/betaflight #10267: Motors don't turn off in 3D mode when disarming in the motors testing tab

tommaier123, 2020-10-11:
Configurator: 10.7.0
Firmware: BTFL 4.2.3
Target: AIRB/OMNIBUSF4SD(STM32F405)

I recently tried out 3D mode and noticed that after testing in the motors tab the motors don't stop correctly when you uncheck the "I understand the risks" switch.

haslinghuis, 2020-10-11:
Please define `correctly`? Either it stops or it won't stop. Please explain.

tommaier123, 2020-10-12:
They don't stop. Instead they slowly spin backwards which corresponds to the throttle value in the motors tab that gets set to 1460 instead of 1500.
[MotorsNotStopping.zip](https://github.com/betaflight/betaflight/files/5366938/MotorsNotStopping.zip)

haslinghuis, 2020-10-12:
Running dshot or analog?

//Note: values may need to be revisited
 neutral3d = (FC.MOTOR_3D_CONFIG.neutral > 1575 || FC.MOTOR_3D_CONFIG.neutral < 1425) ? 1500 : FC.MOTOR_3D_CONFIG.neutral;
        }

tommaier123, 2020-10-12:
Running dshot300

haslinghuis, 2020-10-12:
I don't know why `1460` is the default in firmware. Configurator is setting it to `1500` but firmware overrides?

src/js/tabs/motors.js:        DSHOT_3D_NEUTRAL: 1500
src/js/tabs/motors.js:        var neutral3d;
src/js/tabs/motors.js:            neutral3d = self.DSHOT_3D_NEUTRAL;
src/js/tabs/motors.js:            neutral3d = (FC.MOTOR_3D_CONFIG.neutral > 1575 || FC.MOTOR_3D_CONFIG.neutral < 1425) ? 1500 : FC.MOTOR_3D_CONFIG.neutral;
src/js/tabs/motors.js:                $('div.sliders input').val(neutral3d);
src/js/tabs/motors.js:            zeroThrottleValue = neutral3d;
src/js/tabs/configuration.js:            $('input[name="3dneutral"]').val(FC.MOTOR_3D_CONFIG.neutral);
src/js/tabs/configuration.js:                FC.MOTOR_3D_CONFIG.neutral = parseInt($('input[name="3dneutral"]').val());
src/js/msp/MSPHelper.js:                FC.MOTOR_3D_CONFIG.neutral = data.readU16();
src/js/msp/MSPHelper.js:                .push16(FC.MOTOR_3D_CONFIG.neutral);
src/js/fc.js:            neutral:                    0,
src/js/backup_restore.js:                    neutral:                1460,

RipperDrone, 2020-10-17:
...or is there an overlay considering throttle deadband maybe?

haslinghuis, 2020-10-18:
Please look at: https://github.com/betaflight/betaflight/wiki/3D-Setup

closed 2020-10-18 by mikeller