betaflight/betaflight #10271: ESC RPM Telemetry Errors

Kagey-cmd, 2020-10-13:
I'm getting ESC telemetry errors at certain RPM (with all motors spinning) and during any rapid throttle punches (props off on desk, single motors or all 4). Spinning up a single motor slowly does not produce errors at any RPM.

_Originally posted by @jflyper in https://github.com/betaflight/betaflight/issues/9103#issuecomment-546611212_

Kagey-cmd, 2020-10-13:
I should mention this is with a fresh flash of both BLHeli_32 and Betaflight. Completely default everything, except bidirectional DSHOT.

etracer65, 2020-10-13:
Try BLHeli32 32.7.2 firmware.

narleybarley, 2020-10-13:
i am having same problem, I am running Holybro Kakute f7 V1.5 with 4 in 1 ESC (65amp). I finally got the motors to activate and once they got to 1/4 throttle it was almost like they had a mind of their own and the only way to get them to stop was failsafe or disarm. I have done 3 fresh flashed in betaflight and BL_Heli32 and ran so many different configs trying to get this to work. I just do not know anymore :(.

etracer65, 2020-10-13:
@narleybarley 

> i am having same problem, I am running Holybro Kakute f7 V1.5 with 4 in 1 ESC (65amp). I finally got the motors to activate and once they got to 1/4 throttle it was almost like they had a mind of [...]

This is not the proper place to seek support or help with configuration. See https://drones.stackexchange.com/questions/127/why-do-the-motors-on-my-drone-keep-spinning-up-and-max-out-when-i-run-up-the-thr

Try RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook https://www.facebook.com/groups/betaflightgroup).

Kagey-cmd, 2020-10-13:
> Try BLHeli32 32.7.2 firmware.

@etracer65 
This is the "Rev32.7.2 SBUS and S.PORT testcode" located [here](https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Rev32.7.2%20SBUS%20and%20S.PORT%20testcode), correct?

etracer65, 2020-10-13:
@Kagey-cmd You should contact the BLHeli developers for the correct version to use. Or post questions in the BLHeli32 rcgroups thread. https://www.rcgroups.com/forums/showthread.php?2864611-BLHeli_32-Power-to-perform

Kagey-cmd, 2020-10-13:
Ok... but you're suggesting I try the 32.7.2 firmware, which is in beta still, right?

etracer65, 2020-10-13:
> Ok... but you're suggesting I try the 32.7.2 firmware, which is in beta still, right?

I don't know the status of the BLHeli32 firmware as we don't develop it - ask the BLHeli developers. There was a timing bug that was fixed that affected bidirectional DHSOT and caused error packets. See https://github.com/bitdump/BLHeli/issues/468

closed 2020-10-18 by mikeller

Kagey-cmd, 2020-10-19:
FYI for anyone that finds this page, this was solved by replacing faulty hardware.
New FC (same make/model) fixed it.