betaflight/betaflight #10285: [RFE] Allow change failsafe behavior with an stick

soukron, 2020-10-18:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I must connect a laptop or a phone to my FC to change the behavior of the failsafe to either enable RTH or drop when I could use a stick to change it.

**Describe the solution you'd like**
I would be awesome if we could change with one stick the behavior of the failsafe stage 2 to return to home, drop or land. That would allow us to change easily when using the same drone for different flights without connecting a latop to it.

**Describe alternatives you've considered**
There's no alternative. The only way to change this setting is to connect the drone to a laptop or phone and change it.

**Additional context**
As said, allowing changes with sticks would be perfect for this setting.

This could be done using modes or assigning a channel to the stage 2 failsafe procedure in the failsafe settings page and then setting a range to each procedure (drop, land or gps rescue).

etracer65, 2020-10-18:
This can already be changed int he OSD menus.

soukron, 2020-10-19:
> This can already be changed in the OSD menus.

No for those using DJI :( 

I know this can't be considered an issue for betaflight itself but if the implementation doesn't takes too long and it's not complex would add a lot of added value.

mikeller, 2020-10-19:
@soukron:

> No for those using DJI :(

Unfortunately this can be said of pretty much every setting that can be done through the CMS in OSD, and only a very small fraction of these settings are available on switches at the moment - trying to add any arbitrary ones (or all) is simply not an option.
Fact is, the CMS that is used to drive the menus in OSD can be used in other ways as well, it already is possible to get these menus on HoTT and Spektrum radios, on OpenTX with CRSF, on MWOSD. We recommended to DJI to use this and make all settings available to their users, but they decided to implement their own solution which is rather lacking.

So if you want this changed, please open a complaint / feature request with DJI as a customer, asking them to implement support for the Betaflight CMS.

Solving this by adding 100s of switches just for DJI is not the solution.

soukron, 2020-10-29:
>> So if you want this changed, please open a complaint / feature request with DJI as a customer, asking them to implement support for the Betaflight CMS.
I already did. This request doesn't imply I'm not complaining to them.

>> Solving this by adding 100s of switches just for DJI is not the solution.
Not sure how many requests like this one you have to have "100s of switches". This was only one.

>> trying to add any arbitrary ones (or all) is simply not an option. 
Fair enough. But note that you already have sticks for options which are also available in the menu, I just thought you could add one more which applies to more situations (not only DJI) where you forgot to change the setting before taking off or if you are already flying and change your mind about flying far away (so you need RTH) or entering in a bando that you find and change the setting to avoid a RTH in a closed room.

And for sure, don't get me wrong, Betaflight rocks (I love it) even if it's not a perfect option right now (not saying that it's your fault) when using DJI system. I hope there's a mid-term between DJI's development and yours so nobody prefers to create a fork with dedicated DJI compatibility (they number of digital users is increasing a lot so you would end losing them). On the other hand bear in mind that there are other software solutions out there and I don't think DJI is going to be fully compatible with all of them so some extra work from your side sounds reasonable to benefit the end-users.

I know that my arguments are weak (but still valid), so feel free to close the request at your convenience.

mikeller, 2020-10-29:
@soukron: Sorry, the '100s' was hyperbolic. :smile: But I think you get the gist: At the moment the fact that the DJI system does not support the Betaflight CMS is a drawback. But if we allow every DJI user to name the settings in CMS that they would like to use, and then add them as switches:
- we will end up adding all CMS settings as switches;
- we will not be far off the 100s :wink: 

You are right that there are switches that change things that are changeable in CMS as well. For what I can tell they fall into 2 categories:
1. switches that allow changing of settings that pilots want to be able to change in-flight, where CMS is not available;
2. switches that were added before CMS was implemented, or before CMS had the capability to perform that particular function.

Especially for 1., there is an extra complication for the failsafe switch - this will also require the corresponding AUX channel to be set to 'hold', or else the failsafe will always be set to whatever behaviour is defined for 1500 when failsafe is entered and the default value for the AUX channel is injected.

soukron, 2020-10-30:
> Especially for 1., there is an extra complication for the failsafe switch - this will also require the corresponding AUX channel to be set to 'hold', or else the failsafe will always be set to [...]

Yeah, but that can be changed automatically as soon as the pilot sets the AUX as failsafe switch, all transparently.

But anyway, I get your point and I agree with you in some way, but I had to try it :) I won't waste more of your time so feel free to close the RFE and I hope DJI improves the integration.

soukron, 2021-01-01:
Hey, what about changing the scope of the RFE to enable/disable the GPS rescue mode based on a switch. I am still struggling to find a good and flexible way to configure it on the field without a laptop or a phone. Actually Joshua Bardwell kinda complains about this exact same thing [in one of his videos](https://youtu.be/-bYavyTRvx8?t=135).

etracer65, 2021-01-02:
You can already configure the failsafe to be GPS Rescue or Drop in the OSD menus - no need for a laptop. I said this above. And if you still can't do this because of DJI, then tell DJI to fix their OSD implementation.

Nothing that is affected by failsafe (like radio channels) can control the behavior of failsafe.

soukron, 2021-02-10:
> You can already configure the failsafe to be GPS Rescue or Drop in the OSD menus - no need for a laptop. I said this above. And if you still can't do this because of DJI, then tell DJI to fix their OSD implementation.

Can we forget DJI from the equation? I already got your point there. Anybody can forget to set the procedure properly (using the OSD or any other method) before taking off and this can help to set the right procedure for stage 2 procedure easily during the flight.

mikeller, 2021-02-14:
@soukron: In this case, the correct approach is to land, disarm, use OSD or the TX app to change the desired setting, and take off again.
If you wanted to support changing all of the possible firmware settings that you might forget to set prior to takeoff, you'd need a TX with 100+ channels and switches. ;-)

etracer65, 2021-02-14:
As I mentioned above, the very nature of a failsafe means that the radio channels become unreliable. Having radio channels control the failsafe behavior when the channels themselves are **affected by the failsafe** is a recipe for disaster.

soukron, 2021-02-22:
Feel free to close the request. 

I see you won't change your mind even if the arguments you provide are questionable (@mikeller: landing is not always an option, and @etracer65: channels can be set **on hold** to avoid interferences in the procedure). 

But it is what it is. I exposed a need, a quite simple solution, and you don't want to implement it. Fair enough. But don't try to change my mind: I still think it's a good thing to have it for everyone using Betaflight.

mikeller, 2021-02-22:
@soukron: If landing is not an option then your craft is lost at that point anyway, as being able to enable GPS rescue will not miraculously make landing an option, and your request is pointless.

soukron, 2021-02-25:
> @soukron: If landing is not an option then your craft is lost at that point anyway, as being able to enable GPS rescue will not miraculously make landing an option, and your request is pointless.

I'm not sure what's the point to keep the discussion open if you're not willing to accept the RFE, but I'm not leting you to say that the request is pointless.

Two scenarios to point your wrong approach and how the RFE could help:
 - You forget to enable GPS rescue before taking off... if you land to enable (assuming you can land safely while being far from your position), the home reference will be reset to that new place instead of the original one. Landing doesn't help here, but a switch to enable (and keep the original home reference) would do it.
 - You forget to disable GPS rescue before taking off... and while flying you realize that the conditions for a safe emergency return (if required) are not the best ones. How safe is landing (assuming you can) while being far from your position so you can use the OSD and then you taking off again? Wouldn't be safer use a switch to disable it?

I reiterate: feel free to close the issue, that's okay and you're entitled to do it. But the request is totally legit and useful so don't try to make it pointless only because you don't want to accept it.

closed 2021-04-05 by github-actions[bot]