betaflight/betaflight #10296: HEADFREE + GPS_RESCUE problem.

MichelArhipov, 2020-10-21:
Why add the headfree offset to attitude.values in /betaflight/src/main/flight/imu.c?

STATIC_UNIT_TESTED void imuUpdateEulerAngles(void)
{
…
    if (FLIGHT_MODE(HEADFREE_MODE)) {
       imuQuaternionComputeProducts(&headfree, &buffer);

       attitude.values.roll = lrintf(atan2_approx((+2.0f * (buffer.wx + buffer.yz)), (+1.0f - 2.0f * (buffer.xx + buffer.yy))) * (1800.0f / M_PIf));
       attitude.values.pitch = lrintf(((0.5f * M_PIf) - acos_approx(+2.0f * (buffer.wy - buffer.xz))) * (1800.0f / M_PIf));
       attitude.values.yaw = lrintf((-atan2_approx((+2.0f * (buffer.wz + buffer.xy)), (+1.0f - 2.0f * (buffer.yy + buffer.zz))) * (1800.0f / M_PIf)));
    }
…
       
}

Yes, now attitude.values shows the correct direction of the abstract "nose" in HEADFREE mode,
but the pidLevel function in /betaflight/src/main/flight/pid.c starts working incorrectly.

STATIC_UNIT_TESTED float pidLevel(int axis, const pidProfile_t *pidProfile, const rollAndPitchTrims_t *angleTrim, float currentPidSetpoint) {
…
Now here is angle in body coordinates, but attitude.raw in earth coordinates. 

    const float errorAngle = angle - ((attitude.raw[axis] - angleTrim->raw[axis]) / 10.0f);

…
   
}

Proper flight control in HEADFREE mode is achieved by using the imuQuaternionHeadfreeTransformVectorearthtobody function in /betaflight/src/main/flight/rc.c,
which uses the headfree offset to transfer control signals from earth coordinates to body coordinates but does not use attitude.values in any way.

haslinghuis, 2020-10-24:
Thanks. Feel free to contribute. I am happy to assist or do it for you with your input.

mikeller, 2020-10-24:
@MichelArhipov: Headfree mode has not been maintained in Betaflight for some time, and it is well possible that it is broken and unusable at this point in time. The reason for this is that there do not seem to be too many users with an interest in it. If you have an interest then please feel free to contribute by proposing changes to fix it.

MichelArhipov, 2020-10-26:
Thank you for the response, haslinghuis and mikeller.

If for some purpose it is important to maintain attitude.values in earth coordinates in HEADFREE mode, then in pidLevel it is possible to take the necessary data from rMat:

STATIC_UNIT_TESTED float pidLevel(int axis, const pidProfile_t *pidProfile, const rollAndPitchTrims_t *angleTrim, float currentPidSetpoint) {
…

    const float errorAngle = angle - ((attitude.raw[axis] - angleTrim->raw[axis]) / 10.0f);

//May be add here something like this ---------------------------------------------

if(FLIGHT_MODE(HEADFREE_MODE)){
if(axis == FD_ROLL){errorAngle = angle – ((lrintf(atan2_approx(rMat[2][1], rMat[2][2]) * (1800.0f / M_PIf)) - angleTrim->raw[axis])/10.0f);}
if(axis == FD_PITCH){errorAngle = angle – ((lrintf(((0.5f * M_PIf) - acos_approx(-rMat[2][0])) * (1800.0f / M_PIf)) - angleTrim->raw[axis])/10.0f);}
if(axis == FD_YAW){errorAngle = angle – ((lrintf((-atan2_approx(rMat[1][0], rMat[0][0]) * (1800.0f / M_PIf))) - angleTrim->raw[axis])/10.0f);}
}

//and it solves problem with HEADFREE ---------------------------------------------
…
   
}

haslinghuis, 2020-10-26:
Is there a `rmat` library for C ?
```
    angle = constrainf(angle, -pidProfile->levelAngleLimit, pidProfile->levelAngleLimit);
    float errorAngle = angle - ((attitude.raw[axis] - angleTrim->raw[axis]) / 10.0f);
    if (FLIGHT_MODE(HEADFREE_MODE)) {
        if (axis == FD_ROLL) {
            errorAngle = angle - ((lrintf(atan2_approx(rMat[2][1], rMat[2][2]) * (1800.0f / M_PIf)) - angleTrim->raw[axis]) / 10.0f);
        }
        if (axis == FD_PITCH) {
            errorAngle = angle - ((lrintf(((0.5f * M_PIf) - acos_approx(-rMat[2][0])) * (1800.0f / M_PIf)) - angleTrim->raw[axis]) / 10.0f);
        }
        if (axis == FD_YAW) {
            errorAngle = angle - ((lrintf((-atan2_approx(rMat[1][0], rMat[0][0]) * (1800.0f / M_PIf))) - angleTrim->raw[axis]) / 10.0f);
        }
    }
```

etracer65, 2020-10-26:
GPS Rescue flight control model doesn't support HEADFREE. In fact it would only have the ability to fly **away** from the home point (and then disarm due to sanity checks).

MichelArhipov, 2020-10-27:
I didn't notice it right away - there is a similar problem in setBearing in /betaflight/src/main/flight/gps_rescue.c:

static void setBearing(int16_t desiredHeading)
{
…
    float errorAngle = (attitude.values.yaw / 10.0f) - desiredHeading;

//And a similar solution ----------------------------------------------------------

if(FLIGHT_MODE(HEADFREE_MODE)){
errorAngle = (lrintf(-atan2_approx(rMat[1][0], rMat[0][0]) * (180.0f / M_PIf)))  - desiredHeading;
}

//now it will fly in the right direction to home in HEADFREE mode ----------------- 
…
}

etracer65, 2020-10-27:
No, the solution is to disable HEADFREE mode when GPS Rescue is active. Or better yet completely remove HEADFREE mode from Betaflight.

haslinghuis, 2020-10-28:
@MichelArhipov do you have some input about the missing `rmat` and a hint about implementing this?

MichelArhipov, 2020-10-28:
rMat[3][3] - this is a variable from /betaflight/src/main/flight/imu.c.
I didn't notice that this is not a public variable.

HEADFREE+ANGLE and HEADFREE+HORIZON has the same problem in pidLevel:

STATIC_UNIT_TESTED float pidLevel(int axis, const pidProfile_t *pidProfile, const rollAndPitchTrims_t *angleTrim, float currentPidSetpoint) {
…
Now here is angle in body coordinates, but attitude.raw in earth coordinates. 

    const float errorAngle = angle - ((attitude.raw[axis] - angleTrim->raw[axis]) / 10.0f);
…   
}

MichelArhipov, 2020-10-28:
To access static varable rMat[3][3], I added function float getrMatValue(int x, int y) {return rMat[x], [y]}
to imu.c and replaced rMat[x],[y] with getrMatValue(x,y) in the code being added to pid.c and gps_rescue.c

STATIC_UNIT_TESTED float pidLevel(int axis, const pidProfile_t *pidProfile, const rollAndPitchTrims_t *angleTrim, float currentPidSetpoint) {
…
float errorAngle = angle - ((attitude.raw[axis] - angleTrim->raw[axis]) / 10.0f);

if(FLIGHT_MODE(HEADFREE_MODE)){
if(axis == FD_ROLL){errorAngle = angle - ((lrintf(atan2_approx(getrMatValue(2,1), getrMatValue(2,2)) * (1800.0f / M_PIf)) - angleTrim->raw[axis])/10.0f);}
if(axis == FD_PITCH){errorAngle = angle - ((lrintf(((0.5f * M_PIf) - acos_approx(-getrMatValue(2,0))) * (1800.0f / M_PIf)) - angleTrim->raw[axis])/10.0f);}
if(axis == FD_YAW){errorAngle = angle - ((lrintf((-atan2_approx(getrMatValue(1,0), getrMatValue(0,0)) * (1800.0f / M_PIf))) - angleTrim->raw[axis])/10.0f);}
}
…   
}
static void setBearing(int16_t desiredHeading)
{
…
    float errorAngle = (attitude.values.yaw / 10.0f) - desiredHeading;

if(FLIGHT_MODE(HEADFREE_MODE)){
errorAngle = (lrintf(-atan2_approx(getrMatValue(1,0), getrMatValue(0,0)) * (180.0f / M_PIf)))  - desiredHeading;
    }
…
}

As a result, building succeed. Now HEADFREE+GPS_RESCUE, HEADFREE+ANGLE and HEADFREE+HORISON must work.

haslinghuis, 2020-10-29:
Thank you!!! `GPS_RESCUE` doesn't use `HEADFREE` mode anymore. This new PR should fix `HEADFREE+ANGLE` and `HEADFREE+HORIZON`. Could you please check the PR?

haslinghuis, 2020-10-29:
`make test` seems not happy with `getrMatValue`?

MichelArhipov, 2020-10-29:
I didn't specify that I added declaration float getrMatValue(int x, int y); to imu.h.
Without this declaration, the compiler issued a warning, but not an error.
I build code on local copy betaflight.

haslinghuis, 2020-10-29:
With declaration the target builds fine without warning. Without declaration it give warnings about implicit declaration and type mismatch. `make test` works in neither case, with or without declaration.

etracer65, 2020-10-29:
I'm not sure why people are throwing random code at something that doesn't seem to have any problems. Has anybody presented any evidence that there are any actual problems with HEADFREE outside fo the context of GPS Rescue? There is no need to make any modifications to `pidLevel()` in pid.c. The HEADFREE translation happens by mapping the control axes during RC processing and there is no need to alter the PID controller logic.

https://github.com/betaflight/betaflight/blob/d30f5104d0446dd3f36612e00fa94467d0f97941/src/main/fc/rc.c#L819-L836

MichelArhipov, 2020-10-29:
cCommandBuff. Z = 0; does not guarantee that the control object
will not be rotate around Z under external influence.
Any minor turn around Z will cause pidLevel to malfunction.
I suspect that cCommandBuff. Z = 0; just appeared due to pidLevel malfunction.
After making the suggested changes, we can delete cCommandBuff. Z = 0;

mikeller, 2020-10-29:
Agree with @etracer65. I think the approach to come up with a solution to a perceived problem in an entirely theoretical way is wrong here. If there is a problem with the HEADFREE mode what should be done first is flight tests to establish if the problem actually exists and has a negative impact. And only if this is the case a solution should be proposed, and verified to solve the problem in more flight tests.

mikeller, 2020-10-29:
@MichelArhipov: Have you done any flight tests with HEADFREE, and can you provide a description of what does not work when flying HEADFREE, and preferably some logs of the test flights?

MichelArhipov, 2020-10-29:
I was just about to conduct flight tests, but after reviewing the code and finding obvious errors, I postponed these tests.

closed 2020-10-29 by mikeller

haslinghuis, 2020-11-14:
@MichelArhipov are you still around? Did you have any time to conduct flight test with the new code in 10340? Please let's continue here instead of opening a new issue. But this kind of changes are very critical so expect resistance in acceptation.

MichelArhipov, 2020-11-15:
Hi! 
In the text 10340 I made a typo, mixed up + and - in the first line:

	const float x = (buffer.ww + buffer.xx /*There should be a minus here*/ - buffer.yy - buffer.zz) * v->X + 2.0f * (buffer.xy - buffer.wz) * v->Y + 2.0f * (buffer.xz + buffer.wy) * v->Z;
    	const float y = 2.0f * (buffer.xy + buffer.wz) * v->X + (buffer.ww - buffer.xx + buffer.yy - buffer.zz) * v->Y + 2.0f * (buffer.yz - buffer.wx) * v->Z;
    	const float z = 2.0f * (buffer.xz - buffer.wy) * v->X + 2.0f * (buffer.yz + buffer.wx) * v->Y + (buffer.ww - buffer.xx - buffer.yy + buffer.zz) * v->Z;

But the code I checked on Gozebo10 doesn't have this error.

If you leave this code, then HEADFREE+ANGLE will not work without changes in pidLevel.
But since the changes in pidLevel are not approved, I deleted this code.

/*-------------------------------------------------------------------------------------- 

if (FLIGHT_MODE(HEADFREE_MODE)) {
imuQuaternionComputeProducts(&headfree, &buffer);

   attitude.values.roll = lrintf(atan2_approx((+2.0f * (buffer.wx + buffer.yz)), (+1.0f - 2.0f * (buffer.xx + buffer.yy))) * (1800.0f / M_PIf));
   attitude.values.pitch = lrintf(((0.5f * M_PIf) - acos_approx(+2.0f * (buffer.wy - buffer.xz))) * (1800.0f / M_PIf));
   attitude.values.yaw = lrintf((-atan2_approx((+2.0f * (buffer.wz + buffer.xy)), (+1.0f - 2.0f * (buffer.yy + buffer.zz))) * (1800.0f / M_PIf)));
} else {

----------------------------------------------------------------------------------------*/

   attitude.values.roll = lrintf(atan2_approx(rMat[2][1], rMat[2][2]) * (1800.0f / M_PIf));
   attitude.values.pitch = lrintf(((0.5f * M_PIf) - acos_approx(-rMat[2][0])) * (1800.0f / M_PIf));
   attitude.values.yaw = lrintf((-atan2_approx(rMat[1][0], rMat[0][0]) * (1800.0f / M_PIf)));

// }
…

}
After I tested HEADFREE+HORIZON, I came to the conclusion that this code in rc.c is not superfluous.
It is not necessary to convert the rotation around Z at large angles of roll and pitch.

etracer65, 2020-11-15:
Again, outside the context of GPS Rescue (which doesn't support HEADFREE) nobody has actually explained or demonstrated any problem. Changing the code based on something that "you think is incorrect" is not a good way forward. The HEADFREE code has been the codebase nearly forever - probably dating back to original MultiWii. The fact that **nobody** has ever reported any problems with it in normal use makes me extremely dubious of these claims. Testing in a simulator is not actual flight testing. Please demonstrate an actual flight problem along with supporting documentation like a blackbox log.

mikeller, 2020-11-15:
@MichelArhipov: If you would like to continue this discussion please provide the data you were asked to provide in https://github.com/betaflight/betaflight/issues/10296#issuecomment-718828263.

MichelArhipov, 2020-11-17:
HEADFREE stopped working after commits on 23 Sep 2017 https://github.com/betaflight/betaflight/commit/7146c40ca85110b2b0c9a328ff682bb16d99ce0c#diff-5b076361907539df290ef01d9e0151a0bf6e389c842b3ca24212453ccfa42445
 
In the branch Sep 22 2017 https://github.com/betaflight/betaflight/tree/79651ec6c83f96f222a0ca77679bb1b1fd9c1511
the imuUpdateEulerAngles function does not contain redundant code.

STATIC_UNIT_TESTED void imuUpdateEulerAngles(void)
{
    /* Compute pitch/roll angles */
    attitude.values.roll = lrintf(atan2f(rMat[2][1], rMat[2][2]) * (1800.0f / M_PIf));
    attitude.values.pitch = lrintf(((0.5f * M_PIf) - acosf(-rMat[2][0])) * (1800.0f / M_PIf));
    attitude.values.yaw = lrintf((-atan2f(rMat[1][0], rMat[0][0]) * (1800.0f / M_PIf) + magneticDeclination));

    if (attitude.values.yaw < 0)
        attitude.values.yaw += 3600;

    /* Update small angle state */
    if (rMat[2][2] > smallAngleCosZ) {
        ENABLE_STATE(SMALL_ANGLE);
    } else {
        DISABLE_STATE(SMALL_ANGLE);
    }
}

The updateRcCommands function contains a simple EarthToBody conversion formula that works perfectly.

void updateRcCommands(void)
{
...

if (FLIGHT_MODE(HEADFREE_MODE)) {
        const float radDiff = degreesToRadians(DECIDEGREES_TO_DEGREES(attitude.values.yaw) - headFreeModeHold);
        const float cosDiff = cos_approx(radDiff);
        const float sinDiff = sin_approx(radDiff);
        const float rcCommand_PITCH = rcCommand[PITCH] * cosDiff + rcCommand[ROLL] * sinDiff;
        rcCommand[ROLL] = rcCommand[ROLL] * cosDiff - rcCommand[PITCH] * sinDiff;
        rcCommand[PITCH] = rcCommand_PITCH;
    }
...
}

iNav still uses this code.

mikeller, 2020-11-17:
@MichelArhipov: Looks like the logs from your test flights went missing from your earlier post. Please add them and repost.

MichelArhipov, 2020-11-26:
The reason for my incorrect assumptions were bugs in SITL.
I found these bugs and fixed them (#10367).
I used SITL to investigate the HEADFREE + GPS_RESCUE problem
that I encountered in a real flight.

Thank you all!

mikeller, 2020-11-26:
@MichelArhipov: 'Fixing' SITL by changing it to not be a representation of how the real flight controller code works is wrong.
If there is a problem manifesting in actual test flights, then this should be fixed and the fix verified in actual test flights. Making sure that SITL still works in the same way as the real flight controller code works is a second step after this

PR #10309 (2020-10-29): Disable HEADFREE flight mode when GPS Rescue is active
Fixes #10296

The GPS Rescue flight control model doesn't work in HEADFREE mode.