betaflight/betaflight #10299: Lack of throttle reaction after start

pmarkiewicz, 2020-10-23:
After start copter stopped reacting to throttle, it was stable, fully controllable (I did a flip) apart from throttle.
I believe it's dangerous situation.

Controller is Matek F405 STD, Esc BLHeli 32, frame is 3d printed (this may be a reason for vibrations)

Maybe it's hardware issue (FC was crashed in the past) or maybe it's due to excessive wibrations but I still believe it's a bug.
Log is here https://www.dropbox.com/s/igx8tkce0kjgd0h/LOG00002.BFL?dl=1

pmarkiewicz, 2020-10-23:
Not what is missing, issue is 100% possible  to reproduce, every start end up in same efect.

diff here
```
# 

# diff

# version
# Betaflight / STM32F405 (S405) 4.2.3 Sep 20 2020 / 20:42:51 (2696b7c88) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF405STD, version: 310c299f, date: 2020-05-22T01:08:06Z

# start the command batch
batch start

board_name MATEKF405STD
manufacturer_id MTKS

# name: PEON 230

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SERIAL_TX 11 A15
resource CAMERA_CONTROL 1 A08

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 4 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 1 1375 1625 0 0
aux 1 13 0 1750 1850 0 0
aux 2 35 0 1550 1650 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 500

# rxfail
rxfail 5 s 1500

# master
set acc_calibration = -40,133,-29,1
set min_check = 1025
set max_check = 1975
set rssi_channel = 8
set rc_smoothing_derivative_type = PT1
set fpv_mix_degrees = 30
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 5
set failsafe_throttle = 1350
set failsafe_procedure = AUTO-LAND
set vbat_max_cell_voltage = 435
set small_angle = 180
set pid_process_denom = 2
set pid_in_tlm = ON
set osd_cap_alarm = 1500
set osd_vbat_pos = 440
set osd_rssi_pos = 14352
set osd_link_quality_pos = 230
set osd_tim_2_pos = 14808
set osd_flymode_pos = 97
set osd_g_force_pos = 9
set osd_crosshairs_pos = 14541
set osd_current_pos = 6593
set osd_mah_drawn_pos = 2529
set osd_craft_name_pos = 6636
set osd_gps_speed_pos = 192
set osd_gps_lon_pos = 17
set osd_gps_lat_pos = 1
set osd_gps_sats_pos = 32
set osd_home_dir_pos = 47
set osd_home_dist_pos = 55
set osd_altitude_pos = 6359
set osd_avg_cell_voltage_pos = 14528
set osd_disarmed_pos = 14635
set osd_esc_rpm_pos = 161
set osd_rtc_date_time_pos = 480
set osd_flip_arrow_pos = 2415
set osd_stat_max_spd = OFF
set osd_stat_max_alt = ON
set osd_stat_max_g_force = ON
set debug_mode = GYRO_SCALED
set vtx_band = 4
set vtx_channel = 3
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5780
set vtx_pit_mode_freq = 5780
set vcd_video_system = PAL
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 700
set name = PEON 230

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 6
set vbat_sag_compensation = 100
set iterm_relax_cutoff = 20
set yaw_lowpass_hz = 100
set throttle_boost = 7
set throttle_boost_cutoff = 25
set ff_spike_limit = 70
set ff_smooth_factor = 20

rateprofile 0

# end the command batch
batch end

# 
```

haslinghuis, 2020-10-24:
> After start copter stopped reacting to throttle

Please describe better to understand what's happening.

pmarkiewicz, 2020-10-24:
Copter starts and after about 40% of input throttle motors spins with constant speed without any reaction to throttle stick.
With low throttle everything is ok, above some level motors are not more controlled by throttle stick.
It reacts to every other input apart from throttle.
You can observe this is attached log that throttle is set to 0 and copter is in flight.

bizmar, 2020-10-24:
There's severe P-D oscillation on roll. The default P-term to D-term ratio isn't suited for this quad. Reducing roll P by 10, should stop the over reaction. Gyro data is otherwise relatively low noise, so I don't think it's a hardware issue. 
[image]

pmarkiewicz, 2020-10-24:
I'll check this tomorrow, but this not explain lack of throttle reaction.

bizmar, 2020-10-25:
Look at the screenshot in my previous post. Lines at the too are roll pids, once a feedback loop is "triggered", pids react stronger and stronger, untill they max out. 

Motor command lines at the bottom are oscillating from 0% to 100% motor drive. Since the whole motor range is already "in use" throttle doesn't have any effect.

Disarming the quad is usually the only way to stop it from gaining altitude.

pmarkiewicz, 2020-11-01:
Thank you
ok, reducing helped.
All in all this was surprising reaction especially that everything worked apart from throttle.

closed 2020-11-01 by pmarkiewicz