betaflight/betaflight #10302: No stick input

tedwanday, 2020-10-24:
My receiver tab just stopped working, everything had been fine but a week ago it refused to arm, so I connected it to betaflight and reciever tab has no response at all. I've had a very frustrating week trying to figure it out but it turns out to be called 'no stick input issue'.
I have an xm+ and a mamba f405mini and q x7.
It was working well previously, so the issue is not that I wired it badly or that my serialrx is on the wrong UART. I have rewired the full board anyway to confirm. The xm+ is green so it is bound to the radio and I see rssi aux channel moving. All others dont move at all. The arming disable flags are constantly rx_failsafe and msp, msp is okay but the failsafe is due to this whole issue.
Here is my diff all:
Any help will be much appreciated.
-------------------------------------------------------------------------------------------------------------------------------
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.3 Sep 20 2020 / 20:42:51 (2696b7c88) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FURYF4OSD
manufacturer_id DIAT
mcu_id 003a003c374d500520373947
signature 

# feature
feature MOTOR_STOP

# beeper
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -BAT_CRIT_LOW
beeper -BAT_LOW

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 35 1 1700 2100 0 0

# rxfail
rxfail 3 h
rxfail 4 h
rxfail 5 h

# master
set acc_calibration = -69,2,214,1
set min_check = 1010
set max_check = 2000
set rssi_channel = 16
set rssi_src_frame_errors = OFF
set blackbox_device = NONE
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 370
set small_angle = 180
set gps_provider = UBLOX
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_vbat_pos = 14392
set osd_tim_1_pos = 14827
set osd_flymode_pos = 6188
set osd_throttle_pos = 8426
set osd_crosshairs_pos = 6349
set osd_current_pos = 14839
set osd_mah_drawn_pos = 12673
set osd_gps_speed_pos = 14784
set osd_gps_lon_pos = 8426
set osd_gps_lat_pos = 8426
set osd_gps_sats_pos = 14425
set osd_home_dir_pos = 12782
set osd_home_dist_pos = 14816
set osd_flip_arrow_pos = 14510
set gyro_1_align_yaw = 1800

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
------------------------------------------------------------------------------------------------------------------------------------
Any help will be much appreciated

mikeller, 2020-10-24:
@tedwanday:

> The xm+ is green so it is bound to the radio and I see rssi aux channel moving. 

On the Betaflight side there is no distinction between the RSSI AUX channel and all other channels - so if this channel is moving the signal for all channels is coming from the RX to Betaflight ok, and the problem you have is somewhere between the TX / firmware and the RX - check your OpenTX settings.

tedwanday, 2020-10-24:
The RSSI occassionally responds and also fails like the other channels.
I have open-tx version 2.3.9 and I haven't updated it so it had been working just fine, could something just have malfunctioned?

closed 2021-02-08 by github-actions[bot]