betaflight/betaflight #10312: Where can I download the firmware version of BTFL CC3D-RAID Betaflight 3.2. 5 Firmware mar-10-2018 14:10:41.?

38293312, 2020-10-28:
Where can I download the firmware version of BTFL CC3D-RAID Betaflight 3.2. 5 Firmware mar-10-2018 14:10:41.?

mikeller, 2020-10-28:
https://github.com/betaflight/betaflight/releases/tag/v3.2.5

38293312, 2020-10-28:
> https://github.com/betaflight/betaflight/releases/tag/v3.2.5

Thank you for your help, but the CC3D firmware of this release date cannot be checked for "GYROR" after it is burned in. I checked some information. My version of CC3D is "CC3D PRO raid" and uses MPU6500 sensor, which must be used Only firmware version 3.2.5 with a release date of "mer 10 2018" can be used.

etracer65, 2020-10-28:
The link provided is for the official Betaflight releases. Versions created by third-parties are not supported or available from Betaflight and we have no idea about where such a version is sourced.

38293312, 2020-10-28:
> The link provided is for the official Betaflight releases. Versions created by third-parties are not supported or available from Betaflight and we have no idea about where such a version is sourced.

Oh, it turned out to be like this, I always thought it was the official version, it turned out to be a third-party version. It seems that my CC3D flight controller can't be used anymore.

closed 2021-02-08 by github-actions[bot]

jlbfpv, 2021-03-29:
I wish I wouldve known this before buying 10 boards theese guys put out nothing but crap .seriously I understand you get what you pay for .25.00 each is decent for a cheesy fc but theese guys have bunked me for 6 months now and if I had a way to get to their shop in chi n a I dont want to say what this mother fucker would do .its like they do it on purpose how do they even get product out like this crap.I see they put on a quality control sticker on some of the fc 's but I realized later if you flip the qc sticker over it says SUCKER.ANYWAYthanks for the info .sorry for the rant.peace.china fc production needs to stop until wong quits drinking at lunch.its no good.fright cotrorer no woki.loljkjkjk.

jlbfpv, 2021-03-29:
And even if you do blow the dough on the call communication barrier.

datday10xx, 2021-12-06:
I don't know if this can help you? This is what you are looking for
http://www.5imx.com/forum.php?mod=viewthread&tid=1420139