betaflight/betaflight #10320: Has anyone noticed that Jenkins is no longer working?

TheIsotopes, 2020-11-01:
The Betaflight and BetaFlight_Configurator jobs have not been working for several days.

mikeller, 2020-11-01:
@TheIsotopes: Not me, thanks for bringing this up. Normally @wind0r looks after the Jenkings instance.

A quick check shows the following problems:
- Betaflight - too strict version requirements for the clang used for test builds - fixed in #10321;
- Betaflight configurator - breaks when it tries to build the Chromeos version for which support has been dropped - the Jenkins build configuration is not in the repository for what I can tell, so @wind0r will have to fix this.

wind0r, 2020-11-01:
@TheIsotopes thanks for informing us :) 

@mikeller thanks for the quick check. 

* removed Chromeos from the target list for configurator.
* updated Jenkins and its dependencies.
* replaced default clang (3.8.1) with clang 10 installed via Automatic installation script from https://apt.llvm.org/
* maybe #10321 should be closed
* one test build is currently running. Looks fine :)

closed 2020-11-01 by wind0r