betaflight/betaflight #10322: BF4.3 - on SPRACINGH7NANO esc's not working after flashing latest master

TheIsotopes, 2020-11-01:
**Describe the bug**
esc are not initialized when battery pluged in 
if fc are connected to the configurator esc's constant initialization when a tab is changed

imo #10247 makes this problems, @SteveCEvans pls can you check this

**To Reproduce**
flash latest master

**Flight controller configuration**
```
# Betaflight / SPRACINGH7NANO (SP7N) 4.3.0 Nov  1 2020 / 08:54:31 (49679ce0e) MSP API: 1.44

# start the command batch
batch start

board_name SPRACINGH7NANO

# name: XHOVER

# feature
feature -TELEMETRY
feature -RSSI_ADC
feature -OSD
feature -TRANSPONDER

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW

# map
map TAER1234

# led
led 0 9,2::AW:0
led 1 9,3::AW:0
led 2 9,4::AW:0
led 3 9,5::AW:0
led 4 6,5::AW:0
led 5 6,4::AW:0
led 6 6,3::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1600 2000 0 0
aux 1 13 1 1000 1200 0 0
aux 2 26 0 1800 2000 0 0
aux 3 31 0 1200 1400 0 0
aux 4 36 1 1800 2000 0 0

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_auto_smoothness = 7
set serialrx_provider = SUMD
set blackbox_sample_rate = 1/8
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1550
set use_cbat_alerts = ON
set cbat_alert_percent = 20
set vbat_duration_for_warning = 50
set vbat_duration_for_critical = 20
set ibata_scale = 179
set name = XHOVER

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 7
set iterm_relax_cutoff = 20
set throttle_boost = 7
set throttle_boost_cutoff = 25
set p_pitch = 32
set i_pitch = 63
set d_pitch = 27
set f_pitch = 67
set p_roll = 29
set i_roll = 59
set d_roll = 25
set f_roll = 63
set p_yaw = 31
set i_yaw = 63
set f_yaw = 63
set d_min_roll = 16
set d_min_pitch = 18

rateprofile 0

# rateprofile 0
set rates_type = QUICK
set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 130
set roll_srate = 126
set pitch_srate = 126
set yaw_srate = 99

# end the command batch
batch end
```

mikeller, 2020-11-01:
@TheIsotopes: Can you please provide the output of `resource show all`?

TheIsotopes, 2020-11-01:
@mikeller here the output from `resource show all`

```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: FREE
A05: FREE
A06: FREE
A07: FREE
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: FREE
B01: FREE
B02: QSPI_CLK 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: FREE
B06: I2C_SCL 4
B07: I2C_SDA 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: QSPI_BK2CS 1
B11: FREE
B12: GYRO_CS 1
B13: FREE
B14: FREE
B15: SERIAL_RX 1
C00: ADC_CURR
C01: ADC_BATT
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: FREE
C05: ADC_EXT
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: SPI_SCK 2
D04: FREE
D05: FREE
D06: SPI_MOSI 3
D07: BEEPER
D08: FREE
D09: FREE
D10: FREE
D11: QSPI_BK1IO0 1
D12: QSPI_BK1IO1 1
D13: QSPI_BK1IO3 1
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: QSPI_BK1IO2 1
E03: LED 1
E04: SYSTEM
E05: CAMERA_CONTROL
E06: FREE
E07: QSPI_BK2IO0 1
E08: QSPI_BK2IO1 1
E09: QSPI_BK2IO2 1
E10: QSPI_BK2IO3 1
E11: PREINIT 3
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: GYRO_EXTI
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15:
    CH1: CAMERA_CONTROL
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: DSHOT_BITBANG 1
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

SteveCEvans, 2020-11-01:
@TheIsotopes  Looking into this now. Current master, 49679ce, is working for me with the config below.

The obvious difference apart from the fact you'er using a NANO, not a ZERO, is that you're using the LED strip. When I enable LED_STRIP and  paste in the following fragment of your configuration, my motors still work.

```led 0 9,2::AW:0
led 1 9,3::AW:0
led 2 9,4::AW:0
led 3 9,5::AW:0
led 4 6,5::AW:0
led 5 6,4::AW:0
led 6 6,3::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 1 3
```

Could you please try disabling the LED_STRIP and trying again.

You appear not to be using logging at the moment either. I've disabled that and it still works. The NANO has additional baro drivers linked in compared to the ZERO. I've added those in to no effect either.

I've disabled the OSD too... still works.

My config (without the LEDs):

```# 

# diff all

# version
# Betaflight / SPRACINGH7ZERO (SP7Z) 4.3.0 Nov  1 2020 / 20:15:35 (49679ce) MSP API: 1.44

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name SPRACINGH7ZERO
mcu_id 0025001b3039510f38303435
signature 

# name: Marmotte

# feature
feature -RSSI_ADC
feature -LED_STRIP
feature -TRANSPONDER
feature -AIRMODE
feature GPS
feature ESC_SENSOR

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 7 2 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 950 1050 0 0
aux 1 1 3 1450 1550 0 0
aux 2 46 3 1950 2050 0 0
aux 3 13 2 1950 2050 0 0
aux 4 28 1 1950 2050 0 0
aux 5 35 2 1450 1550 0 0
[... 280 lines cut ...]
TIM14: FREE
TIM15:
    CH1: CAMERA_CONTROL
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: DSHOT_BITBANG 1
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

TheIsotopes, 2020-11-01:
after setting to defaults and minimalistic config

```
# version
# Betaflight / SPRACINGH7NANO (SP7N) 4.3.0 Nov  1 2020 / 15:57:55 (49679ce0e) MSP API: 1.44

# start the command batch
batch start

board_name SPRACINGH7NANO

# feature
feature -TELEMETRY
feature -RSSI_ADC
feature -LED_STRIP
feature -OSD
feature -TRANSPONDER

# master
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600

profile 0

rateprofile 0

# end the command batch
batch end
```
the problem is also not resolved

SteveCEvans, 2020-11-01:
I'll ask @Hydra to confirm and see if he has any thoughts. Failing that I'll ask him to send me one to test. I'll ask him to check and EXTREME too.

TheIsotopes, 2020-11-02:
@SteveCEvans pls see ny comments in #10247

klutvott123, 2020-11-02:
Probably need to do some cache invalidation now that the bitbang dshot buffers are no longer in ```DMA_RAM```? We have had issues like this before.

TheIsotopes, 2020-11-21:
@SteveCEvans are there any news?

SteveCEvans, 2020-11-21:
Just received a 100MHz logic analyser, and the F7NANO is now on it's ways, so things are coming together. iFlight H743 AIO behaved differently to H7ZERO so I have a difference to chase. Need to full understand DSHOT, so some learning to do.

SteveCEvans, 2020-11-22:
Just spotted that I wasn't cleaning/invalidating the correct DSHOT buffer size. Fixed in my h7mscsrc branch. Not tested on an H7NANO yet, but might be worth a quick try.

TheIsotopes, 2020-11-22:
@SteveCEvans ok, i have tested latest master
and included https://github.com/SteveCEvans/betaflight/commit/c63447e7e719e7d2b357524bdfa1f895d6d8cc68 ; https://github.com/SteveCEvans/betaflight/commit/4b77eb8dafd487ff54a295a31492653e7d75d313 ; https://github.com/SteveCEvans/betaflight/commit/0351b78b131e810c660b8a61bc732eff094572a0
seems like everything is going normally. all motors turn without problems and telemetry is also displayed correctly.

SteveCEvans, 2020-11-22:
Thank you @TheIsotopes. Sorry it took me a while to spot what I'd done wrong. I'm going to refactor the code to make the buffer sizes clearer; no changes in functionality, but hopefully more maintainable in future.

TheIsotopes, 2020-11-22:
now i tested #10359 
as before i couldn't find any problems here ... motors and telemetry works correctly.

thx for fixing this @SteveCEvans

SteveCEvans, 2020-11-22:
Thanks @TheIsotopes Just push a fix tweak, making cache management dependent on USE_DSHOT_CACHE_MGMT rather than processor type. I think that's rather tidier. In the process of measuring the speed impact of this.. +ve or -ve.

TheIsotopes, 2020-11-22:
I just saw. will test it again right away.

SteveCEvans, 2020-11-22:
Without cache clean/invalidate, although not working, so not convinced this is in shareable memory.

[image]

With cache clean/invalidate (USE_DSHOT_CACHE_MGMT defined):

[image]

Not sure about the first set of timings; looking into this.

SteveCEvans, 2020-11-22:
Looking in more detail at the option with USE_DSHOT_CACHE_MGMT defined, the clean/invalidate operations account for only ~1us total, so that's not the issue.

[image]

SteveCEvans, 2020-11-22:
This is probably slow because I don't have an ESC connected, so no returning telemetry. The time is being spent looking for a leading zero on a data packet which of course never comes. The code in `decode_bb()` uses `__builtin_expect` to try to optimise things; I replaced it with a simple loop out of curiosity and it makes no difference whatsoever, so I suggest such optimisations be removed as they simply reduce clarity.

mikeller, 2020-11-22:
@SteveCEvans:

> The code in `decode_bb()` uses `__builtin_expect` to try to optimise things; I replaced it with a simple loop out of curiosity and it makes no difference whatsoever, so I suggest such optimisations be removed as they simply reduce clarity.

One thing to keep in mind here is that this code is used on a number of different MCU types starting with STM32F411. So any statement like 'makes no difference whatsoever' will have to be validated against all supported MCU types.

SteveCEvans, 2020-11-23:
Fair comment. Had a quick look at the F411 and it makes no difference there either, however that's not a focus for now...

closed 2020-11-24 by mikeller