betaflight/betaflight #10323: Two UARTS for runcam Hybrid

mfulz, 2020-11-02:
**Is your feature request related to a problem? Please describe.**
Yes it is related to the following issue:
The runcam [hybrid](https://www.runcam.com/download/Hybrid/RC_Hybrid_Manual_EN.pdf) device has two control channels (two uart connections) for controlling either the HD cam via cam buttons (inside the mode configuration) and the other one for the joystick control to change the FPV cam.

For now I've connected both the inputs to separate UARTs and has to switch the uart in the ports tab to be able to control both the cams.

**Describe the solution you'd like**
Implementing a seperation inside the ports tab (bf configurator / bf cli):
RunCam Device Hybrid HD -> for sending the button control inputs via mode
RunCam Device -> for sending the stick control inputs

**Describe alternatives you've considered**
An alternative solution could be to either control both cams (sending both input methods buttons and stick input) simultanously and select two time the runcam device inside the ports tab.

**Additional context**
I could code the patches for bf I think, if someone could point me in the corrections where I would need to search for these functions, etc. as I'm able to program, but just don't have the time to dig deeply into the code to locate every part on my own.

etracer65, 2020-11-02:
Duplicate of #9299.

mfulz, 2020-11-02:
Sorry I forgot to look into closed issues.
Are there any possibilites to assist (very briefly) in pointing me into the right directions to locate the files / code where I've to look into for this rewrite?
I'd really like to help out here.

mikeller, 2020-12-25:
@mfulz: Probably best to hit up @azolyoung, who is the developer who implemented the RunCam integration, and coordinate your help with him.

closed 2021-02-15 by github-actions[bot]