betaflight/betaflight #10330: CPU usage doubles on Nox F411 boards after build 2326

ctzsnooze, 2020-11-07:
On my Nox boards my CPU usage suddenly doubles when I flash build 2336 or higher, compared to the previous successful build, 2326.  

l overclocked all the F411 boards to 120Mhz and always have.  

This is a new development.  CPU usage hasn't been an issue for a long time on these boards.

It doesn't seem to matter if I use the custom Nox build or the unified target.

Build 2326 has CPU at 5-6% going to only 9-10% when motors are running rpm with DShot telemetry, it just about doubles in 2336 with no change to settings.

Strangely the NoxeV1 board with FlywooF411 isn't affected.

Can't really figure it out, there were a few commits in the middle, but not sure how any could have caused this problem.

ctzsnooze, 2020-11-07:
@SteveCEvans by stepping backwards through master, it appears to have started at [PR10303](https://github.com/betaflight/betaflight/pull/10303)

SteveCEvans, 2020-11-07:
@ctzsnooze The only F411 FC I have is the REVONANO.

With master (6f4d179) and no overclocking I see 4-5% CPU load reported in the configurator, and a `tasks` dump gives:

```# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms
00 - (         SYSTEM)     10       2       1    0.5%    0.5%         0
01 - (         SYSTEM)    996       8       1    1.2%    0.5%        23
02 - (           GYRO)   8000      21      10   17.3%    8.5%      3100
03 - (         FILTER)   4000      25      15   10.5%    6.5%      2488
04 - (            PID)   4000      53      35   21.7%   14.5%      5858
05 - (            ACC)    996      18       9    2.2%    1.3%       395
06 - (       ATTITUDE)    100      13      10    0.6%    0.6%        42
07 - (             RX)     33      82      47    0.7%    0.6%        65
08 - (         SERIAL)    100  473018       2 4730.6%    0.5%       887
09 - (       DISPATCH)    997       9       1    1.3%    0.5%        32
11 - (BATTERY_CURRENT)     50       2       1    0.5%    0.5%         1
12 - ( BATTERY_ALERTS)      5       5       2    0.5%    0.5%         0
13 - (         BEEPER)    100       6       1    0.5%    0.5%         5
15 - (           BARO)    186      13       4    0.7%    0.5%        27
16 - (       ALTITUDE)     40      17       9    0.5%    0.5%        15
21 - (            CMS)     20       9       2    0.5%    0.5%         1
22 - (        VTXCTRL)      5       2       1    0.5%    0.5%         0
23 - (        CAMCTRL)      5       2       1    0.5%    0.5%         0
25 - (    ADCINTERNAL)      1       3       2    0.5%    0.5%         0
RX Check Function                   5       2                         2
Total (excluding SERIAL)                        60.7%   38.0%
```

Could you please post tasks dumps for immediately before/after #10303 to see if there's a task load that'll give us a clue.

What SPI devices are there on this FC?

SteveCEvans, 2020-11-08:
I note that the NOX has the gyro, baro and MAX7456 all on the same SPI bus (2), with the flash on SPI bus 1.

The only obvious difference I can see is that the MAX7456 driver will return the SPI clock speed to ~1MHz as the bus is shared, whereas previously it would have been SPI_CLOCK_FAST, or approx. 21MHz. That's quicker than the baro is meant to run.

Could you please try changing the definition of MAX7456_MAX_SPI_SHARED_CLK in src//main/drivers/max7456.c to 21000000 and see what that does.

Thanks

ctzsnooze, 2020-11-08:
Thanks - you're spot on!  That fixes it!
The Nox has a baro module, but no mag.  Both are disabled in the CLI with
```set mag_hardware = NONE
set baro_hardware = NONE
```

The following are with the Nox overclocked to 120Mhz.  The code is in full flight configuration but motors are not armed.
Current master shows 10-11% CPU.  Tasks:
```# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms
00 - (         SYSTEM)      9       3       1    0.5%    0.5%         0
01 - (         SYSTEM)    985       4       0    0.8%    0.0%        16
02 - (           GYRO)   8000      49      38   39.7%   30.9%      8480
03 - (         FILTER)   4000      35      26   14.5%   10.9%      3032
04 - (            PID)   4000     102      78   41.3%   31.7%      9346
07 - (             RX)     32      84      47    0.7%    0.6%        45
08 - (         SERIAL)    100  505256       2 5053.0%    0.5%       956
09 - (       DISPATCH)    986       6       1    1.0%    0.5%        21
10 - (BATTERY_VOLTAGE)    199       6       2    0.6%    0.5%        13
12 - ( BATTERY_ALERTS)      5       3       2    0.5%    0.5%         0
13 - (         BEEPER)    100       5       1    0.5%    0.5%         3
17 - (      TELEMETRY)    249      53       1    1.8%    0.5%        39
20 - (            OSD)     60     562      29    3.8%    0.6%        51
22 - (            CMS)     19       4       2    0.5%    0.5%         0
23 - (        VTXCTRL)      5       6       1    0.5%    0.5%         0
24 - (        CAMCTRL)      5       3       1    0.5%    0.5%         0
26 - (    ADCINTERNAL)      2       3       1    0.5%    0.5%         0
RX Check Function                   5       2                         1
Total (excluding SERIAL)                       107.7%   79.7%```

Modified master, with max7456 to 21000000 - 5% CPU - like builds before #10303.  Tasks:

```Task list             rate/hz  max/us  avg/us maxload avgload  total/ms
00 - (         SYSTEM)     10       7       1    0.5%    0.5%         0
01 - (         SYSTEM)    988       4       1    0.8%    0.5%        21
02 - (           GYRO)   8000      23       9   18.9%    7.7%      2766
03 - (         FILTER)   4000      40      26   16.5%   10.9%      4111
04 - (            PID)   4000     110      78   44.5%   31.7%     12511
07 - (             RX)     33      80      46    0.7%    0.6%        60
08 - (         SERIAL)    100  535234       1 5352.8%    0.5%       996
09 - (       DISPATCH)    991       9       1    1.3%    0.5%        29
10 - (BATTERY_VOLTAGE)    200      11       3    0.7%    0.5%        17
12 - ( BATTERY_ALERTS)      5       3       2    0.5%    0.5%         0
13 - (         BEEPER)    100       7       2    0.5%    0.5%         4
17 - (      TELEMETRY)    249      53       1    1.8%    0.5%        44
20 - (            OSD)     60     562      31    3.8%    0.6%        69
22 - (            CMS)     20       4       1    0.5%    0.5%         1
23 - (        VTXCTRL)      5       3       1    0.5%    0.5%         0
24 - (        CAMCTRL)      5       3       0    0.5%    0.0%         0
26 - (    ADCINTERNAL)      1       4       2    0.5%    0.5%         0
RX Check Function                   4       2                         2
Total (excluding SERIAL)                        92.5%   56.5%```

mikeller, 2020-11-08:
Good findings @SteveCEvans and @ctzsnooze.
It looks like resetting the SPI clock to 1 MHz at the end of a MAX7456 transaction is too conservative - the value before #10303 was ~20 MHz. But all in all I think resetting the SPI bus clock at the end of the MAX7456 transaction is a 'workaround' for specific designs that should not be there in the first place.

The proper solution for the 'workaround' of unilaterally switching SPI clock speeds (that is only done in the MAX7456 driver) is to start using SPI transactions (#7424). We've had them for a while, and we really should start using them as they solve exactly this problem in a proper way.

@ctzsnooze: Can you please try if the test targets in #10333 fixes this problem?

ctzsnooze, 2020-11-08:
Thanks, Mike.
With the F411 test target in #10333, the CPU runs a tiny bit higher, at about 6%, with this result in Tasks:
```
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms
00 - (         SYSTEM)      9       3       1    0.5%    0.5%         0
01 - (         SYSTEM)    994       9       1    1.3%    0.5%        17
02 - (           GYRO)   8000      31      15   25.3%   12.5%      2220
03 - (         FILTER)   4000      40      26   16.5%   10.9%      2048
04 - (            PID)   4000     103      78   41.7%   31.7%      6287
07 - (             RX)     31      93      49    0.7%    0.6%        31
08 - (         SERIAL)    100  599996      11 6000.4%    0.6%      1099
09 - (       DISPATCH)    990       8       0    1.2%    0.0%        14
10 - (BATTERY_VOLTAGE)    200       6       2    0.6%    0.5%         9
12 - ( BATTERY_ALERTS)      5       3       1    0.5%    0.5%         0
13 - (         BEEPER)    100       8       1    0.5%    0.5%         2
20 - (      TELEMETRY)    249      64       1    2.0%    0.5%        27
23 - (            OSD)     59     555      28    3.7%    0.6%        33
25 - (            CMS)     18       4       1    0.5%    0.5%         0
26 - (        VTXCTRL)      5      34      16    0.5%    0.5%         2
27 - (        CAMCTRL)      5       3       1    0.5%    0.5%         0
29 - (    ADCINTERNAL)      2       4       1    0.5%    0.5%         0
RX Check Function                   4       2                         1
Total (excluding SERIAL)                        96.5%   61.3%
```

SteveCEvans, 2020-11-08:
@mikeller The SPI DMA code I mentioned to you handles this already. Not ready for a PR yet as I need to add H7 support. I think this may be worth pursuing.

mikeller, 2020-11-08:
@SteveCEvans: I think the introduction of SPI transactions is definitely worth pursuing - this has been planned / ready with limited testing for some time, and seeing the current problem I think it is time to pull the trigger on it now.

SteveCEvans, 2020-11-08:
The current issue could easily be fixed by allowing targets to override a `SHARED_SPI_CLK` definition used instead of the current value applied by the MAX7456 driver.

The pursuit I was referring to was my DMA SPI bus driver!

mikeller, 2020-11-08:
@SteveCEvans:

> The current issue could easily be fixed by allowing targets to override a SHARED_SPI_CLK definition used instead of the current value applied by the MAX7456 driver.

This sounds like a pretty bad workaround for something that is done wrong in the first place - the MAX7456 driver should not attempt to set the SPI bus speed in lieu of other devices using it.

> The pursuit I was referring to was my DMA SPI bus driver!

We can use SPI transactions as a stepping stone on the way to this.

SteveCEvans, 2020-11-08:
👍🏻

SteveCEvans, 2020-11-10:
@mikeller Please see https://github.com/SteveCEvans/betaflight/tree/bb_dma for a preview of the DMA SPI bus support. No H7 support yet (my next task), but F4 and F7 working fine. Key changes are that there is now a distinction between a bus device driver (represented by a busDevice_t structure) and an external device (represented by a extDevice_t structure) of which many may be associated with a given bus driver. Polled operation (faster than the current driver) is used for small transfers where the overhead of a DMA is too much, and DMAs are automatically used where appropriate. All device initialisation is done is polled mode, and then DMA descriptors are assigned for DMA operation if not already taken. The SPI API has also been updated. Seems as good a place to mention this as any. I'll raise a PR once H7 is supported and I've tidied up a bit.

PR #10333 (2020-11-22): Enabled SPI Transaction support for F4 and F7.
Fixes #10330. 

Brought on by changes in #10303, which exposed an ugly 'workaround' for setting the SPI clock to a constant speed after every MAX7456 transaction, discussed in #10330. 

@jflyper, do you see any reason to not do this? I think we are better off to enable this now and fix any remaining bugs, instead of tweaking 'workarounds' like the one exposed in #10330. 

Test targets:

[betaflight_4.3.0_STM32F745_266264262.zip](https://github.com/betaflight/betaflight/files/5506232/betaflight_4.3.0_STM32F745_266264262.zip)
[betaflight_4.3.0_STM32 [... cut]
closed 2020-11-22 by mikeller