betaflight/betaflight #10332: GHST driver causes high CPU load as new frame types are introduced

tonycake, 2020-11-07:
Beta builds of the Ghost firmware, which add new frame types, not yet recognized by the Betaflight driver, cause high CPU load. 

In ghstProcessFrame, frame types GHST_UL_RC_CHANS_HS4_5TO8, GHST_UL_RC_CHANS_HS4_9TO12, GHST_UL_RC_CHANS_HS4_13TO16 are correctly supported, and new frames should drop through the default: case. 

The default case causes a false value to be returned from ghstProcessFrame, which appears to be incorrect, and causes continuous callbacks. 

Proposal is to return true in the default case to silently ignore new frames which will be added to the Ghost firmware in the future.

mikeller, 2020-11-07:
@tonycake:

> Proposal is to return true in the default case to silently ignore new frames which will be added to the Ghost firmware in the future.

Yes, that sounds about right. For what I can tell there is no case (yet) where the `rcProcessFrameFn` function should return `false` in the GHST implementation.

mikeller, 2020-12-25:
Fixed by https://github.com/betaflight/betaflight/pull/10360.

closed 2021-02-15 by github-actions[bot]