betaflight/betaflight #10335: GHST Protocol No TM Stream

bobbycisneros, 2020-11-09:
**Describe the bug**
There is no TM stream when using the GHST protocol

I switched to the IRC Ghost protocol and rebound my Rx with the same.  TX/RX connected well and the test flight was a success; however, there is no TM stream coming down from the drone.  

**Flight controller configuration**
```
# 

# diff all

# version
# Betaflight / DALRCF722DUAL (DLF7) 4.3.0 Nov  7 2020 / 17:05:47 (norevision) MSP API: 1.44

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name DALRCF722DUAL
manufacturer_id DALR
mcu_id 004b00383338510339393832
signature 

# name: Apex2

# feature
feature GPS
feature TELEMETRY
feature LED_STRIP

# serial
serial 3 2 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# beeper
beeper -RX_SET
beeper -ON_USB

# beacon
beacon RX_LOST

# map
map TAER1234

# aux
aux 0 0 0 900 1250 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 900 1300 0 0
aux 3 46 3 900 1200 0 0
aux 4 13 3 900 1700 0 0
aux 5 35 2 1475 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set gyro_lowpass2_hz = 325
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -53,-21,-38,1
set align_mag = CW270FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 2700
set mag_hardware = NONE
set mag_calibration = 46,-75,-21
set baro_hardware = NONE
set serialrx_provider = SRXL2
set serialrx_halfduplex = ON
set min_throttle = 1070
set dshot_idle_value = 500
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set failsafe_delay = 8
set failsafe_procedure = GPS-RESCUE
set vbat_scale = 162
set ibata_scale = 140
set ibata_offset = -500
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_initial_alt = 75
set gps_rescue_descent_dist = 120
set gps_rescue_ascend_rate = 200
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set osd_units = IMPERIAL
set osd_warn_link_quality = ON
set osd_alt_alarm = 120
set osd_vbat_pos = 2497
set osd_rssi_pos = 2474
set osd_link_quality_pos = 2479
set osd_tim_1_pos = 2518
set osd_tim_2_pos = 2486
set osd_remaining_time_estimate_pos = 152
set osd_flymode_pos = 2105
set osd_throttle_pos = 2305
set osd_vtx_channel_pos = 33
set osd_crosshairs_pos = 2285
set osd_ah_pos = 2158
set osd_current_pos = 2465
set osd_mah_drawn_pos = 471
set osd_craft_name_pos = 2529
set osd_gps_speed_pos = 2169
set osd_gps_lon_pos = 2113
set osd_gps_lat_pos = 2081
set osd_gps_sats_pos = 2145
set osd_home_dir_pos = 2140
set osd_home_dist_pos = 2201
set osd_compass_bar_pos = 47
set osd_altitude_pos = 2231
set osd_warnings_pos = 14729
set osd_battery_usage_pos = 417
set osd_disarmed_pos = 2538
set osd_nheading_pos = 90
set osd_core_temp_pos = 417
set osd_stat_endbatt = ON
set osd_stat_max_alt = ON
set vtx_band = 2
set vtx_channel = 1
set vtx_power = 4
set vtx_freq = 5733
set gyro_rpm_notch_harmonics = 1
set name = Apex2

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set feedforward_transition = 80
set iterm_relax = RPY
set p_pitch = 53
set i_pitch = 59
set d_pitch = 35
set f_pitch = 90
set p_roll = 53
set i_roll = 59
set p_yaw = 29
set i_yaw = 63
set d_min_roll = 22

profile 1

# profile 1
set feedforward_transition = 80
set iterm_relax = RPY
set p_pitch = 53
set i_pitch = 59
set d_pitch = 45
set f_pitch = 90
set p_roll = 53
set i_roll = 59
set d_roll = 45
set p_yaw = 29
set i_yaw = 63
set d_min_roll = 22

profile 2

# profile 2
set p_pitch = 72
set i_pitch = 50
set d_pitch = 25
set f_pitch = 90
set p_roll = 70
set i_roll = 50
set d_roll = 25
set p_yaw = 82
set i_yaw = 50
set d_min_pitch = 0

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 12
set thr_expo = 50
set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 150
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 59
set pitch_srate = 59
set yaw_srate = 59
set tpa_rate = 35

rateprofile 1

# rateprofile 1
set thr_mid = 12
set thr_expo = 50
set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 150
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 59
set pitch_srate = 59
set yaw_srate = 59
set tpa_rate = 35

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 2
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: LED_STRIP
B00: GYRO_CS 1
B01: CAMERA_CONTROL
B02: FLASH_CS
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: GYRO_EXTI
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_CURR
C01: ADC_BATT
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: LED_STRIP
TIM3:
    CH4: CAMERA_CONTROL
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: LED_STRIP
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```
**Setup / Versions**
DALRCF722DUAL
RX: IRC Atto
VTX: Lumenier Lux

tonycake, 2020-11-09:
Coming in ghost v1010, expected this week.

tonycake, 2020-12-04:
Fixed in #10360, with Ghost firmware v1.0.1.2

bobbycisneros, 2020-12-04:
Tested the fix on the bench. Open flight would not discover any TM data. Is there a new version of open flight needed that complements the TM fix?

closed 2021-02-15 by github-actions[bot]