betaflight/betaflight #10340: HEADFREE fix

MichelArhipov, 2020-11-11:
The imuQuaternionHeadfreeTransformVectorearthtobody function in /betaflight/src/main/flight/rc.c uses
a formula from classical mathematics, where the positive direction of the rotation angle along the axis is counterclockwise.
But betaflight uses the positive direction of the rotation angle along the axis clockwise. I rewrote the formula.

void imuQuaternionHeadfreeTransformVectorEarthToBody(t_fp_vector_def *v)
{
 …

/* Classical formula -----------------------------------------------
    const float x = (buffer.ww + buffer.xx - buffer.yy - buffer.zz) * v->X + 2.0f * (buffer.xy + buffer.wz) * v->Y + 2.0f * (buffer.xz - buffer.wy) * v->Z;
    const float y = 2.0f * (buffer.xy - buffer.wz) * v->X + (buffer.ww - buffer.xx + buffer.yy - buffer.zz) * v->Y + 2.0f * (buffer.yz + buffer.wx) * v->Z;
    const float z = 2.0f * (buffer.xz + buffer.wy) * v->X + 2.0f * (buffer.yz - buffer.wx) * v->Y + (buffer.ww - buffer.xx - buffer.yy + buffer.zz) * v->Z;
----------------------------------------------------------------------*/

    const float x = (buffer.ww + buffer.xx + buffer.yy - buffer.zz) * v->X + 2.0f * (buffer.xy - buffer.wz) * v->Y + 2.0f * (buffer.xz + buffer.wy) * v->Z;
    const float y = 2.0f * (buffer.xy + buffer.wz) * v->X + (buffer.ww - buffer.xx + buffer.yy - buffer.zz) * v->Y + 2.0f * (buffer.yz - buffer.wx) * v->Z;
    const float z = 2.0f * (buffer.xz - buffer.wy) * v->X + 2.0f * (buffer.yz + buffer.wx) * v->Y + (buffer.ww - buffer.xx - buffer.yy + buffer.zz) * v->Z;

…

}

Since no one answered me, why add the headfree offset to attitude.values in /betaflight/src/main/flight/imu. c,
which causes pidLevel malfunction in /betaflight/src/main/flight/pid. c, (Issue #10296) I removed this conversion:

STATIC_UNIT_TESTED void imuUpdateEulerAngles(void)
{
…
/*
    if (FLIGHT_MODE(HEADFREE_MODE)) {
       imuQuaternionComputeProducts(&headfree, &buffer);

       attitude.values.roll = lrintf(atan2_approx((+2.0f * (buffer.wx + buffer.yz)), (+1.0f - 2.0f * (buffer.xx + buffer.yy))) * (1800.0f / M_PIf));
       attitude.values.pitch = lrintf(((0.5f * M_PIf) - acos_approx(+2.0f * (buffer.wy - buffer.xz))) * (1800.0f / M_PIf));
       attitude.values.yaw = lrintf((-atan2_approx((+2.0f * (buffer.wz + buffer.xy)), (+1.0f - 2.0f * (buffer.yy + buffer.zz))) * (1800.0f / M_PIf)));
    } else {
*/

       attitude.values.roll = lrintf(atan2_approx(rMat[2][1], rMat[2][2]) * (1800.0f / M_PIf));
       attitude.values.pitch = lrintf(((0.5f * M_PIf) - acos_approx(-rMat[2][0])) * (1800.0f / M_PIf));
       attitude.values.yaw = lrintf((-atan2_approx(rMat[1][0], rMat[0][0]) * (1800.0f / M_PIf)));

 //   }
…
       
}

After changes in imu.c no need to make changes to pidLevel and the following code was superfluous in / betaflight/src/main/flight/rc. c:

FAST_CODE FAST_CODE_NOINLINE void updateRcCommands(void)
{
…
if (FLIGHT_MODE(HEADFREE_MODE)) {
        static t_fp_vector_def  rcCommandBuff;

        rcCommandBuff.X = rcCommand[ROLL];
        rcCommandBuff.Y = rcCommand[PITCH];
/*--------------------------------------------------------
        if ((!FLIGHT_MODE(ANGLE_MODE) && (!FLIGHT_MODE(HORIZON_MODE)) && (!FLIGHT_MODE(GPS_RESCUE_MODE)))) {
----------------------------------------------------------*/

            rcCommandBuff.Z = rcCommand[YAW];
/*--------------------------------------------------------
        } else {
            rcCommandBuff.Z = 0;
        }
----------------------------------------------------------*/

        imuQuaternionHeadfreeTransformVectorEarthToBody(&rcCommandBuff);
        rcCommand[ROLL] = rcCommandBuff.X;
        rcCommand[PITCH] = rcCommandBuff.Y;

/*----------------------------------------------------------
        if ((!FLIGHT_MODE(ANGLE_MODE)&&(!FLIGHT_MODE(HORIZON_MODE)) && (!FLIGHT_MODE(GPS_RESCUE_MODE)))) {
----------------------------------------------------------*/

            rcCommand[YAW] = rcCommandBuff.Z;
//        }

    }
…
}

I built the code on a local copy of betaflight

make TARGET=SITL

and ran it with Gazebo10.

HEADFREE and HEADFREE+ANGLE works fine.

MichelArhipov, 2020-11-11:
[A small video fragment from Gazebo10](http://st.ural.ru/MichelArhipov/headfree+angle.avi)

mikeller, 2020-11-11:
Duplicate of #10296.

closed 2020-11-11 by mikeller

haslinghuis, 2020-11-12:
@etracer65 I like this proposal but it's above my grade - what's your point?

mikeller, 2020-11-12:
@haslinghuis: Let's keep the discussion on this focused on #10296.