betaflight/betaflight #10341: Out put yaw pitch roll command to quick test new drone design.

ZhangBo-China, 2020-11-11:
To put it simply, I want to output yaw pitch roll command that solved by flight-controller with serial ports. In fact, I am working on making a new kind of drone, and I have packed up the yaw pitch roll command adapt to my drone.  I want to test my design  quickly, but I am not familiar with flight-controller.  
For example, I want the drone maintain self-stability, but there is a disturbance, and the drone begin to spin right, flight-controller sense the spin, sent a command to motion-solver to make the drone  spin left, and the command sent to the motion-solver is what I want.
I am familiar with STM32, and I am  using a board with F407VE to packed up the yaw pitch roll command adapting to my drone that I mentioned. I also have a hobbywing-XRotor-FlightController-F4-G3. 
Thanks guys!
I accidentally know from Tom Stanton' Swashplateless Helicopter that the  INAV may do that, but I did not find the solution, either.

mikeller, 2020-11-11:
@ZhangBo-China: Sounds like the MSP protocol that is used between the flight controller and configurator  and other apps is the way to get what you need. Have a look at how the communication with the flight controller is implemented in configurator, especially pertaining to the RX channel display and 3D model on the setup tab.

ZhangBo-China, 2020-11-12:
> @ZhangBo-China: Sounds like the MSP protocol that is used between the flight controller and configurator and other apps is the way to get what you need. Have a look at how the communication with the [...]
Thanks for answering! I have seen the Configuration/MSP, it sames that I can enable Telemetry Output with many protocol, if I choose MAVLink, for example, I wander what data the flight controller sent by it, and if there is a way that I can make the flight controller send the data I want. Do I have to change the  source code of betaflight?

mikeller, 2020-11-12:
@ZhangBo-China: Yes, if you want to implement your own protocol instead of using one of the existing ones you will have to modify the firmware. Keep in mind that such modifications will not be accepted into the firmware if they are just re-implementing functionality that already exists in other protocols, like in what you are proposing.

ZhangBo-China, 2020-12-04:
OK! Thanks for answering. I will try.